COVESA/vsomeip #572: Linux to Linux Communication

jsb9945, 2023-11-13:
I don't know why it's not working. I'm asking for your help
I am currently executing requests and responses in sample.
I didn't modify the cpp file and only the config file appears on the wiki, so I modified it and am running it.

=======================================================
 **s32_udp.json**

{
    "unicast" : "192.168.0.145",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/var/log/vsomeip.log" },
        "dlt" : "false"
    },
    "applications" :
    [
        {
            "name" : "client-sample",
            "id" : "0x1313"
        }
    ],
    "routing" : "client-sample",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}
================================================
 **desktop_udp.json**

{
    "unicast" : "192.168.0.60",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/tmp/vsomeip.log" },
        "dlt" : "false"
    },
    "applications" :
    [
        {
            "name" : "service-sample",
            "id" : "0x1212"
        }
    ],
    "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "unreliable" : "30509"
        }
    ],
    "routing" : "service-sample",
    "service-discovery" :
    {
        "enable" : "true",
[... 80 lines cut ...]
2023-11-14 00:13:49.190650 [info] shutdown thread id from application: 1212 (service-sample) is: 7fbd25b41700 TID: 97527
2023-11-14 00:13:49.191443 [info] Watchdog is disabled!
Application service-sample is registered.
2023-11-14 00:13:49.191846 [info] io thread id from application: 1212 (service-sample) is: 7fbd26c3c740 TID: 97524
2023-11-14 00:13:49.192025 [info] io thread id from application: 1212 (service-sample) is: 7fbd24b3f700 TID: 97529
2023-11-14 00:13:49.192043 [info] OFFER(1212): [1234.5678:0.0] (true)
2023-11-14 00:13:49.192810 [info] vSomeIP 3.1.37.1 | (default)
2023-11-14 00:13:49.193121 [info] Listening at /tmp/vsomeip-1212
2023-11-14 00:13:49.193257 [info] Network interface "enp42s0" state changed: up
2023-11-14 00:13:49.193302 [info] OFFER(1212): [1235.5678:0.0] (true)
2023-11-14 00:13:49.193550 [info] Port configuration missing for [1235.5678]. Service is internal.
2023-11-14 00:13:49.193717 [info] Route "default route (0.0.0.0/0) if: enp42s0 gw: 192.168.0.1" state changed: up
2023-11-14 00:13:49.194146 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992 (1703936) local port:30490
2023-11-14 00:13:49.194276 [debug] Joining to multicast group 224.224.224.245 from 192.168.0.60
2023-11-14 00:13:49.194429 [info] udp_server_endpoint_impl<multicast>: SO_RCVBUF is: 212992 (1703936) local port:30490
2023-11-14 00:13:49.195019 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992 (1703936) local port:30509
2023-11-14 00:13:49.195118 [info] SOME/IP routing ready.
2023-11-14 00:13:49.195308 [warning] Route "224.0.0.0/4 if: enp42s0 gw: n/a" state changed: up
2023-11-14 00:13:49.195409 [warning] Route "224.224.224.245/32 if: enp42s0 gw: n/a" state changed: up
2023-11-14 00:13:59.196106 [info] vSomeIP 3.1.37.1 | (default)
2023-11-14 00:13:59.215791 [info] STOP OFFER(1212): [1234.5678:0.0] (true)
2023-11-14 00:13:59.216295 [info] STOP OFFER(1212): [1235.5678:0.0] (true)

I'd appreciate your help.
Thank you.

fcmonteiro, 2023-11-13:
Your host 2 is using vSomeIP 3.1.x
Please retest both hosts with the latest vSomeIP 3.4.x

jsb9945, 2023-11-13:
I changed it to 3.4 and ran it, but it says it's the same..

fcmonteiro, 2023-11-14:
` 2023-11-14 00:13:49.193550 [info] Port configuration missing for [1235.5678]. Service is internal.`

You are not providing the port configuration in the host2 json file, and the offers are not being sent to the network.
Please try to add something like this in your json file (adapt if necessary),
`    "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "reliable" : { "port" : "31111", "enable-magic-cookies" : "false" },
            "unreliable" : "31111"
        }
    ],`

Also check with wireshark if you see the offers/finds in the network.

closed 2024-05-08 by goncaloalmeida