COVESA/vsomeip #579: Attribute change from unreliable to reliable

andrasi, 2023-11-20:
Hello,

I have a service which sends the temperature via an attribute and I want to switch it over TCP.

With the configuration detailed below I get the following mesage on client side:

`[warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [043e.567c.80fc] 3`

VSomeIP version used: 3.1.20.3

My configuration looks like below:

`#server-side deployment:
.depl

define my.someip.deployment for interface my.tempmon.MyService {
    SomeIpServiceID = 1086

    method getTemperature {
        SomeIpMethodID = 32700
    }

    attribute temperatureNotifier {
        SomeIpAttributeReliable = true
        SomeIpNotifierID = 38103
        SomeIpGetterID = 24004
        SomeIpNotifierEventGroups = { 33020 }
    }

}

define my.someip.deployment for provider as Service {
    instance my.tempmon.MyService {
        InstanceId = "my.tempmon.MyService"
        SomeIpInstanceID = 22140
    }
}

#server-side definition:
.fidl

interface MyService {
    version { major 1 minor 0 }
  
    attribute Float temperatureNotifier readonly
   
    method getTemperature {
        out {
			Float temperature
		}
	}
}

#vsomeip.json server side:

{
    "unicast" : "192.168.225.1",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/var/log/vsomeip.log" },
        "dlt" : "false"
    },
    "applications" :
    [
        {
            "name" : "MyApp",
            "id" : "0x1212"
        }
    ],
    "services" :
    [
        {
            "service" : "0x043E",
            "instance" : "0x567C",
            "reliable" : "30509"
        }
	],
	"routing" : "MyApp",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}
`

`#vsomeip.json client side:

{

    "unicast" : "192.168.225.13",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/var/log/vsomeip.log" },
        "dlt" : "false"
    },

    "applications" :
    [
        {
            "name" : "client-sample",
            "id" : "0x1313"
        }
    ],

    "services": [
            {
            "service" : "0x043E",
            "instance" : "0x567C",
            "unicast" : "192.168.225.1",
            "reliable" : "30509"
            }
    ],
    "routing" : "client-sample",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}`

Both client & server were built against the same .fdepl&fidl files.

Any ideas?

Thank you!

andrasi, 2023-11-20:
Hi @goncaloalmeida ! Thanks for the fast answer. Do you know of any issues regarding my problem which have been fixed in the newer versions?

andrasi, 2023-11-23:
Hi @goncaloalmeida ! Followed your advice and switched to the latest versions and now I get this new error:

`Configuration module could not be loaded`

goncaloalmeida, 2024-09-30:
@andrasi _Configuration module could not be loaded!_ usually points to an issue with LD_LIBRARY_PATH.
Please check if the path actually points to the vsomeip libraries, and try running ldconfig to update the library cache.

See https://github.com/COVESA/vsomeip/issues/466 for more info.

cc @duartenfonseca

goncaloalmeida, 2025-03-05:
@andrasi is your problem solved?
cc @duartenfonseca