COVESA/vsomeip #588: ! Failed to load a two json files in a single process for two application with the usage of VSOMEIP_CONFIGURATION_application this configuration

vramasubbu, 2023-12-15:
I have make a vsomeip communication with two VLANS by using of single process. 
so i have created two json files set the env path by using this environment VSOMEIP_CONFIGURATION_<application>.

When i launched the process am able to see the two some-ip application are Initializing . but it loading only one json file at the beginning only . its not taking the other json file.

Please let me know if am doing anything wrong here.

[image]

[image]

fcmonteiro, 2023-12-15:
Hi @vramasubbu, this feature was reverted temporally because it was failing some internal tests.
We are currently working to fix it.
Thanks

vramasubbu, 2023-12-19:
@fcmonteiro  , Thankyou , 
Is there any other possible methods to load two json files in a single process ? 
for the two VLAN communications through Single Process.

zhaoxin39913, 2023-12-26:
I can load multiple JSON files for different vsomeip instances in a single process, but I meet another issue https://github.com/COVESA/vsomeip/issues/540.

duartenfonseca, 2025-03-24:
This appears to not occur anymore

closed 2025-03-24 by duartenfonseca