COVESA/vsomeip #590: Communication between 2 Devives

akashnagpalofficial, 2023-12-15:
Hello I have two devices
1. QNX OS 
vSomeIp v3.20.0.3

2. Pi Os
vSomeIp v3.4.10

I have tried subscribe-sample.cpp and notify-sample.cpp

In both the terminal there is no error seen but it does not seem to be worked.

Can someone help on this if they have successfully able to try with the above combination of devices.

In my case I am able to execute it successfully on the same node.

reymor, 2024-01-02:
Hello @akashnagpalofficial 

There are a lot of things that could be wrong and you are not providing enough information. I could suggest to use a network packet analyzer in one side like wireshark/tshark/tcpdump. 

Have you create the route?, I mean you have a service discovery which sends messages by a multicast IP. Assuming that you are running the service discovery in both devices. 

What is you .json communication? 

Check this: [Communication between two devices](https://github.com/COVESA/vsomeip/wiki/vsomeip-in-10-minutes#communication-between-two-devices)

duartenfonseca, 2024-09-26:
closed due to no response from reporter

closed 2024-09-26 by duartenfonseca