COVESA/vsomeip #592: not sent Subscription: race between services with shared endpoint

akhzarj, 2023-12-18:
Description:

In some conditions we have observe that Subscribtion() message is not send to wire and it is simply dropped when endpoint connection establishment return false while previously another service/event Subscription()  already in progress, look like some race condition is there. More details could be found in provided diagramm (Obsidian canvas or exported png).

Provided artifacts and links:
[image]

[vsomeip.zip Obsidian canvas ](https://github.com/COVESA/vsomeip/files/13705141/vsomeip.zip)

[Obsidian download](https://obsidian.md/download)

RuiG19, 2024-01-15:
Hello @akhzarj 

I've tried to reproduce he issue you described, with two example applications, one offering two services on the same endpoint, and the other subscribing to them. 
The subscriber application cycles between subscription and stop subscription.
However, it was not able to reproduce the issue.

Examples used:
Application offer services: [examples/service.cpp](https://github.com/RuiG19/vsomeip/blob/test_covesa_592/examples/service.cpp)
Application subscribing to the services: [examples/client.cpp](https://github.com/RuiG19/vsomeip/blob/test_covesa_592/examples/client.cpp)

@akhzarj do you have any hint to what I could change in the examples to help reproduce the issue?
Also, which version of the vsomeip-lib were you using, and is this issue reoccurring in your sw?

mstv, 2024-03-19:
Hello @RuiG19,

this race condition can be seen with a client subscribing to two services offered in one Offer message (after StopOffer for these services). `_endpoint->set_established(true);` is called too late.
It nearly always occurs with a longer delay in `routing_manager_impl::service_endpoint_connected`:
https://github.com/COVESA/vsomeip/blob/55284c518f805a2e0194e23770f7c0f2b352ed64/implementation/routing/src/routing_manager_impl.cpp#L4736-L4742
https://github.com/COVESA/vsomeip/blob/55284c518f805a2e0194e23770f7c0f2b352ed64/implementation/routing/src/routing_manager_impl.cpp#L4307-L4321