COVESA/vsomeip #600: how to use special client port in RR model?

fengmao31, 2023-12-26:
It look that the ecu limit the client's port to connect. how can I set the client port for the RR model.
```
2023-12-26 14:22:49.826915 [info] Using configuration file: "./vsomeip-client.json".
2023-12-26 14:22:49.827376 [info] Parsed vsomeip configuration in 0ms
2023-12-26 14:22:49.827479 [info] Configuration module loaded.
2023-12-26 14:22:49.827526 [info] Security disabled!
2023-12-26 14:22:49.827569 [info] Initializing vsomeip (3.4.9.1) application "ADCC".
2023-12-26 14:22:49.828330 [info] Instantiating routing manager [Host].
2023-12-26 14:22:49.829058 [info] create_routing_root: Routing root @ /tmp/vsomeip-0
2023-12-26 14:22:49.829645 [info] Service Discovery enabled. Trying to load module.
2023-12-26 14:22:49.831666 [info] Service Discovery module loaded.
2023-12-26 14:22:49.831971 [info] Application(ADCC, 5555) is initialized (11, 100).
2023-12-26 14:22:49.832100 [info] REQUEST(5555): [b005.0001:1.1]
2023-12-26 14:22:49.832560 [info] create_local_server: Listening @ /tmp/vsomeip-5555
2023-12-26 14:22:49.833002 [info] Starting vsomeip application "ADCC" (5555) using 2 threads I/O nice 255
2023-12-26 14:22:49.833467 [info] Client [5555] routes unicast:192.168.62.11, netmask:255.255.255.0
2023-12-26 14:22:49.834364 [info] Watchdog is disabled!
2023-12-26 14:22:49.834763 [info] io thread id from application: 5555 (ADCC) is: 7f539d5d2040 TID: 209801
2023-12-26 14:22:49.834842 [info] io thread id from application: 5555 (ADCC) is: 7f539acb9700 TID: 209807
2023-12-26 14:22:49.835718 [info] vSomeIP 3.4.9.1 | (default)
2023-12-26 14:22:49.835823 [info] shutdown thread id from application: 5555 (ADCC) is: 7f539bcbb700 TID: 209805
2023-12-26 14:22:49.836044 [info] main dispatch thread id from application: 5555 (ADCC) is: 7f539c4bc700 TID: 209804
CLIENT: Service [b005.1] is NOT available.
2023-12-26 14:22:49.836227 [info] Network interface "vlan62" state changed: up
before while running_------------0
2023-12-26 14:22:49.836937 [info] Route "239.0.0.255/32 if: vlan62 gw: n/a" state changed: up
2023-12-26 14:22:49.837261 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992 (1703936) local port:30490
2023-12-26 14:22:49.837378 [debug] Joining to multicast group 239.0.0.255 from 192.168.62.11
2023-12-26 14:22:49.837531 [info] SOME/IP routing ready.
2023-12-26 14:22:49.837666 [info] udp_server_endpoint_impl<multicast>: SO_RCVBUF is: 212992 (1703936) local port:30490
2023-12-26 14:22:50.107278 [info] endpoint_manager_impl::create_remote_client: 192.168.62.2:30500 reliable: 1 using local port: 0
2023-12-26 14:22:50.108105 [info] endpoint_manager_impl::create_remote_client: 192.168.62.2:30500 reliable: 0 using local port: 0
2023-12-26 14:22:50.109233 [info] udp_client_endpoint_impl::connect: SO_RCVBUF is: 212992 (1703936) local port:0 remote:192.168.62.2:30500
CLIENT: Service [b005.1] is available.
blocked_------------1
Client/Session [5555/0001] sent a request to Service [b005.0001]
2023-12-26 14:22:50.118134 [warning] tcp_client_endpoint receive_cbk: Connection reset by peer(104) local: 192.168.62.11:38737 remote: 192.168.62.2:30500
2023-12-26 14:22:50.118243 [warning] tcp_client_endpoint receive_cbk restarting.
2023-12-26 14:22:50.118447 [warning] service_endpoint_disconnected: lost connection to remote service: [b005.0001]
2023-12-26 14:22:50.118493 [warning] tce::restart: local:  remote: 192.168.62.2:30500
CLIENT: Service [b005.1] is NOT available.
2023-12-26 14:22:50.125080 [warning] connect_cbk: resume sending to: 192.168.62.2:30500
CLIENT: Service [b005.1] is available.
2023-12-26 14:22:50.128561 [warning] tcp_client_endpoint receive_cbk: Connection reset by peer(104) local: 192.168.62.11:42929 remote: 192.168.62.2:30500
2023-12-26 14:22:50.128667 [warning] tcp_client_endpoint receive_cbk restarting.
2023-12-26 14:22:50.128838 [warning] service_endpoint_disconnected: lost connection to remote service: [b005.0001]
2023-12-26 14:22:50.128905 [warning] tce::restart: local:  remote: 192.168.62.2:30500
CLIENT: Service [b005.1] is NOT available.
CLIENT: Service [b005.1] is available.
2023-12-26 14:22:50.138938 [warning] tcp_client_endpoint receive_cbk: Connection reset by peer(104) local: 192.168.62.11:34551 remote: 192.168.62.2:30500
2023-12-26 14:22:50.139134 [warning] tcp_client_endpoint receive_cbk restarting.
2023-12-26 14:22:50.139284 [warning] service_endpoint_disconnected: lost connection to remote service: [2203.0001]
2023-12-26 14:22:50.139323 [warning] service_endpoint_disconnected: lost connection to remote service: [2239.0001]
2023-12-26 14:22:50.139369 [warning] service_endpoint_disconnected: lost connection to remote service: [223b.0001]
2023-12-26 14:22:50.139397 [warning] service_endpoint_disconnected: lost connection to remote service: [224a.0001]
2023-12-26 14:22:50.139424 [warning] service_endpoint_disconnected: lost connection to remote service: [22d4.0001]
2023-12-26 14:22:50.139504 [warning] service_endpoint_disconnected: lost connection to remote service: [22d7.0001]
2023-12-26 14:22:50.139531 [warning] service_endpoint_disconnected: lost connection to remote service: [22db.0001]
2023-12-26 14:22:50.139565 [warning] service_endpoint_disconnected: lost connection to remote service: [22dd.0001]
2023-12-26 14:22:50.139600 [warning] service_endpoint_disconnected: lost connection to remote service: [22e1.0001]
2023-12-26 14:22:50.139632 [warning] service_endpoint_disconnected: lost connection to remote service: [2306.0001]
2023-12-26 14:22:50.139660 [warning] service_endpoint_disconnected: lost connection to remote service: [232c.0001]
2023-12-26 14:22:50.139698 [warning] service_endpoint_disconnected: lost connection to remote service: [2407.0001]
2023-12-26 14:22:50.139761 [warning] service_endpoint_disconnected: lost connection to remote service: [4902.0002]
2023-12-26 14:22:50.139788 [warning] service_endpoint_disconnected: lost connection to remote service: [6402.0001]
2023-12-26 14:22:50.139819 [warning] service_endpoint_disconnected: lost connection to remote service: [6403.0001]
2023-12-26 14:22:50.139845 [warning] service_endpoint_disconnected: lost connection to remote service: [6404.0001]
2023-12-26 14:22:50.139871 [warning] service_endpoint_disconnected: lost connection to remote service: [6411.0001]
2023-12-26 14:22:50.139896 [warning] service_endpoint_disconnected: lost connection to remote service: [6415.0001]
2023-12-26 14:22:50.139923 [warning] service_endpoint_disconnected: lost connection to remote service: [641b.0001]
[... 288 lines cut ...]
    [
        {    
            "service" : "0xB005",
            "instance" : "0x0001",
            "unreliable" : ["30501","30502","30503"]
        }
    ],
    "routing" : "ADCC",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "0",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "30",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "1000",
        "request_response_delay" : "0"
    }
}
```
This vsomeip.josn can work in event model but cannot work in RR model.

fengmao31, 2023-12-26:
I try to search the answer and try to set the `message->set_reliable(false);`
https://github.com/COVESA/vsomeip/issues/56
but I get the result below. I donnot know where the 00 coming from because I cannot catch any someip package not sd package by wireshark.
```
2023-12-26 15:24:43.031923 [info] Using configuration file: "./vsomeip-client.json".
2023-12-26 15:24:43.032271 [info] Parsed vsomeip configuration in 0ms
2023-12-26 15:24:43.032332 [info] Configuration module loaded.
2023-12-26 15:24:43.032372 [info] Security disabled!
2023-12-26 15:24:43.032422 [info] Initializing vsomeip (3.4.9.1) application "ADCC".
2023-12-26 15:24:43.032821 [info] Instantiating routing manager [Host].
2023-12-26 15:24:43.033348 [info] create_routing_root: Routing root @ /tmp/vsomeip-0
2023-12-26 15:24:43.033681 [info] Service Discovery enabled. Trying to load module.
2023-12-26 15:24:43.034944 [info] Service Discovery module loaded.
2023-12-26 15:24:43.035287 [info] Application(ADCC, 5555) is initialized (11, 100).
2023-12-26 15:24:43.035425 [info] REQUEST(5555): [b005.0001:1.1]
2023-12-26 15:24:43.035807 [info] create_local_server: Listening @ /tmp/vsomeip-5555
2023-12-26 15:24:43.036312 [info] Starting vsomeip application "ADCC" (5555) using 2 threads I/O nice 255
before while running_------------0
2023-12-26 15:24:43.036938 [info] Client [5555] routes unicast:192.168.62.11, netmask:255.255.255.0
2023-12-26 15:24:43.037019 [info] shutdown thread id from application: 5555 (ADCC) is: 7f4f1cea8700 TID: 237155
2023-12-26 15:24:43.036977 [info] main dispatch thread id from application: 5555 (ADCC) is: 7f4f14ea8700 TID: 237154
2023-12-26 15:24:43.037915 [info] Watchdog is disabled!
CLIENT: Service [b005.1] is NOT available.
2023-12-26 15:24:43.038322 [info] io thread id from application: 5555 (ADCC) is: 7f4f1dfbe040 TID: 237151
2023-12-26 15:24:43.038346 [info] io thread id from application: 5555 (ADCC) is: 7f4f177fe700 TID: 237157
2023-12-26 15:24:43.039548 [info] vSomeIP 3.4.9.1 | (default)
2023-12-26 15:24:43.040752 [info] Network interface "vlan62" state changed: up
2023-12-26 15:24:43.041899 [info] Route "239.0.0.255/32 if: vlan62 gw: n/a" state changed: up
2023-12-26 15:24:43.042397 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992 (1703936) local port:30490
2023-12-26 15:24:43.042653 [debug] Joining to multicast group 239.0.0.255 from 192.168.62.11
2023-12-26 15:24:43.043140 [info] SOME/IP routing ready.
2023-12-26 15:24:43.043346 [info] udp_server_endpoint_impl<multicast>: SO_RCVBUF is: 212992 (1703936) local port:30490
2023-12-26 15:24:43.661009 [info] endpoint_manager_impl::create_remote_client: 192.168.62.2:30500 reliable: 1 using local port: 0
2023-12-26 15:24:43.661298 [info] configuration_impl:find_specific_port #1: service: b005 instance: 1 reliable: 0 return specific port: 30501
2023-12-26 15:24:43.661362 [info] endpoint_manager_impl::create_remote_client: 192.168.62.2:30500 reliable: 0 using local port: 30501
2023-12-26 15:24:43.661964 [info] udp_client_endpoint_impl::connect: SO_RCVBUF is: 212992 (1703936) local port:30501 remote:192.168.62.2:30500
CLIENT: Service [b005.1] is available.
blocked_------------1
Client/Session [5555/0001] sent a request to Service [b005.0001]
2023-12-26 15:24:43.670667 [warning] tcp_client_endpoint receive_cbk: Connection reset by peer(104) local: 192.168.62.11:40227 remote: 192.168.62.2:30500
2023-12-26 15:24:43.670766 [warning] tcp_client_endpoint receive_cbk restarting.
2023-12-26 15:24:43.670980 [warning] service_endpoint_disconnected: lost connection to remote service: [b005.0001]
2023-12-26 15:24:43.671034 [warning] tce::restart: local:  remote: 192.168.62.2:30500
CLIENT: Service [b005.1] is NOT available.
0 b005
CLIENT: Received message with Client/Session [5555/0001] 00
2023-12-26 15:24:43.810826 [warning] SD messages lost from 192.168.62.7 to 239.0.0.255 - session_id[14078]
CLIENT: Service [b005.1] is available.
2023-12-26 15:24:43.679971 [warning] tcp_client_endpoint receive_cbk: Connection reset by peer(104) local: 192.168.62.11:39727 remote: 192.168.62.2:30500
2023-12-26 15:24:43.680364 [warning] tcp_client_endpoint receive_cbk restarting.
CLIENT: Service [b0052023.1] is -12-NOT available.26
 15:24:43.680559 [warning] service_endpoint_disconnected: lost connection to remote service: [b005.0001]
2023-12-26 15:24:43.680662 [warning] tce::restart: local:  remote: 192.168.62.2:30500
CLIENT: Service [b005.1] is available.
2023-12-26 15:24:44.690255 [warning] tcp_client_endpoint receive_cbk: Connection reset by peer(104) local: 192.168.62.11:35711 remote: 192.168.62.2:30500
2023-12-26 15:24:44.690540 [warning] tcp_client_endpoint receive_cbk restarting.
2023-12-26 15:24:44.690785 [warning] service_endpoint_disconnected: lost connection to remote service: [6415.0001]
2023-12-26 15:24:44.690859 [warning] service_endpoint_disconnected: lost connection to remote service: [a40b.0001]
2023-12-26 15:24:44.690959 [warning] service_endpoint_disconnected: lost connection to remote service: [b005.0001]
2023-12-26 15:24:44.691013 [warning] service_endpoint_disconnected: lost connection to remote service: [b029.0001]
2023-12-26 15:24:44.691061 [warning] service_endpoint_disconnected: lost connection to remote service: [c8bc.0001]
2023-12-26 15:24:44.691188 [warning] tce::restart: local:  remote: 192.168.62.2:30500
CLIENT: Service [b005.1] is NOT available.
CLIENT: Service [b005.1] is available.
[... cut ...]
2023-12-26 15:24:44.715463 [warning] service_endpoint_disconnected: lost connection to remote service: [c890.0001]
2023-12-26 15:24:44.715494 [warning] service_endpoint_disconnected: lost connection to remote service: [c893.0001]
2023-12-26 15:24:44.715525 [warning] service_endpoint_disconnected: lost connection to remote service: [c8bc.0001]
2023-12-26 15:24:44.715563 [warning] tce::restart: local:  remote: 192.168.62.2:30500
2023-12-26 15:24:44.715618 [info] Exiting vsomeip application...
```

wulijiao, 2024-04-26:
Hello, may I ask if your problem has been resolved? I also encountered the same problem.