COVESA/vsomeip #603: Multiple routing managers in single process don't work

zhaoxin39913, 2024-01-03:
Hi COVESA team,

We have an user secnario to run multiple routing managers within a single process, however according to our test result this feature is not fully supported in latest version V3.4.9-r1because of the issue https://github.com/COVESA/vsomeip/issues/540 . It would be appreciated if you can let us know when this feature can be fully supported , or how to fix the issue we found. Thanks.

zhaoxin39913, 2024-01-17:
This is issue is getting fixed by commenting out below code (between "#ifdef 0" and "endif" in below function in service_discovery_impl.cpp.  

```
bool service_discovery_impl::check_ipv4_address(
        const boost::asio::ip::address& its_address) const {
    //Check unallowed ipv4 address
    bool is_valid = true;

    static const boost::asio::ip::address_v4::bytes_type its_unicast_address =
            unicast_.to_v4().to_bytes();
    const boost::asio::ip::address_v4::bytes_type endpoint_address =
            its_address.to_v4().to_bytes();
    static const boost::asio::ip::address_v4::bytes_type its_netmask =
            configuration_->get_netmask().to_v4().to_bytes();

    //same address as unicast address of DUT not allowed
    if (its_unicast_address == endpoint_address) {
        VSOMEIP_ERROR << "Subscriber's IP address is same as host's address! : "
                << its_address;
        is_valid = false;
    } else {
#if 0
        const std::uint32_t self = VSOMEIP_BYTES_TO_LONG(its_unicast_address[0],
                its_unicast_address[1], its_unicast_address[2], its_unicast_address[3]);
        const std::uint32_t remote = VSOMEIP_BYTES_TO_LONG(endpoint_address[0],
                endpoint_address[1], endpoint_address[2], endpoint_address[3]);
        const std::uint32_t netmask = VSOMEIP_BYTES_TO_LONG(its_netmask[0],
                its_netmask[1], its_netmask[2], its_netmask[3]);
        if ((self & netmask) != (remote & netmask)) {
            VSOMEIP_ERROR<< "Subscriber's IP isn't in the same subnet as host's IP: "
                    << its_address;
            is_valid = false;
#endif
        }
    }
    return is_valid;
}
```

duartenfonseca, 2024-11-04:
@zhaoxin39913 have you tried using the fix mentioned here: https://github.com/akhzarj/vsomeip/commit/655b0a6da0c77cea026fa35361a9ccde43d9711e.
does it still occurs?

NemanjaTrifunovicRTRK, 2024-11-26:
Hello everyone,

There is a solution for using multiple routing managers with a single process.

1. For the solution, it is mandatory to use the version of vsomeip which supports multiple application configurations with a single process - currently, it is vsomeip 3.5.1. 
2. It is not needed to change the source code of vsomeip libraries by commenting in service_discovery_impl.cpp.
3. For enabling multiple routing managers, the libraries must be built with preprocessor define **VSOMEIP_ENABLE_MULTIPLE_ROUTING_MANAGERS** set to 1. With cmake build, it is enabled by adding flag **-DENABLE_MULTIPLE_ROUTING_MANAGERS=true**.
4. For applications which needs to be **routing manager host**, set usage of the different configuration files.  Each configuration file must have set value of **"routing"** to the name of the application which needs to be the routing manager host. Configuration pre application can be set using environment variable **VSOMEIP_CONFIGURATION_<application_name>=<path_to_configuration_json_file>.**

This solution is tested on Android 14.

closed 2025-01-27 by duartenfonseca

NemanjaTrifunovicRTRK, 2025-11-05:
Hello @getgou,

Analyzing your configuration, I didn't find that you use a different **routing manager**, especially in the same process, so I don't  think you have this issue. 
Generally, the configuration looks fine.