COVESA/vsomeip #604: Support for Package manager

reymor, 2024-01-04:
Hello guys, 

I have seem multiple times problem with library installation. It's true that it is not a complex library and if we follow the instructions it is kind of straightforward. However, it is true that if you face any problem sometimes you need to know different aspects of compilations process, build system, etc. 

So, would you think that it is a good idea to provide vsomeip through a package manager? 

For instance conan or vcpkg. 

If you believe that it is a good idea. I could start working on this. It is just that I do not want to work on something that you feel it is not worthwhile. For me idea behind this is to provide a easy start for learning about someip and kind of clear way of getting the library.

I know that the main problem here is that there different "tunning" parameters that are use in compilation time, but I think that we could provide a package with "default" configuration and if someone needs something specific you will need to compile from source.

Thanks in advance.

goncaloalmeida, 2024-01-05:
@reymor this is a good idea :)
Feel free to start that and if you need help we can discuss details on the Discussion tab.

closed 2024-05-07 by goncaloalmeida

ruchirchauhan, 2024-05-22:
Hello,

Since this discussion is closed, Is there any update on this ? If not, I am available to take up a task of making it available through Conan.io. 

Thanks

goncaloalmeida, 2024-05-22:
@ruchirchauhan no, no updated on this.
sure feel free to create that :) 
Thanks