COVESA/vsomeip #605: Some/IP-SD session counter issue with ipv6 and local link addresses

wennmbj, 2024-01-09:
We discovered an issue in the service discovery module with ipv6 and local link addresses. We use one Linux ECU with vsomeip and an ECU with Autosar Classic. On receiving side of the subscribe ack the session id is incorrect.
[image]

The SD module handles two session counters for one receiving ECU. One for subscribing to a service offered by another ECU and one for the subscribe ack when an ECU subscribed a service from the host ECU.
For the ack in service_discovery_impl::get_session(const boost::asio::ip::address &_address) the address object contains the device e.g. fe80::a8bb:ccff:fedd:2%eth0. For other sd operations the object contains only the address e.g. fe80::a8bb:ccff:fedd:2.

wennmbj, 2024-10-17:
@fcmonteiro & @duartenfonseca: Are you considering merging my fix into main?

duartenfonseca, 2024-10-17:
@wennmbj yes, we are, but we would need some suggestions in how to reproduce on our side the mentioned bug

wennmbj, 2024-10-18:
@duartenfonseca: It is not that difficult. Use one Linux ECU with vsomeip and one other vsomeip stack vendor e.g. Vector Autosar Classic. You just need to use IPv6 local link addresses. Not sure if I can write a unit-test for that case. We are testing this on integration test level with physical hardware.

closed 2026-04-21 by fcmonteiro