COVESA/vsomeip #607: Not able to communicate between 2 linux machines

Vignesh-Murugappan, 2024-01-10:
Hello Everyone!

Need your help on the below error.

I tried to establish someip communication between 2 linux machines.
I have used the service-example.cpp and client-example.cpp files along with their respective json config files whichever mentioned in https://github.com/COVESA/vsomeip/wiki/vsomeip-in-10-minutes , I just changed host machines IP address in place of unicast_address in json file.

But I didn't able to make two machines communicate. I have attached the cpp files and json files which I have used.

Your responses would help me a lot to move forward ! Thanks in Advance...

[service.json](https://github.com/COVESA/vsomeip/files/13882926/service.json)
[client.json](https://github.com/COVESA/vsomeip/files/13882929/client.json)
[service-example.txt](https://github.com/COVESA/vsomeip/files/13882933/service-example.txt)
[client-example.txt](https://github.com/COVESA/vsomeip/files/13882934/client-example.txt)

wennmbj, 2024-01-10:
Would be helpful to see the log. But I guess the *.json files are not loaded. By default it is /etc/vsomeip.
You can also set a specific file with the environment variable: VSOMEIP_CONFIGURATION.
In general this question here is more an item for Discussions and not an Issue.

Vignesh-Murugappan, 2024-01-10:
Dear @wennmbj 

Thanks for your response! I have json file as below,

I ran service-example like below in one linux machine 1:

root@che7-l24712:/home/che7/work/vsomeip-master/examples# VSOMEIP_CONFIGURATION=service.json ../build/examples/service-example
2024-01-10 09:37:46.236103 [info] Using configuration file: "service.json".
2024-01-10 09:37:46.236768 [info] Parsed vsomeip configuration in 0ms
2024-01-10 09:37:46.236880 [info] Configuration module loaded.
2024-01-10 09:37:46.236981 [info] Security disabled!
2024-01-10 09:37:46.237034 [info] Initializing vsomeip (3.4.10) application "World".
2024-01-10 09:37:46.237254 [info] Instantiating routing manager [Host].
2024-01-10 09:37:46.238897 [info] create_routing_root: Routing root @ /tmp/vsomeip-0
2024-01-10 09:37:46.239680 [info] Service Discovery enabled. Trying to load module.
2024-01-10 09:37:46.242178 [info] Service Discovery module loaded.
2024-01-10 09:37:46.242691 [info] Application(World, 1212) is initialized (11, 100).
2024-01-10 09:37:46.243526 [info] create_local_server: Listening @ /tmp/vsomeip-1212
2024-01-10 09:37:46.243786 [info] OFFER(1212): [1234.5678:0.0] (true)
2024-01-10 09:37:46.243879 [info] Starting vsomeip application "World" (1212) using 2 threads I/O nice 255
2024-01-10 09:37:46.244714 [info] Client [1212] routes unicast:10.176.27.28, netmask:255.255.255.0
2024-01-10 09:37:46.246919 [info] Watchdog is disabled!
2024-01-10 09:37:46.247497 [info] io thread id from application: 1212 (World) is: 7f2d40198000 TID: 17661
2024-01-10 09:37:46.245839 [info] main dispatch thread id from application: 1212 (World) is: 7f2d3f888700 TID: 17663
2024-01-10 09:37:46.245657 [info] shutdown thread id from application: 1212 (World) is: 7f2d3f087700 TID: 17664
2024-01-10 09:37:46.248264 [info] io thread id from application: 1212 (World) is: 7f2d37fff700 TID: 17666
2024-01-10 09:37:46.250360 [info] vSomeIP 3.4.10 | (default)
2024-01-10 09:37:46.252919 [info] Network interface "enx607d090e3e23" state changed: up
2024-01-10 09:37:46.254793 [error] netlink_connector::receive_cbk received error message: 2 seq 3
2024-01-10 09:37:56.252683 [info] vSomeIP 3.4.10 | (default)

Client-example in linux machine 2

root@che7-l24731:/home/che7/vsomeip/vsomeip-master# VSOMEIP_CONFIGURATION=examples/client.json build/examples/client-example
2024-01-10 09:37:50.176149 [info] Using configuration file: "examples/client.json".
2024-01-10 09:37:50.176937 [info] Parsed vsomeip configuration in 0ms
2024-01-10 09:37:50.177067 [info] Configuration module loaded.
2024-01-10 09:37:50.177147 [info] Security disabled!
2024-01-10 09:37:50.177209 [info] Initializing vsomeip (3.4.10) application "Hello".
2024-01-10 09:37:50.177453 [info] Instantiating routing manager [Host].
2024-01-10 09:37:50.179124 [info] create_routing_root: Routing root @ /tmp/vsomeip-0
2024-01-10 09:37:50.179933 [info] Service Discovery enabled. Trying to load module.
2024-01-10 09:37:50.182756 [info] Service Discovery module loaded.
2024-01-10 09:37:50.183252 [info] Application(Hello, 1313) is initialized (11, 100).
2024-01-10 09:37:50.183439 [info] REQUEST(1313): [1234.5678:255.4294967295]
2024-01-10 09:37:50.184220 [info] create_local_server: Listening @ /tmp/vsomeip-1313
2024-01-10 09:37:50.184904 [info] Starting vsomeip application "Hello" (1313) using 2 threads I/O nice 255
2024-01-10 09:37:50.185728 [info] Client [1313] routes unicast:10.176.27.41, netmask:255.255.255.0
2024-01-10 09:37:50.186549 [info] shutdown thread id from application: 1313 (Hello) is: 7f504ecbf700 TID: 17793
2024-01-10 09:37:50.187978 [info] Watchdog is disabled!
2024-01-10 09:37:50.186947 [info] main dispatch thread id from application: 1313 (Hello) is: 7f504f4c0700 TID: 17792
2024-01-10 09:37:50.188648 [info] io thread id from application: 1313 (Hello) is: 7f50505d0ec0 TID: 17789
CLIENT: Service [1234.5678] is NOT available.
2024-01-10 09:37:50.189747 [info] io thread id from application: 1313 (Hello) is: 7f504dcbd700 TID: 17795
2024-01-10 09:37:50.191586 [info] vSomeIP 3.4.10 | (default)
2024-01-10 09:37:50.193334 [info] Network interface "enx607d090e3676" state changed: up
2024-01-10 09:37:50.195134 [error] netlink_connector::receive_cbk received error message: 2 seq 3
2024-01-10 09:38:00.193490 [info] vSomeIP 3.4.10 | (default)

wennmbj, 2024-01-11:
I would expect also messages like:
[debug] Joining to multicast group ... from ....
[info] SOME/IP routing ready.
Have you checked your network settings and do you have enabled multicasts?

Vignesh-Murugappan, 2024-01-12:
Hi @wennmbj Thanks for the input. I'll double check my network settings. Meanwhile, Do we need to run routingmanagerd as well?

wennmbj, 2024-01-12:
No the application in the routing section in the conf will take care about the routing e.g.
    "routing" : "World",
    "service-discovery" :

hugojacob, 2024-01-16:
This seems to be a duplicate of issue #492 .

Klanor, 2024-01-23:
Not sure what the reason for this issue but rebuilding the vsomeip lib fixed this problem for me

Vignesh-Murugappan, 2024-01-24:
Hi @Klanor 

Could you please help me with your vsomeip version? I'm using 3.4.10, still see issues.

vromanen, 2024-01-24:
> Hi @Klanor
> 
> Could you please help me with your vsomeip version? I'm using 3.4.10, still see issues.
> 
Hi!

vsomeip - 3.4.10
boost - 1.74.0
I noticed this issue in my test setup, communication in this setup between Ubuntu host machine and Ubuntu guest on Docker which has a virtual adapter connection.
host Ubuntu has version 22.04.3 LTS
guest Ubuntu in Docker has version 20.04.6 LTS
This issue was observed after some updates of the core libs in host machine.
Rebuild of the someip lib resolved this issue for me (or this issue is still present but I have not observed it anymore ~10 tries)

Klanor, 2024-01-24:
> Hi @Klanor
> 
> Could you please help me with your vsomeip version? I'm using 3.4.10, still see issues.

Hi @Vignesh-Murugappan !

vsomeip - 3.4.10
boost - 1.74.0
I noticed this issue in my test setup, communication in this setup between Ubuntu host machine and Ubuntu guest on Docker which has a virtual adapter connection.
host Ubuntu has version 22.04.3 LTS
guest Ubuntu in Docker has version 20.04.6 LTS
This issue was observed after some updates of the core libs in host machine.
Rebuild of the someip lib resolved this issue for me (or this issue is still present but I have not observed it anymore ~10 tries)

Vignesh-Murugappan, 2024-03-05:
Dear All, 

Issue has been resolved after taking the "NL retries patch" https://github.com/COVESA/vsomeip/pull/591.

Thank all for your inputs.

Vignesh-Murugappan, 2024-03-05:
Closing this ticket

closed 2024-03-05 by Vignesh-Murugappan