COVESA/vsomeip #609: 1 Configuration module could not be loaded!

9109547, 2024-01-10:
I will port VSOMIP to the ARM platform.
Run hello_ World_ Client, prompt "1 Configuration module could not be loaded!".
I have tried setting up LD_ LIBRARY_ PATH, followed by ldconfig, but did not solve the problem.
Can someone help me solve this problem？

reymor, 2024-01-10:
Hello @9109547, there is not magic trick with this. I have using vsomeip2 and vsomeip3 in `arm` and `aarch64` platform without any problem. 

Follow this: https://github.com/COVESA/vsomeip/issues/468

If you follow different issues related with this, you will end with a functional working library.

Best regards,

9109547, 2024-01-11:
I have resolved the issue after “export LD-LIBRARY-PATH”. Thanks @reymor

closed 2024-01-11 by 9109547