COVESA/vsomeip #614: /tmp/vsomeip.lck file not removed upon application termination

srchaitu2008, 2024-01-23:
### vSomeip Version

v3.1.20.3

### Environment

Debian 11

### Describe the bug

/tmp/vsomeip.lck file is not deleted when the application is terminated by Ctrl-C.

This issue is present in v3.3.6 as well.

### Reproduction Steps

1) Run the application request-sample. (Present in build/examples folder)
2) /tmp/vsomeip.lck file and some other files are created in /tmp folder.(Image AfterAppStart attached)
3) Stop the application by pressing Ctrl-C.
4) /tmp/vsomeip.lck file is not removed where as other 2 files are gone.(Image AfterAppStop attached)

### Expected behaviour

/tmp/vsomeip.lck should be removed upon application termination.

### Logs and Screenshots

$ ./request-sample 
2024-00-23 16:41:50.159126 [info] Parsed vsomeip configuration in 0ms
2024-00-23 16:41:50.159602 [info] Configuration module loaded.
2024-00-23 16:41:50.159677 [info] Initializing vsomeip application "".
2024-00-23 16:41:50.160096 [info] Instantiating routing manager [Host].
2024-00-23 16:41:50.160379 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
2024-00-23 16:41:50.160846 [info] Service Discovery enabled. Trying to load module.
2024-00-23 16:41:50.162401 [info] Service Discovery module loaded.
2024-00-23 16:41:50.162678 [info] Application(unnamed, 0100) is initialized (11, 100).
Client settings [protocol=UDP:quiet=false:cycle=1000]
2024-00-23 16:41:50.162879 [info] Starting vsomeip application "" (0100) using 2 threads I/O nice 255
2024-00-23 16:41:50.163370 [info] main dispatch thread id from application: 0100 () is: 7f94300dd700 TID: 3494855
2024-00-23 16:41:50.163663 [info] shutdown thread id from application: 0100 () is: 7f942f8dc700 TID: 3494856
2024-00-23 16:41:50.164472 [info] Watchdog is disabled!
Service [1234.5678] is NOT available.
Service [1235.5678] is NOT available.
2024-00-23 16:41:50.165065 [info] REQUEST(0100): [1234.5678:255.4294967295]
2024-00-23 16:41:50.165216 [info] io thread id from application: 0100 () is: 7f94309c9d40 TID: 3494853
2024-00-23 16:41:50.165247 [info] io thread id from application: 0100 () is: 7f942e8da700 TID: 3494858
2024-00-23 16:41:50.165970 [info] Listening at /tmp/vsomeip-100
2024-00-23 16:41:50.166527 [info] vSomeIP 3.1.20.3 | (default)
2024-00-23 16:41:50.167221 [info] Network interface "lo" state changed: up
2024-00-23 16:42:00.167687 [info] vSomeIP 3.1.20.3 | (default)
^C2024-00-23 16:42:03.156776 [info] Stopping vsomeip application "" (0100).
2024-00-23 16:42:03.162412 [info] Exiting vsomeip application...

[image]
[image]

goncaloalmeida, 2024-01-23:
hello @srchaitu2008 
please retest with the last release. vsomeip 3.1.* is no longer supported: https://github.com/COVESA/vsomeip/discussions/420

closed 2024-01-23 by goncaloalmeida

srchaitu2008, 2024-01-23:
Hi @goncaloalmeida 

I will test with the latest release.

This issue is present in v3.3.6 as well.

srchaitu2008, 2024-01-23:
Hi @goncaloalmeida 

I tested with latest version 3.4.10 and the vsomeip.lck file is not deleted in this version as well. Will raise a new issue.