COVESA/vsomeip #615: /tmp/vsomeip.lck file not removed upon application termination

srchaitu2008, 2024-01-23:
### vSomeip Version

v3.4.10

### Environment

Debian 11

### Describe the bug

/tmp/vsomeip.lck file is not deleted when the application is terminated by Ctrl-C.

### Reproduction Steps

1) Run the application request-sample. (Present in build/examples folder)
2) /tmp/vsomeip.lck file and some other files are created in /tmp folder.(Image AfterAppStart_v3_4_10 attached)
3) Stop the application by pressing Ctrl-C.
4) /tmp/vsomeip.lck file is not removed where as other 2 files are gone.(Image AfterAppStop_v3_4_10 attached)

### Expected behaviour

/tmp/vsomeip.lck should be removed upon application termination.

### Logs and Screenshots

$ ./request-sample 
2024-01-23 19:47:39.673820 [info] Parsed vsomeip configuration in 0ms
2024-01-23 19:47:39.675272 [info] Configuration module loaded.
2024-01-23 19:47:39.675549 [info] Security disabled!
2024-01-23 19:47:39.675734 [info] Initializing vsomeip (3.4.10) application "".
2024-01-23 19:47:39.676952 [info] Instantiating routing manager [Host].
2024-01-23 19:47:39.678806 [info] create_routing_root: Routing root @ /tmp/vsomeip-0
2024-01-23 19:47:39.680399 [info] Service Discovery enabled. Trying to load module.
2024-01-23 19:47:39.685672 [info] Service Discovery module loaded.
2024-01-23 19:47:39.686495 [info] Application(unnamed, 0100) is initialized (11, 100).
Client settings [protocol=UDP:quiet=false:cycle=1000]
2024-01-23 19:47:39.687280 [info] Starting vsomeip application "" (0100) using 2 threads I/O nice 255
2024-01-23 19:47:39.689089 [info] Client [0100] routes unicast:127.0.0.1, netmask:255.255.255.0
2024-01-23 19:47:39.689130 [info] shutdown thread id from application: 0100 () is: 7fef7f032700 TID: 3686445
2024-01-23 19:47:39.691913 [info] Watchdog is disabled!
2024-01-23 19:47:39.693228 [info] io thread id from application: 0100 () is: 7fef80937500 TID: 3686441
2024-01-23 19:47:39.693197 [info] io thread id from application: 0100 () is: 7fef7e030700 TID: 3686447
2024-01-23 19:47:39.695333 [info] vSomeIP 3.4.10 | (default)
2024-01-23 19:47:39.695823 [info] main dispatch thread id from application: 0100 () is: 7fef7f833700 TID: 3686444
2024-01-23 19:47:39.696269 [info] Network interface "lo" state changed: up
Service [1234.5678] is NOT available.
Service [1235.5678] is NOT available.
2024-01-23 19:47:39.697195 [info] REQUEST(0100): [1234.5678:255.4294967295]
2024-01-23 19:47:39.698875 [info] create_local_server: Listening @ /tmp/vsomeip-100
2024-01-23 19:47:49.695987 [info] vSomeIP 3.4.10 | (default)
^C2024-01-23 19:47:50.409647 [info] Stopping vsomeip application "" (0100).
2024-01-23 19:47:50.413186 [info] Exiting vsomeip application...
[image]
[image]

srchaitu2008, 2024-02-05:
Can I get an update on this issue please?

srchaitu2008, 2024-02-22:
Hi @goncaloalmeida @reymor 

Can I get an update on this issue please?

I am eagerly waiting for the response.

Please do let me know in case anything is not clear.

Thanks,
Ravi

fcmonteiro, 2024-02-22:
Hi @srchaitu2008, I've tried this on my side and the file was deleted.
Can you check compiling vsomeip with the flag `-DENABLE_SIGNAL_HANDLING=1` to check if it changes the behaviour?
Something like,
```
cmake -Bbuild -DCMAKE_INSTALL_PREFIX=install_folder -DENABLE_SIGNAL_HANDLING=1 vsomeip-lib
cmake --build build --target install -j 16
```

srchaitu2008, 2024-02-22:
Hi @fcmonteiro ,

At my end, I have compiled the vsomeip library by using the below command(from vsomeip-3.4.10/build folder)

`cmake -DENABLE_SIGNAL_HANDLING=1; make`

I compiled the examples by using the below command(from vsomeip-3.4.10/build/example)

`make`

Also, I tried to kill the application by Ctrl-C and also by sending SIGTERM and the file was not deleted in both the cases.

Can you please help with the below info from your end:

1) What version of vsomeip library did you try this on?
2) Will the command you mentioned make any difference when compared to the one I am using?
3) How did you kill the application ?

I appreciate your help in assisting with this issue.

Thanks,
Ravi

BrunoLSilvaCTW, 2024-03-08:
I am able to reproduce the issue by build with `-D ENABLE_SIGNAL_HANDLING=1`, and terminating the `request-sample` with `Ctrl+C`:

```bash
2024-03-08 08:38:52.352657 [info] vSomeIP 3.4.10.4 | (default)
^C2024-03-08 08:38:53.914363 [info] Stopping vsomeip application "client-sample" (0100).
2024-03-08 08:38:53.920808 [info] Exiting vsomeip application...

$ ll /tmp | rg vsomeip
.-w-------     0 vscode 2024-03-08 08:37 vsomeip.lck
```

It also only seems to happen the the application is the routing manager.

cc. @fcmonteiro

fcmonteiro, 2024-03-12:
Hi @srchaitu2008,
I was able to reproduce the issue, and it will be fixed in the next release of vSomeIP.
Can you test the following patch? [456patch.zip](https://github.com/COVESA/vsomeip/files/14570709/456patch.zip)
Thanks

srchaitu2008, 2024-04-08:
Hi @fcmonteiro ,

The patch works fine.

Thanks,
Ravi

closed 2024-11-04 by duartenfonseca