COVESA/vsomeip #617: qnx build v3.4.10 error

CuriousFu, 2024-01-25:
#### - qcc toolchain: aarch64-unknown-nto-qnx7.1.0-g++
#### - vsomeip version:3.4.10
#### - boost - version:boot_1_81_0
---
#### errors：
[image]
[image]
----

CuriousFu, 2024-01-25:
#### details:
```
<char>; std::__1::basic_string<_CharT, _Traits, _Allocator>::allocator_typr = std::__1::allocator<char>]'
 basic_string<_CharT, _Traits, _Allocator>::basic_string(basic_string&& __str, const allocator_type& __a)
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/public/qnx710/target/qnx7/usr/include/c++/v1/string:1890:1: note:   candidate expects 2 arguments, 1 provided
/home/public/qnx710/target/qnx7/usr/include/c++/v1/string:1872:1: note: candidate: 'std::__1::basic_string<_CharT, _Traits, _Allocator>::basic_string(std::__1::basic_string<_CharT, _Traits, _Allocator>&&) [with _CharT = char; _Traits = std::__1::char_traits<char>; _Allocator = std::__1::allocator<char>]'
 basic_string<_CharT, _Traits, _Allocator>::basic_string(basic_string&& __str)
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/public/qnx710/target/qnx7/usr/include/c++/v1/string:1872:1: note:   no known conversion for argument 1 from 'int' to 'std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&&'
/home/public/qnx710/target/qnx7/usr/include/c++/v1/string:1855:1: note: candidate: 'std::__1::basic_string<_CharT, _Traits, _Allocator>::basic_string(const std::__1::basic_string<_CharT, _Traits, _Allocator>&, const allocator_type&) [with _CharT = char; _Traits = std::__1::char_traits<char>; _Allocator = std::__1::allocator<char>; std::__1::basic_string<_CharT, _Traits, _Allocator>::allocator_type = std::__1::allocator<char>]'
 basic_string<_CharT, _Traits, _Allocator>::basic_string(
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/public/qnx710/target/qnx7/usr/include/c++/v1/string:1855:1: note:   candidate expects 2 arguments, 1 provided
/home/public/qnx710/target/qnx7/usr/include/c++/v1/string:1842:1: note: candidate: 'std::__1::basic_string<_CharT, _Traits, _Allocator>::basic_string(const std::__1::basic_string<_CharT, _Traits, _Allocator>&) [with _CharT = char; _Traits = std::__1::char_traits<char>; _Allocator = std::__1::allocator<char>]'
 basic_string<_CharT, _Traits, _Allocator>::basic_string(const basic_string& __str)
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/public/qnx710/target/qnx7/usr/include/c++/v1/string:1842:1: note:   no known conversion for argument 1 from 'int' to 'const std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >&'
/home/public/qnx710/target/qnx7/usr/include/c++/v1/string:1740:1: note: candidate: 'std::__1::basic_string<_CharT, _Traits, _Allocator>::basic_string(const allocator_type&) [with _CharT = char; _Traits = std::__1::char_traits<char>; _Allocator = std::__1::allocator<char>; std::__1::basic_string<_CharT, _Traits, _Allocator>::allocator_type = std::__1::allocator<char>]'
 basic_string<_CharT, _Traits, _Allocator>::basic_string(const allocator_type& __a)
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/public/qnx710/target/qnx7/usr/include/c++/v1/string:1740:1: note:   no known conversion for argument 1 from 'int' to 'const allocator_type&' {aka 'const std::__1::allocator<char>&'}
/home/public/qnx710/target/qnx7/usr/include/c++/v1/string:1729:1: note: candidate: 'std::__1::basic_string<_CharT, _Traits, _Allocator>::basic_string() [with _CharT = char; _Traits = std::__1::char_traits<char>; _Allocator = std::__1::allocator<char>]'
 basic_string<_CharT, _Traits, _Allocator>::basic_string()
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/public/qnx710/target/qnx7/usr/include/c++/v1/string:1729:1: note:   candidate expects 0 arguments, 1 provided
In file included from /home/jfu2/qnx/boost_1_81_0/include/boost/asio/ip/tcp.hpp:25,
                 from /home/jfu2/mylibpackags/buildvsomeipproject/vsomeip-3.4.10/implementation/endpoints/src/client_endpoint_impl.cpp:13:
/home/jfu2/qnx/boost_1_81_0/include/boost/asio/ip/basic_resolver.hpp:645:41: error: no matching function for call to 'std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::basic_string(int&)'
         static_cast<std::string>(service), resolve_flags);
                                         ^
In file included from /home/public/qnx710/target/qnx7/usr/include/c++/v1/__locale:15,
                 from /home/public/qnx710/target/qnx7/usr/include/c++/v1/ios:216,
                 from /home/public/qnx710/target/qnx7/usr/include/c++/v1/ostream:138,
                 from /home/public/qnx710/target/qnx7/usr/include/c++/v1/istream:164,
                 from /home/public/qnx710/target/qnx7/usr/include/c++/v1/iomanip:48,
                 from /home/jfu2/mylibpackags/buildvsomeipproject/vsomeip-3.4.10/implementation/endpoints/src/client_endpoint_impl.cpp:7:
/home/public/qnx710/target/qnx7/usr/include/c++/v1/string:2122:1: note: candidate: 'std::__1::basic_string<_CharT, _Traits, _Allocator>::basic_string(std::initializer_list<_Ep>, const _Allocator&) [with _CharT = char; _Traits = std::__1::char_traits<char>; _Allocator = std::__1::allocator<char>]'
 basic_string<_CharT, _Traits, _Allocator>::basic_string(
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/public/qnx710/target/qnx7/usr/include/c++/v1/string:2122:1: note:   candidate expects 2 arguments, 1 provided
/home/public/qnx710/target/qnx7/usr/include/c++/v1/string:2110:1: note: candidate: 'std::__1::basic_string<_CharT, _Traits, _Allocator>::basic_string(std::initializer_list<_Ep>) [with _CharT = char; _Traits = std::__1::char_traits<char>; _Allocator = std::__1::allocator<char>]'
 basic_string<_CharT, _Traits, _Allocator>::basic_string(
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
make[4]: *** [CMakeFiles/vsomeip3.dir/all] Error 2
Makefile:135: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/vsomeip.dir/build.make:85: recipe for target 'vsomeip-prefix/src/vsomeip-stamp/vsomeip-build' failed
make[2]: *** [vsomeip-prefix/src/vsomeip-stamp/vsomeip-build] Error 2
CMakeFiles/Makefile2:82: recipe for target 'CMakeFiles/vsomeip.dir/all' failed
make[1]: *** [CMakeFiles/vsomeip.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2
`

duartenfonseca, 2024-09-24:
@CuriousFu did you try the suggestion?

closed 2024-10-28 by duartenfonseca