COVESA/vsomeip #626: Can't see outgoing offer messages in wireshark when using notify-sample

mogar, 2024-02-09:
### vSomeip Version

v3.4.10

### Boost Version

1.71.0-6ubuntu6

### Environment

Ubuntu 20.04.6

### Describe the bug

I'm running the notify-sample example, but don't see any offer messages being sent over my network. Note that if I run client and server on the same machine, the client does receive notifications normally.

I'm monitoring via wireshark, and if I manually transmit a multicast packet then it does show up in wireshark. I manually send multicast packets with `echo -n "hello" >/dev/udp/239.22.0.2/30490`

I start the example with:
`env VSOMEIP_CONFIGURATION=../../config/vsomeip.json VSOMEIP_APPLICATION_NAME=testnotifier ./notify-sample`

This is my config file:
```
{
    "unicast" : "192.168.0.141",
    "logging" :
    {
        "level" : "verbose",
        "console" : "true",
        "file" : { "enable" : "true", "path" : "/tmp/vsomeip.txt" },
        "dlt" : "false"
    },
    "applications" :
    [
        {
            "name" : "testnotifier",
            "id" : "0x1343"
        }
    ],
    "services":
    [
        {
            "service":"1234",
            "instance":"0x5678",
            "reliable" : { "port" : "31111", "enable-magic-cookies" : "false" },
            "unreliable":"31111"
        }
    ],
    "routing" : "testnotifier",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "239.22.0.2",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}
```

This is my log output:
```
2024-02-09 11:53:11.427908 [info] Using configuration file: "../../config/vsomeip.json".
2024-02-09 11:53:11.429913 [info] Parsed vsomeip configuration in 0ms
2024-02-09 11:53:11.430587 [info] Configuration module loaded.
2024-02-09 11:53:11.431225 [info] Security disabled!
2024-02-09 11:53:11.431774 [info] Initializing vsomeip (3.4.10) application "testnotifier".
2024-02-09 11:53:11.433106 [info] Instantiating routing manager [Host].
2024-02-09 11:53:11.434889 [info] create_routing_root: Routing root @ /tmp/vsomeip-0
2024-02-09 11:53:11.436472 [info] Service Discovery enabled. Trying to load module.
2024-02-09 11:53:11.441097 [info] Service Discovery module loaded.
2024-02-09 11:53:11.442270 [info] Application(testnotifier, 1343) is initialized (11, 100).
2024-02-09 11:53:11.442935 [info] offer_event: Event [0009.0064.0002] uses configured cycle time 0ms
2024-02-09 11:53:11.443846 [info] REGISTER EVENT(1343): [0009.0064.0002:is_provider=true]
2024-02-09 11:53:11.444540 [info] Starting vsomeip application "testnotifier" (1343) using 2 threads I/O nice 255
2024-02-09 11:53:11.446517 [info] Client [1343] routes unicast:10.136.24.163, netmask:255.255.255.0
2024-02-09 11:53:11.447522 [info] create_local_server: Listening @ /tmp/vsomeip-1343
2024-02-09 11:53:11.446618 [info] shutdown thread id from application: 1343 (testnotifier) is: 7fa4d7fff700 TID: 1963998
2024-02-09 11:53:11.446750 [info] main dispatch thread id from application: 1343 (testnotifier) is: 7fa4dcc95700 TID: 1963997
2024-02-09 11:53:11.449044 [info] OFFER(1343): [0009.0064:0.0] (true)
2024-02-09 11:53:11.449582 [info] Watchdog is disabled!
Setting event (Length=1).
Application testnotifier is registered.
2024-02-09 11:53:11.454342 [info] io thread id from application: 1343 (testnotifier) is: 7fa4de5a7040 TID: 1963993
2024-02-09 11:53:11.454875 [info] io thread id from application: 1343 (testnotifier) is: 7fa4d6ffd700 TID: 1964000
2024-02-09 11:53:11.455425 [info] vSomeIP 3.4.10 | (default)
Setting event (Length=2).
Setting event (Length=3).
Setting event (Length=4).
Setting event (Length=5).
Setting event (Length=6).
Setting event (Length=7).
Setting event (Length=8).
Setting event (Length=9).
Setting event (Length=1).
2024-02-09 11:53:21.458274 [info] vSomeIP 3.4.10 | (default)
2024-02-09 11:53:21.460301 [info] STOP OFFER(1343): [0009.0064:0.0] (true)
^C2024-02-09 11:53:26.920810 [info] STOP OFFER(1343): [0009.0064:0.0] (true)
2024-02-09 11:53:27.468292 [info] Stopping vsomeip application "testnotifier" (1343).
```

mogar, 2024-02-09:
It looks like the problem was that I had the unicast addresses swapped in my local and remote config files.

closed 2024-02-09 by mogar

rangasridhar07-droid, 2026-02-12:
I use the correct address but still i face the issue where i am not able to capture packets in wireshark