COVESA/vsomeip #629: Configuring incomplete errros occured while adding vsomeip [BUG]: <title>

dappchandu, 2024-02-15:
### vSomeip Version

2.1

### Boost Version

1.71

### Environment

Ubuntu

### Describe the bug

CMake Error at external/vsomeip/CMakeLists.txt:551 (configure_file):
  configure_file attempted to configure a file:
  /home/chandrakala/Desktop/FOTA_PROJECT_TEMPLATE/external/vsomeip/implementation/configuration/include/internal.hpp
  into a source directory.

-- Configuring incomplete, errors occurred!
See also "/home/chandrakala/Desktop/FOTA_PROJECT_TEMPLATE/build/debug/CMakeFiles/CMakeOutput.log".
See also "/home/chandrakala/Desktop/FOTA_PROJECT_TEMPLATE/build/debug/CMakeFiles/CMakeError.log".
make: *** [Makefile:65: all] Error 1
can anyone help me to resolve this thing , Thanks in advance.

### Logs and Screenshots

[image]

fcmonteiro, 2024-02-15:
Hi @dappchandu we have dropped support of vsomeip 2.x.
Please retest with the latest vSomeIP 3.4.10
Thanks

closed 2024-02-15 by fcmonteiro

dappchandu, 2024-02-15:
> Hi @dappchandu we have dropped support of vsomeip 2.x. Please retest with the latest vSomeIP 3.4.10 Thanks

can you provide the link please