COVESA/vsomeip #630: configuring incomplete error

dappchandu, 2024-02-15:
### vSomeip Version

3.4.10

### Boost Version

1.71

### Environment

Ubuntu 20.04.6

### Describe the bug

CMake Error at external/vsomeip/CMakeLists.txt:551 (configure_file):
  configure_file attempted to configure a file:
  /home/chandrakala/Desktop/FOTA_PROJECT_TEMPLATE/external/vsomeip/implementation/configuration/include/internal.hpp
  into a source directory.

-- Configuring incomplete, errors occurred!
See also "/home/chandrakala/Desktop/FOTA_PROJECT_TEMPLATE/build/debug/CMakeFiles/CMakeOutput.log".
See also "/home/chandrakala/Desktop/FOTA_PROJECT_TEMPLATE/build/debug/CMakeFiles/CMakeError.log".
make: *** [Makefile:65: all] Error 1

### Logs and Screenshots

CMake Error at external/vsomeip/CMakeLists.txt:551 (configure_file):
  configure_file attempted to configure a file:
  /home/chandrakala/Desktop/FOTA_PROJECT_TEMPLATE/external/vsomeip/implementation/configuration/include/internal.hpp
  into a source directory.

-- Configuring incomplete, errors occurred!
See also "/home/chandrakala/Desktop/FOTA_PROJECT_TEMPLATE/build/debug/CMakeFiles/CMakeOutput.log".
See also "/home/chandrakala/Desktop/FOTA_PROJECT_TEMPLATE/build/debug/CMakeFiles/CMakeError.log".
make: *** [Makefile:65: all] Error 1
  while adding vsomeip as a git submodule

BrunoLSilvaCTW, 2024-03-08:
Hi @dappchandu,

Can you share with us the steps to reproduce this?

Thanks.

dappchandu, 2024-03-14:
I have added boost and vsomeip as submodules bujt vsomeip is unable to find the boost 
CMake Error in external/vsomeip/CMakeLists.txt:
  Found relative path while evaluating include directories of "vsomeip3":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/CMakeLists.txt:
  Found relative path while evaluating include directories of "vsomeip3-cfg":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/CMakeLists.txt:
  Found relative path while evaluating include directories of "vsomeip3-sd":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/CMakeLists.txt:
  Found relative path while evaluating include directories of "vsomeip3-e2e":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error: install(EXPORT "vsomeip3Targets" ...) includes target "vsomeip3" which requires target "boost_thread" that is not in any export set.
CMake Error: install(EXPORT "vsomeip3Targets" ...) includes target "vsomeip3" which requires target "boost_filesystem" that is not in any export set.
CMake Error in external/vsomeip/examples/routingmanagerd/CMakeLists.txt:
  Found relative path while evaluating include directories of
  "routingmanagerd":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/tools/CMakeLists.txt:
  Found relative path while evaluating include directories of "vsomeip_ctrl":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/examples/CMakeLists.txt:
  Found relative path while evaluating include directories of
  "request-sample":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/examples/CMakeLists.txt:
  Found relative path while evaluating include directories of
  "response-sample":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/examples/CMakeLists.txt:
  Found relative path while evaluating include directories of
  "subscribe-sample":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/examples/CMakeLists.txt:
  Found relative path while evaluating include directories of
  "notify-sample":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/CMakeLists.txt:
  export called with target "vsomeip3" which requires target "boost_thread"
  that is not in any export set.

CMake Error in external/vsomeip/CMakeLists.txt:
  export called with target "vsomeip3" which requires target
  "boost_filesystem" that is not in any export set.

-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_TOOLCHAIN_FILE

CMake Generate step failed.  Build files cannot be regenerated correctly.
make: *** [Makefile:65: all] Error 1
got this error

duartenfonseca, 2024-11-11:
here is an example of building vsomeip with boost and vsomeip as external modules that seems to work, can you try to adapt it to your case?

closed 2025-01-07 by duartenfonseca