COVESA/vsomeip #633: After installing the library trying to add in the cmakelists.txt for a project got error configuring incomplete error

dappchandu, 2024-02-16:
### vSomeip Version

3.4.10

### Boost Version

1.71

### Environment

Ubuntu 20.04.6

### Describe the bug

cmake -S . -B "./build/debug" -DCMAKE_TOOLCHAIN_FILE=cmake/toolchain/linux_native_x86_64.cmake
CMake Error at CMakeLists.txt:32 (target_link_libraries):
  Cannot specify link libraries for target "fota_template" which is not built
  by this project.

-- Configuring incomplete, errors occurred!
See also "/home/chandrakala/Desktop/FOTA_PROJECT_TEMPLATE/build/debug/CMakeFiles/CMakeOutput.log".
make: *** [Makefile:65: all] Error 1

### Reproduction Steps

After downloading vsomeip 3.4.10 , run mkdir build , cd build , cmake .. , make , sudo make install.
Got the library in my system and while trying to add in the project using the 
set(VSOMEIP3_DIR /home/chandrakala/install_folder/lib/cmake/vsomeip3)

# Find vsomeip package
find_package(vsomeip3 REQUIRED)

# Your existing code here ...

target_link_libraries(fota_template vsomeip3)

dappchandu, 2024-02-21:
# disable in source build
set(CMAKE_DISABLE_SOURCE_CHANGES OFF) if it is set to OFF then i'm able to install it .
If it is set to ON i'm unable to install the library.
I want to set it on if i'm not using the vsomeip library how can i made it .

goncaloalmeida, 2024-05-08:
@dappchandu could you please explain why you think this is a vsomeip issue?
Thank you

closed 2024-11-11 by duartenfonseca