COVESA/vsomeip #642: [BUG]: 1 Configuration module could not be loaded!

anonymousAlpha-96, 2024-02-27:
### vSomeip Version

v3.4.10

### Boost Version

1.71

### Environment

UBUNTU 20.04.6LTS

### Describe the bug

Hi team,

Newly I started working on this application, so I followed the steps which is provided by the group [ https://github.com/COVESA/capicxx-someip-tools/wiki/CommonAPI-C---SomeIP-in-10-minutes and https://github.com/COVESA/capicxx-dbus-tools/wiki/CommonAPI-C---D-Bus-in-10-minutes]. Once after the execution of HelloworldService and HelloworldClient application I'm getting an error like **_1 Configuration module could not be loaded!_**  and the command prompt stops responding so any suggestions or supports would be helpful.

Thanks!
[image]

### Logs and Screenshots

[image]

BrunoLSilvaCTW, 2024-02-27:
Hi @anonymousAlpha-96,

`1 Configuration module could not be loaded!` usually points to an issue with `LD_LIBRARY_PATH`. Check if the path actually points to the vsomeip libraries, and try running `ldconfig` to update the library cache.

See #466 for more info.

anonymousAlpha-96, 2024-02-27:
hi @BrunoLSilvaCTW 

I cross checked the above-mentioned issue ticket, but I guess both issues are different. I built the project inside the directory of **capicxx-someip-runtime** repository and the above mentioned is built inside the **vsomeip** directory.
I tried the above-mentioned (#466) procedures, but the issue remains the same.
kindly suggest any other ways.

thanks in advance!

BrunoLSilvaCTW, 2024-02-27:
Can you share the output of the following command?

```bash
find "../install_folder/lib:$PWD/build/" -iname "*libvsomeip3-cfg*"
```

Run it in the same folder where you run the executable. The output should look something like this:

```bash
$ find ./build ./local -iname "*libvsomeip3-cfg*"
./build/vsomeip/libvsomeip3-cfg.so
./build/vsomeip/libvsomeip3-cfg.so.3
./build/vsomeip/libvsomeip3-cfg.so.3.5.0
./local/lib/libvsomeip3-cfg.so
./local/lib/libvsomeip3-cfg.so.3
./local/lib/libvsomeip3-cfg.so.3.5.0
```

anonymousAlpha-96, 2024-02-27:
Hi @BrunoLSilvaCTW ,

Results for the above-mentioned commands:

**_find ./build ./local -iname "*libvsomeip3-cfg*"_**
[image]
[image]

**_find ../install_folder/lib "$PWD/build/" -iname "*libvsomeip3-cfg*"_**
[image]

BrunoLSilvaCTW, 2024-02-27:
It does seem like the libraries are in the correct path. Can you also run this command?

```bash
LD_LIBRARY_PATH="../install_folder/lib:$PWD/build/" COMMONAPI_CONFIG=commonapi4someip.ini strace -o strace.txt build/HelloWorldService
```

This will generate a `strace.txt` file with the system calls that the application is performing. Please, upload the file here so we can have a look.

anonymousAlpha-96, 2024-02-27:
Hi, I'm attaching the strace file for the further reference:

[strace.txt](https://github.com/COVESA/vsomeip/files/14422015/strace.txt)

BrunoLSilvaCTW, 2024-02-27:
```log
openat(AT_FDCWD, "../install_folder/lib/libvsomeip3-cfg.so.3", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
```

It seems that the executable is unable to find the library. Can you try specifying the complete path when calling the executable?
Something like:

```bash
LD_LIBRARY_PATH=/full/path/to/install_folder/lib
```

anonymousAlpha-96, 2024-02-27:
Hi @BrunoLSilvaCTW ,

Yup it works. Thanks a lot.

I'm attaching the final output screen for the future reference with the appropriate command:
_**LD_LIBRARY_PATH="/mnt/d/CPPDemo/ComSomSetUP/install_folder/lib:$PWD/build/" COMMONAPI_CONFIG=commonapi4someip.ini strace -o strace.txt build/HelloWorldService**_

[image]

closed 2024-02-27 by BrunoLSilvaCTW