COVESA/vsomeip #645: [BUG]:

brenkem, 2024-03-06:
### vSomeip Version

v3.4.10

### Boost Version

1.80.0

### Environment

Embedded LINUX (toolchain-11.2-armv7ahf)

### Describe the bug

When I try to compile the mentioned version of vsomeip I am getting the following error and the build fails:

```
PATH="/home/brenkem/git/FIRMWARE_BUILD_TEMPLATE_EFUSA9X/output/host/bin:/home/brenkem/git/FIRMWARE_BUILD_TEMPLATE_EFUSA9X/output/host/sbin:/home/brenkem/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/opt/puppetlabs/bin:/home/brenkem/toolchains/fs-toolchain-11.2-armv7ahf/bin"  /usr/bin/make -j5  -C /home/brenkem/git/FIRMWARE_BUILD_TEMPLATE_EFUSA9X/output/build/vsomeip-3.4.10/
make[1]: Verzeichnis „/home/brenkem/git/FIRMWARE_BUILD_TEMPLATE_EFUSA9X/output/build/vsomeip-3.4.10“ wird betreten
make[2]: Verzeichnis „/home/brenkem/git/FIRMWARE_BUILD_TEMPLATE_EFUSA9X/output/build/vsomeip-3.4.10“ wird betreten
make[3]: Verzeichnis „/home/brenkem/git/FIRMWARE_BUILD_TEMPLATE_EFUSA9X/output/build/vsomeip-3.4.10“ wird betreten
make[3]: Verzeichnis „/home/brenkem/git/FIRMWARE_BUILD_TEMPLATE_EFUSA9X/output/build/vsomeip-3.4.10“ wird verlassen
make[3]: Verzeichnis „/home/brenkem/git/FIRMWARE_BUILD_TEMPLATE_EFUSA9X/output/build/vsomeip-3.4.10“ wird betreten
[  0%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/client_endpoint_impl.cpp.o
[  0%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/credentials.cpp.o
[  4%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/endpoint_definition.cpp.o
[  4%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/endpoint_manager_base.cpp.o
[  4%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/endpoint_impl.cpp.o
<command-line>: error: "_FORTIFY_SOURCE" redefined [-Werror]
<command-line>: note: this is the location of the previous definition
cc1plus: all warnings being treated as errors
CMakeFiles/vsomeip3.dir/build.make:89: recipe for target 'CMakeFiles/vsomeip3.dir/implementation/endpoints/src/credentials.cpp.o' failed
make[3]: *** [CMakeFiles/vsomeip3.dir/implementation/endpoints/src/credentials.cpp.o] Error 1
make[3]: *** Auf noch nicht beendete Prozesse wird gewartet …
<command-line>: error: "_FORTIFY_SOURCE" redefined [-Werror]
<command-line>: note: this is the location of the previous definition
cc1plus: all warnings being treated as errors
CMakeFiles/vsomeip3.dir/build.make:103: recipe for target 'CMakeFiles/vsomeip3.dir/implementation/endpoints/src/endpoint_definition.cpp.o' failed
make[3]: *** [CMakeFiles/vsomeip3.dir/implementation/endpoints/src/endpoint_definition.cpp.o] Error 1
^CCMakeFiles/vsomeip3.dir/build.make:75: recipe for target 'CMakeFiles/vsomeip3.dir/implementation/endpoints/src/client_endpoint_impl.cpp.o' failed
make[3]: *** [CMakeFiles/vsomeip3.dir/implementation/endpoints/src/client_endpoint_impl.cpp.o] Unterbrechung
CMakeFiles/vsomeip3.dir/build.make:131: recipe for target 'CMakeFiles/vsomeip3.dir/implementation/endpoints/src/endpoint_manager_base.cpp.o' failed
make[3]: *** [CMakeFiles/vsomeip3.dir/implementation/endpoints/src/endpoint_manager_base.cpp.o] Unterbrechung
CMakeFiles/vsomeip3.dir/build.make:117: recipe for target 'CMakeFiles/vsomeip3.dir/implementation/endpoints/src/endpoint_impl.cpp.o' failed
make[3]: *** [CMakeFiles/vsomeip3.dir/implementation/endpoints/src/endpoint_impl.cpp.o] Unterbrechung
CMakeFiles/Makefile2:716: recipe for target 'CMakeFiles/vsomeip3.dir/all' failed
make[2]: *** [CMakeFiles/vsomeip3.dir/all] Unterbrechung
Makefile:145: recipe for target 'all' failed
make[1]: *** [all] Unterbrechung
package/pkg-generic.mk:290: recipe for target '/home/brenkem/git/FIRMWARE_BUILD_TEMPLATE_EFUSA9X/output/build/vsomeip-3.4.10/.stamp_built' failed
make: *** [/home/brenkem/git/FIRMWARE_BUILD_TEMPLATE_EFUSA9X/output/build/vsomeip-3.4.10/.stamp_built] Unterbrechung
```

### Reproduction Steps

The used toolchain (toolchain-11.2-armv7ahf) with gcc version 11.2 seems to autoset the macro "_FORTIFY_SOURCE". I was able to get it building by patching the CMakeList.txt according [1]. I attached the patch unter [2].

[1]:
https://patchwork.ozlabs.org/project/qemu-devel/patch/9021f2f9e51e4c7a253d1993ea05f87d0718752f.1359627279.git.mprivozn@redhat.com/

[2]:
```
--- output/build/vsomeip-3.4.10_old/CMakeLists.txt	2023-11-29 15:24:23.000000000 +0100
+++ output/build/vsomeip-3.4.10/CMakeLists.txt	2024-03-06 09:07:47.740029902 +0100
@@ -76,7 +76,7 @@
 
     set(NO_DEPRECATED "")
     set(OPTIMIZE "")
-    set(OS_CXX_FLAGS "${OS_CXX_FLAGS} -D_GLIBCXX_USE_NANOSLEEP -pthread -O -Wall -Wextra -Wformat -Wformat-security -Wconversion -fexceptions -fstrict-aliasing -fstack-protector-strong -fasynchronous-unwind-tables -fno-omit-frame-pointer -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Wpedantic -Werror -fPIE")
+    set(OS_CXX_FLAGS "${OS_CXX_FLAGS} -D_GLIBCXX_USE_NANOSLEEP -pthread -O -Wall -Wextra -Wformat -Wformat-security -Wconversion -fexceptions -fstrict-aliasing -fstack-protector-strong -fasynchronous-unwind-tables -fno-omit-frame-pointer -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Wpedantic -Werror -fPIE")
 
     # force all use of std::mutex and std::recursive_mutex to use runtime init
     # instead of static initialization so mutexes can be hooked to enable PI as needed

```

### Expected behaviour

CMake/MAKE should simply build the project just the way it was working with older gcc versions. Maybe the cross compiling gcc version differs from the x86 version.

sgpires, 2024-03-13:
Hi @brenkem , do you want to create a PR with this change?

brenkem, 2024-03-13:
Hi,

sure. Lets discuss it [here].

[here]:
https://github.com/COVESA/vsomeip/pull/650

closed 2024-03-13 by brenkem