COVESA/vsomeip #647: [BUG]: Unicast service offer not always received if the service offer is sent to vsomeip via unicast UDP.

AlessandroGiusa, 2024-03-07:
### vSomeip Version

3.1.37.1

### Boost Version

1.71

### Environment

Linux

### Describe the bug

Hi,

Could someone please review the function:

```
void endpoint_manager_impl::add_remote_service_info(
        service_t _service, instance_t _instance,
        const std::shared_ptr<endpoint_definition>& _ep_definition) 
```
**implementation/endpoints/src/endpoint_manager_impl.cpp**

Question:
I'm curious why the **unreliable** case is not reported.

In my setup, this results in the service offer sent via unicast UDP being practically ignored because the client_endpoint appears not to be connected later in the processing, even though it actually is (since we got the service offer).

Suggested Fix: Also evaluate must_report for the **unreliable** case.

```
 // if (_ep_definition->is_reliable()) {
            its_endpoint = find_remote_client(_service, _instance, _ep_definition->is_reliable());
            must_report = (its_endpoint && its_endpoint->is_established_or_connected());
            if (must_report)
                its_info = rm_->find_service(_service, _instance);
// }
```

By applying this fix, everything works fine.

The function is basically unchanged from 3.1.37.1 to latest version.

Thanks

BrunoLSilvaCTW, 2024-03-08:
Hi @AlessandroGiusa,

We no longer support vSomeIP version 3.1.37.1.

Please use the latest version (3.4.10), and check if the issue still persists.

closed 2024-03-08 by BrunoLSilvaCTW

AlessandroGiusa, 2024-03-08:
Thank you for the comment. As said in my initial issue description, the code causing the issue is the same in top of tree, hence same behavior can be assumed.

Could you please reopen and consider this issue?

Thanks

BrunoLSilvaCTW, 2024-03-08:
We prefer not to assume.

Please, test with the latest version, and if the issue persists, also share with us the steps to reproduce it.

Thanks.

AlessandroGiusa, 2024-03-08:
Unfortunately I can't update as easy to the latest version of vsomeip.

Thanks