COVESA/vsomeip #648: [HELP]: segementation fault or [error] application_impl::start() caught exception: deque::_M_new_elements_at_front when trying to request to server

athul-am, 2024-03-11:
### vSomeip Version

v3.4.9

### Boost Version

1.71

### Environment

arm-linux-gnueabihf-gcc,arm-linux-gnueabihf-g++

### Describe the bug

i am trying to send some request from my development board (ie.nxp s32grbd2) to another development board(ie.nxp imx6ullevk).I am able to establish the connection with discovery and offering but when i try to send a request from the s32g board it is giving segmentation fault or [error] application_impl::start() caught exception: deque::_M_new_elements_at_front when trying to request to server this message and progress is halted. 
i have installed the library on both boards . s32g is a 64bit board but the imx6 one is a 32bit board is this causing the issue.
Please help what i am doing wrong this instance.

### Reproduction Steps

1.start the client from the S32g board.
2.start the server from the IMX6 board.
3.Once Connection Established, try to send a request.
4.Segmentation fault

### Expected behaviour

The Request is to be accepted and the callback to be invoked which will process the request and send the appropriate response.

### Logs and Screenshots

client log:
2024-03-11 10:28:18.664619 [info] Using configuration file: "../01_Configuration/iNetTest_GW_Single_VM.json".
2024-03-11 10:28:18.667229 [info] Parsed vsomeip configuration in 2ms
2024-03-11 10:28:18.667999 [info] Configuration module loaded.
2024-03-11 10:28:18.669885 [info] Security disabled!
2024-03-11 10:28:18.670277 [info] Initializing vsomeip (3.4.9.1) application "gw".
2024-03-11 10:28:18.672271 [info] Instantiating routing manager [Host].
2024-03-11 10:28:18.679826 [info] create_routing_root: Routing root @ /tmp/vsomeip-0
2024-03-11 10:28:18.683040 [info] Service Discovery enabled. Trying to load module.
2024-03-11 10:28:18.699353 [info] Service Discovery module loaded.
2024-03-11 10:28:18.702289 [info] Application(gw, 0044) is initialized (11, 100).
2024-03-11 10:28:18.703607 [info] REQUEST(0044): [0003.0004:1.0]
2024-03-11 10:28:18.709704 [info] create_local_server: Listening @ /tmp/vsomeip-44
[INETMAN] Register complete for SID = 3, IID = 4, MID = 1

2024-03-11 10:28:18.712244 [info] REQUEST(0044): [0001.0001:1.0]
2024-03-11 10:28:18.713120 [info] Avoid trigger SD find-service message for local service/instance/major/minor: 1/1/1/0
[INETMAN] Register complete for SID = 1, IID = 1, MID = 1

2024-03-11 10:28:18.714080 [info] REQUEST(0044): [0002.0001:255.4294967295]
2024-03-11 10:28:18.714560 [info] Avoid trigger SD find-service message for local service/instance/major/minor: 2/1/255/4294967295
[INETMAN] Register complete for SID = 2, IID = 1, MID = 1

2024-03-11 10:28:18.715946 [info] REQUEST(0044): [0001.0002:1.0]
[INETMAN] Register complete for SID = 1, IID = 2, MID = 1

2024-03-11 10:28:18.716896 [info] REQUEST(0044): [0002.0002:255.4294967295]
[INETMAN] Register complete for SID = 2, IID = 2, MID = 1

2024-03-11 10:28:18.718578 [info] Starting vsomeip application "gw" (0044) using 2 threads I/O nice 255
2024-03-11 10:28:18.722405 [info] Client [0044] routes unicast:192.168.11.20, netmask:255.255.255.0
2024-03-11 10:28:18.721857 [info] main dispatch thread id from application: 0044 (gw) is: 7f7d7401c0 TID: 2598
2024-03-11 10:28:18.722383 [info] shutdown thread id from application: 0044 (gw) is: 7f7cf3f1c0 TID: 2599
2024-03-11 10:28:18.729801 [info] Watchdog is disabled!
( Service : 0x1, instance: 0x1 ) is not available :(

[INETMAN] EVT_ECU_DISCONNECTED evnt is sent on PIPE
[AUTHGW] Received events from pipe
( Service : 0x1, instance: 0x2 ) is not available :(

[INETMAN] EVT_ECU_DISCONNECTED evnt is sent on PIPE
[AUTHGW] ECU Disconnected message is received
[AUTHGW] Received events from pipe
[AUTHGW] ECU Disconnected message is received
( Service : 0x2, instance: 0x1 ) is not available :(

[INETMAN] EVT_ECU_DISCONNECTED evnt is sent on PIPE
[AUTHGW] Received events from pipe
[AUTHGW] ECU Disconnected message is received
( Service : 0x2, instance: 0x2 ) is not available :(

[INETMAN] EVT_ECU_DISCONNECTED evnt is sent on PIPE
( Service : 0x3, instance: 0x4 ) is not available :(

[AUTHGW] Received events from pipe[INETMAN] N-ECU disconnected message is received.
[AUTHGW] ECU Disconnected message is received

2024-03-11 10:28:18.737904 [info] io thread id from application: 0044 (gw) is: 7f7e86a1c0 TID: 2596
2024-03-11 10:28:18.737892 [info] io thread id from application: 0044 (gw) is: 7f6f7fe1c0 TID: 2601
2024-03-11 10:28:18.748262 [info] vSomeIP 3.4.9.1 | (default)
2024-03-11 10:28:18.752821 [info] Network interface "pfe0" state changed: up
2024-03-11 10:28:18.756867 [info] Route "239.192.255.251/32 if: pfe0 gw: n/a" state changed: up
2024-03-11 10:28:18.759591 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992 (1703936) local port:30490
2024-03-11 10:28:18.760935 [debug] Joining to multicast group 239.192.255.251 from 192.168.11.20
2024-03-11 10:28:18.762093 [info] SOME/IP routing ready.
2024-03-11 10:28:18.763474 [info] udp_server_endpoint_impl<multicast>: SO_RCVBUF is: 212992 (1703936) local port:30490
2024-03-11 10:28:22.628702 [info] endpoint_manager_impl::create_remote_client: 192.168.11.22:30509 reliable: 0 using local port: 0
2024-03-11 10:28:22.631034 [info] udp_client_endpoint_impl::connect: SO_RCVBUF is: 212992 (1703936) local port:0 remote:192.168.11.22:30509
( Service : 0x1, instance: 0x1 ) is available :)

[INETMAN] EVT_ECU_CONNECTED evnt is sent on PIPE
[AUTHGW] Received events from pipe
[AUTHGW] ECU Connected message is received
Service id added 1 ecuid added 1 at index[0]
( Service : 0x2, instance: 0x1 ) is available :)

[INETMAN] EVT_ECU_CONNECTED evnt is sent on PIPE
[AUTHGW] Sending SN Request
[MSGROUTER] Packing data for sedning SN_Request
Sent request to server!

server log:
2024-03-11 06:50:13.235601 [info] Using configuration file: "../01_Configuration/iNetTest_ASECU1_Single_VM.json".
2024-03-11 06:50:13.248081 [info] Parsed vsomeip configuration in 3ms
2024-03-11 06:50:13.253258 [info] Configuration module loaded.
2024-03-11 06:50:13.258191 [info] Security disabled!
2024-03-11 06:50:13.264345 [info] Initializing vsomeip (3.4.9.1) application "asecu_1".
2024-03-11 06:50:13.269585 [info] Instantiating routing manager [Host].
2024-03-11 06:50:13.281291 [info] create_routing_root: Routing root @ /tmp/vsomeip-0
2024-03-11 06:50:13.289291 [info] Service Discovery enabled. Trying to load module.
2024-03-11 06:50:13.316755 [info] Service Discovery module loaded.
2024-03-11 06:50:13.328571 [info] Application(asecu_1, 0055) is initialized (11, 100).
2024-03-11 06:50:13.340345 [info] create_local_server: Listening @ /tmp/vsomeip-55
2024-03-11 06:50:13.348841 [info] OFFER(0055): [0001.0001:0.0] (true)
[INETMAN] Register complete for SID = 1, IID = 1, MID = 1

2024-03-11 06:50:13.361728 [info] OFFER(0055): [0002.0001:0.0] (true)
[INETMAN] Register complete for SID = 2, IID = 1, MID = 1

2024-03-11 06:50:13.371541 [info] Starting vsomeip application "asecu_1" (0055) using 2 threads I/O nice 255
2024-03-11 06:50:13.382011 [info] Client [0055] routes unicast:192.168.11.22, netmask:255.255.255.0
2024-03-11 06:50:13.396411 [info] Watchdog is disabled!
2024-03-11 06:50:13.407218 [info] io thread id from application: 0055 (asecu_1) is: 76a41450 TID: 508
2024-03-11 06:50:13.412315 [info] shutdown thread id from application: 0055 (asecu_1) is: 75057450 TID: 511
2024-03-11 06:50:13.429551 [info] vSomeIP 3.4.9.1 | (default)
2024-03-11 06:50:13.444981 [info] Network interface "eth0" state changed: up
2024-03-11 06:50:13.462585 [info] Route "239.192.255.251/32 if: eth0 gw: n/a" state changed: up
2024-03-11 06:50:13.456251 [info] io thread id from application: 0055 (asecu_1) is: 74055450 TID: 513
2024-03-11 06:50:13.451005 [info] main dispatch thread id from application: 0055 (asecu_1) is: 75858450 TID: 510
2024-03-11 06:50:13.481468 [info] udp_server_endpoint_impl: SO_RCVBUFFORCE successful.
2024-03-11 06:50:13.486561 [info] udp_server_endpoint_impl: SO_RCVBUF is: 1703936 (1703936) local port:30490
2024-03-11 06:50:13.491288 [debug] Joining to multicast group 239.192.255.251 from 192.168.11.22
2024-03-11 06:50:13.501458 [info] udp_server_endpoint_impl<multicast>: SO_RCVBUFFORCE: successful.
2024-03-11 06:50:13.503568 [info] udp_server_endpoint_impl<multicast>: SO_RCVBUF is: 1703936 (1703936) local port:30490
2024-03-11 06:50:13.512411 [info] udp_server_endpoint_impl: SO_RCVBUFFORCE successful.
2024-03-11 06:50:13.519845 [info] udp_server_endpoint_impl: SO_RCVBUF is: 1703936 (1703936) local port:30509
2024-03-11 06:50:13.524608 [info] SOME/IP routing ready.
2024-03-11 06:50:13.115025 [error] application_impl::start() caught exception: deque::_M_new_elements_at_front

athul-am, 2024-03-11:
hi @fcmonteiro ,
thank you for the response , i am trying but not able to get the backtrace i cant install any new things in the hardware as it has basic linux without any distribution.
now when i try to compile the hello world examples i am getting this error 
[image]
but my code is still getting compiled and starting someip thread

athul-am, 2024-03-11:
hi @fcmonteiro ,
please let me know if this backtrace is enough 
[backtrace.txt](https://github.com/COVESA/vsomeip/files/14557929/backtrace.txt)

fcmonteiro, 2024-03-11:
@athul-am not really.
I was expecting to see some calls in Thread 1.

athul-am, 2024-03-12:
hi @fcmonteiro ,
sorry for that please find the thread 1 backtrace
[backtracethread1.txt](https://github.com/COVESA/vsomeip/files/14567277/backtracethread1.txt)
please let me know if this is enough

sgpires, 2024-03-12:
Hi @athul-am ,
Can you monitor the memory consumption during the execution, and check if you are running out of memory?

athul-am, 2024-03-13:
hi @sgpires ,
i checked the memory conception it is at 1.0%-1.5% and my system only has this thread running

athul-am, 2024-03-14:
hi @fcmonteiro and @sgpires ,
[image]
this is the valgrind report statement i dont know if this will be helpful

athul-am, 2024-03-15:
Hi @sgpires  and @fcmonteiro,
i was able to to resolve the issue by downgrading the version to 1.7 but still not sure of the root cause of issue.

Thank you for the responses

closed 2024-03-15 by fcmonteiro