COVESA/vsomeip #658: How to fix the issue: "1 Configuration module could not be loaded!" in Android env?

canaanfredi, 2024-03-27:
### vSomeip Version

v3.3.8

### Boost Version

1.84.0

### Environment

Android NDK r25c

### Describe the bug

As I cross compile vsomeip by Android NDK(version: r25c) in my Mac and I **adb push** all the folder to Android AVD.
And I chmod +x for this.
Since I **adb shell** and want run the hello world it shows
[image]
How to fix it?

canaanfredi, 2024-03-27:
I use this cmd below:
```
cmake -B ./build -G Ninja -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK_HOME/build/cmake/android.toolchain.cmake -DANDROID_ABI=x86_64 -S .
cmake --build ./build
```

canaanfredi, 2024-03-27:
build details: https://github.com/canaanfredi/vsomeip-anroid-cmake

canaanfredi, 2024-03-28:
https://github.com/COVESA/vsomeip/blob/master/implementation/configuration/include/internal_android.hpp
[image]
Notice that this header file we tried to rename libvsomeip3-cfg.so TO libvsomeip_cfg.so but we got this:
[image]

007herelsp, 2024-03-28:
dynamic_pointer_cast this issue where some objects return nullptr Created outside of the boundaries of the shared library.
See below discuss for more details.
https://github.com/android/ndk/issues/519
https://github.com/android/ndk/issues/533

haohaolee, 2024-03-28:
Hi @007herelsp 

Thanks very much for the info. I have carefully studied the discussion for all these, my current understanding is:
The only approach for Android is linking the cfg library explicitly. Do I understand correctly?

I have tried some things like
- `dlopen` with DL_GLOBAL
- add a key function to `configuration_plugin`

None of the above worked.

ps: I personally does not like using dynamic_cast between module boundaries, which leads to a dark place in the ABI spec. Maybe we should redesign the plugin system rely on more stable interfaces.

Best regards,

007herelsp, 2024-03-28:
> Hi @007herelsp
> 
> Thanks very much for the info. I have carefully studied the discussions for all these, my current understanding is: The only approach for Android is linking the cfg library explicitly. Do I understand correctly?
> 
> I have tried some things like
> 
> * `dlopen` with DL_GLOBAL
> * add a key function to `configuration_plugin`
> 
> None of the above worked.
> 
> ps: I personally do not like using dynamic_cast cross module boundaries, which leads to a dark place in the ABI spec. Maybe we should redesign the plugin system to rely on more stable interfaces.
> 
> Best regards,

I'm going to submit the code to fix this problem

KooWu, 2024-11-01:
@007herelsp I am very concerned about whether there is a stable version available for Android.

007herelsp, 2024-11-17:
I have fixed the problem and used it, everything works fine No problem found

duartenfonseca, 2024-11-18:
@007herelsp what is the fix you used?

duartenfonseca, 2024-11-18:
this one? https://github.com/007herelsp/vsomeip/commit/f22cf239ef4205c2aae55400c43c7cbf6ab471fb#diff-757cd72756ef920fd2357653c76c087315969557e257a0fc8532ed2c6dca1649L130

007herelsp, 2024-11-23:
Please refer to this submission
https://github.com/007herelsp/vsomeip/commit/7d2f99109478d08ca5123763ea58357d3fbb2c9d#diff-3f0b51061e075d4605bf22f1d35d3946e158349b1562fbfd5a567d812fdda39bR9