COVESA/vsomeip #664: Avoid trigger SD find-service message for local service/instance/major/minor

manojkumarms1999, 2024-04-09:
### vSomeip Version

v3.1.20

### Boost Version

1.73

### Environment

ubuntu

### Describe the bug

I am using an AAA app to act as client on device
on the Other side, I am Offering a same service aaa with service id d0

I was able to subscribe and all worked

suddenly on the same system
These logs arrive at the vsomeip side and Subscribe is not happening 

04-08 17:31:46.411  1652 19174 I vsomeip : Starting vsomeip application "AAA" (3dd0) using 2 threads I/O nice 255
04-08 17:31:46.412  1652 19176 I vsomeip : shutdown thread id from application: 3dd0 (AAA) is: 6dfee2fcb0 TID: 19176
04-08 17:31:46.412  1652 19177 I vsomeip : io thread id from application: 3dd0 (AAA) is: 6dfdd31cb0 TID: 19177
04-08 17:31:46.412  1652 19175 I vsomeip : main dispatch thread id from application: 3dd0 (AAA) is: 6dfef2dcb0 TID: 19175
04-08 17:31:46.413  1652 19174 I vsomeip : io thread id from application: 3dd0 (AAA) is: 6e01129cb0 TID: 19174
04-08 17:31:46.413  1652 19174 I vsomeip : Listening at /dev/socket/vsomeip/SOME_IP_Unsecure_Domain_01-3dd0
04-08 17:31:46.413  1652 19174 I vsomeip : Client 3dd0 (AAA) successfully connected to routing  ~> registering..
04-08 17:31:46.414  1652  1663 I vsomeip : Application/Client 3dd0 is registering.
04-08 17:31:46.414  1652  1662 I vsomeip : Client [100] is connecting to [3dd0] at /dev/socket/vsomeip/SOME_IP_Unsecure_Domain_01-3dd0
04-08 17:31:46.415  1652 19177 I vsomeip : Application/Client 3dd0 (AAA) is registered.
04-08 17:31:46.415  1652  1657 I vsomeip : REGISTERED_ACK(3dd0)
04-08 17:31:46.415  1652  1657 I vsomeip : REGISTER EVENT(3dd0): [00d0.0001.8001:is_provider=0:reliability=ff]
04-08 17:31:46.422  1652  1657 I vsomeip : REQUEST(3dd0): [00d0.0001:1.4294967295]
04-08 17:31:46.422  1652  1657 I vsomeip : Avoid trigger SD find-service message for local service/instance/major/minor: d0/1/1/4294967295
04-08 17:31:46.507  1651  1665 I vsomeip : vSomeIP 3.1.20 | (default)
04-08 17:31:55.820  1652  1657 I vsomeip : vSomeIP 3.1.20 | (default)
04-08 17:31:56.516  1651  1656 I vsomeip : vSomeIP 3.1.20 | (default)
04-08 17:32:05.821  1652  1657 I vsomeip : vSomeIP 3.1.20 | (default)
04-08 17:32:06.522  1651  1665 I vsomeip : vSomeIP 3.1.20 | (default)
04-08 17:32:15.821  1652  1663 I vsomeip : vSomeIP 3.1.20 | (default)

on the pcap, find are not sent, but still on the server side
app is sending a OFFER service

please share your concerns !!

manojkumarms1999, 2024-05-09:
https://github.com/MicrochipTech/some-ip/blob/master/vsomeip/implementation/routing/src/routing_manager_impl.cpp#L335

how to come up with this check here !

fcmonteiro, 2024-05-09:
Hi @manojkumarms1999, please retest with the latest vSomeIP 3.4.x.
We've dropped support to 3.1.x

manojkumarms1999, 2024-05-09:
Hi @fcmonteiro 
the same issues are not occurred on 3.4.x.

Brisingr001, 2024-08-10:
noticing the same error in v3.4

aadarshkt, 2024-10-29:
Noticing the same error with notify-example in v3.5.1