COVESA/vsomeip #665: how to send notification message

fengmao31, 2024-04-10:
### vSomeip Version

v3.4.9.1

### Boost Version

1.84

### Environment

ubuntu20

### Describe the bug

```
   std::shared_ptr<vsomeip::message> notification = vsomeip::runtime::get()->create_notification();
        notification->set_service(SAMPLE_SERVICE_ID);
        notification->set_instance(SAMPLE_INSTANCE_ID);
        notification->set_method(SAMPLE_EVENT_ID);
        app->send(notification);
        //app->notify(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_EVENT_ID, payload);
```
I want to send the notification by message interface. However, I cannot find the way to setup event_id. I try to send message, but the client cannot get the notification.   
` //app->notify(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_EVENT_ID, payload);`can work.

nidheesh-ss, 2024-05-28:
You can send the notification by creating and offering event id in service side and subscribe to this event on client side and you can simple use the app->notify (service_id,instance_id,event_id) 
it will work

duartenfonseca, 2024-09-06:
@fengmao31 have you tried the suggestion?

fengmao31, 2024-09-12:
I've tried it before and it worked, but why doesn't the standard interface function properly? It lacks organization when I'm packaging the library.

duartenfonseca, 2024-09-12:
@fengmao31 
https://github.com/COVESA/vsomeip/blob/master/examples/request-sample.cpp take a look at this example

fengmao31, 2024-09-12:
> @fengmao31 https://github.com/COVESA/vsomeip/blob/master/examples/request-sample.cpp看一下这个例子

I know the example. But you can't send notification message by interface as request interface.

duartenfonseca, 2024-09-25:
have you taken a look at the notify-sample / subscribe-sample? 

On the notify sample side (the app that offers) will do:
app_->offer_event
app_->offer_service
app_->notify

on the subscribe sample side (the client that consumes the offer):

app_->register_message_handler (to read the response from the service)

app_->request_event
app_->subscribe
app_->request_service

this is how it works on that example. 

please take a look at the [vsomeip in 10 minutes](https://github.com/COVESA/vsomeip/wiki/vsomeip-in-10-minutes)

fengmao31, 2024-09-25:
> have you taken a look at the notify-sample / subscribe-sample?
>
> On the notify sample side (the app that offers) will do: app_->offer_event app_->offer_service app_->notify
>
> on the subscribe sample [...]

I know the example. and I can use app_->notify to send the message well. But I can't send notification message by interface as request interface though the message have the notification attribute。

fengmao31, 2024-09-25:
> have you taken a look at the notify-sample / subscribe-sample?
>
> On the notify sample side (the app that offers) will do: app_->offer_event app_->offer_service app_->notify
>
> on the subscribe sample [...]
```
   std::shared_ptr<vsomeip::message> notification = vsomeip::runtime::get()->create_notification();
        notification->set_service(SAMPLE_SERVICE_ID);
        notification->set_instance(SAMPLE_INSTANCE_ID);
        notification->set_method(SAMPLE_EVENT_ID);
        app->send(notification);
        //app->notify(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_EVENT_ID, payload);
```

duartenfonseca, 2024-10-28:
the behaviour you describe is not intended, can you explain better in which setting would you use this like so? maybe we can help with another way of achieving what you want

closed 2024-11-09 by fengmao31