COVESA/vsomeip #666: When testing a cycle event, calling application_impl::stop_offer_event is invalid.

hjkcst, 2024-04-10:
### vSomeip Version

v3.4

### Boost Version

1.7

### Environment

Embedded linux,ubuntu1808

### Describe the bug

When testing a cycle event, calling application_impl::stop_offer_event is invalid. The reference count for the event was not actually cleared, causing the cycle event to still notify.

The root cause is here.
file path:implementation\routing\src\event.cpp
[image]

This problem can be solved with a few modifications.
[image]

### Reproduction Steps

server:
0.Register handler(state、message)
1.offer_service
2.offer_event (set sycle event)
3.notify 
4.stop_offer_event
result:
The server is still notifying, and the client can still receive the notification message.

client:
0.Register handler(state、massage、availability)
1.request_service
2.request_event
3.subscribe(sycle event)
result:
It can receive a notification and continue to receive notifications after stop_offer_event on the server.

### Expected behaviour

server: cycle event can stop notify.
client: stopped receiving notifications

zhaoxin39913, 2024-04-12:
Yes, I come across the same issue, even after stopping service the event still can not be stopped.

hjkcst, 2024-04-26:
Hello, everyone

Does anyone know how to promote the modification of this problem?

thanks,
hjkcst

zhaoxin39913, 2024-04-29:
Hi @hjkcst ，

I think you can follow the community process to contribute your change to related git branch.

hjkcst, 2024-05-06:
HI, everyone

Has you known this problem?
[image]

BrunoLSilvaCTW, 2024-05-06:
Hi @hjkcst,

You need to create a fork first. See this [page](https://docs.github.com/en/get-started/exploring-projects-on-github/contributing-to-a-project) for instructions on how to do it.

duartenfonseca, 2025-04-10:
hi @hjkcst @007herelsp @zhaoxin39913  i am trying to reproduce your problem. for that i used the following network_test: [cyclic_event_test]( https://github.com/COVESA/vsomeip/tree/master/test/network_tests/cyclic_event_tests).
i did a change to add a stop_offer_event after the notify of the service. 

i wasn't able to see the same problem. can you point out some code for the example, or why in my case it didn't work?