COVESA/vsomeip #679: CommonAPI SOME/IP stub implementation _client->getHostAddress() returns "0.0.0.0" rather than true address

srl100, 2024-04-19:
### vSomeip Version

v3.4.10

### Boost Version

1.81

### Environment

Embedded Linux

### Describe the bug

Stub implementations are passed the 'const std::shared_ptr<CommonAPI::ClientId> _client' as their first parameter, but the _client->getHostAddress() does not return the IP address of the client.

### Reproduction Steps

Create a simple method in .fidl, .fdepl.
Implement the stub implementation of this method, and look at the content of the _client->getHostAddress() when the method is invoked from a remote client.

### Expected behaviour

I expect to be able to read the IP address of the client in _client->getHostAddress().

### Logs and Screenshots

Stub Implementation:
```
void VideoCameraStubImpl::setCamExclusive(const std::shared_ptr<CommonAPI::ClientId> _client, uint32_t _ttl, setCamExclusiveReply_t _reply)
{
    (void)_ttl;
    std::cout << "getUid:" << _client->getUid() << std::endl;
    std::cout << "getGid:" << _client->getGid() << std::endl;
    std::cout << "getEnv:" << _client->getEnv() << std::endl;
    std::cout << "getHostAddress:" << _client->getHostAddress() << std::endl;
    _reply();
}
```

Console output:
```
getUid:4294967295
getGid:4294967295
getEnv:
getHostAddress:0.0.0.0
```

lutzbichler, 2024-08-06:
The expectation is "wrong". The host address is provided if you are using TCP to communicate between vsomeip application (no "real" SOME/IP).

closed 2024-08-06 by lutzbichler

srl100, 2024-08-06:
@lutzbichler What do you mean by 'no "real" SOME/IP'? in your comment? Is this lack of UDP host address documented anywhere?

lutzbichler, 2024-08-06:
As said, the host address is filled when you run the vsomeip protocol between applications via TCP instead of UDS. There is no relationship to UDP or SOME/IP.

srl100, 2024-08-06:
@lutzbichler Sorry, I still don't understand you.

From github: "The vsomeip stack implements the http://some-ip.com/ (Scalable service-Oriented MiddlewarE over IP (SOME/IP)) protocol.", so there's a direct relationship between vsomeip and SOME/IP.

From the SOME/IP specifications we see that there are transport protocol bindings defined for both TCP and UDP (note - not UDS here!).

So I think that it is reasonable to assume that the host address would be filled in for either TCP or UDP transport protocols.

lutzbichler, 2024-08-06:
You are mixing up SOME/IP communication between different hosts and vsomeip communication on a single host. Applications on a single host use the vsomeip protocol to communicate to each other and this protocol runs via UDS(!) or TCP. And all these additional elements of the CommonAPI client identifier (uid, gid, env, host address) only apply to communication on a single host.

srl100, 2024-08-09:
@lutzbichler Any ideas where I can ask for help on determining the client IP address in server code? One of the other COVESA repositories, perhaps?