COVESA/vsomeip #680: [BUG]: How to enable someip to communicate between docker container and non container application

shubham25091998, 2024-04-24:
### vSomeip Version

3.4.10

### Boost Version

1.71

### Environment

Ubuntu 20.04

### Describe the bug

I tried a communication between someip server application running in docker container with someip client application running directly as normal process. I have connected docker container to host network also, but no communication is happening between client and server.
Client and server able to communicate when both run in container.
So, is there any way in when docker and non docker application communicate over someip.

giovanniojan, 2024-05-02:
as a workaround, I suggest you to bind /tmp host folder to guest (see https://docs.docker.com/storage/bind-mounts/) ; in this case the communication will be made via unix domain sockets

goncaloalmeida, 2024-05-03:
@shubham25091998 this is not a defect/bug. Please use the discussion tab

closed 2024-05-03 by goncaloalmeida