COVESA/vsomeip #684: [BUG]: Eventregistration with dynamic instance ids failure

wennmbj, 2024-04-26:
### vSomeip Version

v3.4.9.1

### Boost Version

1.78

### Environment

Embedded

### Describe the bug

With the usage of dynamic instance ids, the event is not correctly registered at the routing node in routing_manager_stub. A remote client is then not able to subscribe to an event (NACK). In the routing node the error process_eventgroupentry: Received a SubscribeEventGroup entry for unknown eventgroup from ... is shown.

### Expected behaviour

Event is registered at the routing node.

### Logs and Screenshots

[image]

duartenfonseca, 2024-09-06:
@wennmbj could you add some reproduction steps for this error?

duartenfonseca, 2024-09-06:
@wennmbj closing this as this is not a feature from vsomeip

closed 2024-09-06 by duartenfonseca