COVESA/vsomeip #688: [BUG]: Build error on Ubuntu 24.04

xbroquer, 2024-05-03:
### vSomeip Version

3.4.10

### Boost Version

1.83

### Environment

Ubuntu 24.04

### Describe the bug

Hi,

There is a build failure on u24.04 (tested on master and 3.4.10): default boost version (1.83)

`[  4%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/routing/src/routing_manager_client.cpp.o
[  9%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/security/src/policy.cpp.o
In file included from /usr/include/boost/icl/concept/interval_set.hpp:14,
                 from /usr/include/boost/icl/associative_interval_container.hpp:16,
                 from /usr/include/boost/icl/interval_base_set.hpp:24,
                 from /usr/include/boost/icl/interval_set.hpp:14,
                 from /usr/include/boost/icl/interval_map.hpp:13,
                 from /home/az01958/src/vsomeip/implementation/security/src/../include/policy.hpp:15,
                 from /home/az01958/src/vsomeip/implementation/security/src/policy.cpp:10:
In function ‘typename Type::iterator boost::icl::segmental::join_under(Type&, const typename Type::value_type&, typename Type::iterator) [with Type = boost::icl::interval_set<unsigned int>]’,
    inlined from ‘boost::icl::interval_set<DomainT, Compare, Interval, Alloc>::iterator boost::icl::interval_set<DomainT, Compare, Interval, Alloc>::add_over(const interval_type&, iterator) [with DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]’ at /usr/include/boost/icl/interval_set.hpp:180:49,
    inlined from ‘boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::iterator boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::_add(const segment_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]’ at /usr/include/boost/icl/interval_base_set.hpp:513:32,
    inlined from ‘SubType& boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::add(const segment_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]’ at /usr/include/boost/icl/interval_base_set.hpp:258:13,
    inlined from ‘typename boost::enable_if<boost::icl::is_interval_set<Type>, Type>::type& boost::icl::add(Type&, const typename Type::segment_type&) [with Type = interval_set<unsigned int>]’ at /usr/include/boost/icl/concept/interval_set.hpp:71:22,
    inlined from ‘typename boost::enable_if<boost::icl::is_interval_set<Type>, Type>::type& boost::icl::add(Type&, const typename Type::element_type&) [with Type = interval_set<unsigned int>]’ at /usr/include/boost/icl/concept/interval_set.hpp:79:20,
    inlined from ‘SubType& boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::add(const element_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]’ at /usr/include/boost/icl/interval_base_set.hpp:252:24,
    inlined from ‘SubType& boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::insert(const element_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]’ at /usr/include/boost/icl/interval_base_set.hpp:289:19,
    inlined from ‘bool vsomeip_v3::policy::deserialize(const vsomeip_v3::byte_t*&, uint32_t&)’ at /home/az01958/src/vsomeip/implementation/security/src/policy.cpp:71:23:
/usr/include/boost/icl/detail/interval_set_algo.hpp:358:5: error: writing 9 bytes into a region of size 8 [-Werror=stringop-overflow=]
  358 |     const_cast<value_type&>(key_value<Type>(first_))
      |     ^~~~~~~~~~
/home/az01958/src/vsomeip/implementation/security/src/policy.cpp: In member function ‘bool vsomeip_v3::policy::deserialize(const vsomeip_v3::byte_t*&, uint32_t&)’:
/home/az01958/src/vsomeip/implementation/security/src/policy.cpp:70:37: note: at offset 40 into destination object ‘its_gid_set’ of size 48
   70 |     boost::icl::interval_set<gid_t> its_gid_set;
      |                                     ^~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/vsomeip3.dir/build.make:1224: CMakeFiles/vsomeip3.dir/implementation/security/src/policy.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /usr/include/boost/icl/concept/interval_set.hpp:14,
                 from /usr/include/boost/icl/associative_interval_container.hpp:16,
                 from /usr/include/boost/icl/interval_base_set.hpp:24,
                 from /usr/include/boost/icl/interval_set.hpp:14,
                 from /home/az01958/src/vsomeip/implementation/routing/src/../include/../../configuration/include/configuration.hpp:16,
                 from /home/az01958/src/vsomeip/implementation/routing/src/../include/routing_manager_base.hpp:28,
                 from /home/az01958/src/vsomeip/implementation/routing/src/../include/routing_manager_client.hpp:19,
                 from /home/az01958/src/vsomeip/implementation/routing/src/routing_manager_client.cpp:24:
In function ‘typename Type::iterator boost::icl::segmental::join_under(Type&, const typename Type::value_type&, typename Type::iterator) [with Type = boost::icl::interval_set<unsigned int>]’,
    inlined from ‘boost::icl::interval_set<DomainT, Compare, Interval, Alloc>::iterator boost::icl::interval_set<DomainT, Compare, Interval, Alloc>::add_over(const interval_type&, iterator) [with DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]’ at /usr/include/boost/icl/interval_set.hpp:180:49,
    inlined from ‘boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::iterator boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::_add(const segment_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]’ at /usr/include/boost/icl/interval_base_set.hpp:513:32,
    inlined from ‘SubType& boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::add(const segment_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]’ at /usr/include/boost/icl/interval_base_set.hpp:258:13,
    inlined from ‘typename boost::enable_if<boost::icl::is_interval_set<Type>, Type>::type& boost::icl::add(Type&, const typename Type::segment_type&) [with Type = interval_set<unsigned int>]’ at /usr/include/boost/icl/concept/interval_set.hpp:71:22,
    inlined from ‘typename boost::enable_if<boost::icl::is_interval_set<Type>, Type>::type& boost::icl::add(Type&, const typename Type::element_type&) [with Type = interval_set<unsigned int>]’ at /usr/include/boost/icl/concept/interval_set.hpp:79:20,
    inlined from ‘SubType& boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::add(const element_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]’ at /usr/include/boost/icl/interval_base_set.hpp:252:24,
    inlined from ‘SubType& boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::insert(const element_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]’ at /usr/include/boost/icl/interval_base_set.hpp:289:19,
    inlined from ‘void vsomeip_v3::routing_manager_client::on_update_security_credentials(const vsomeip_v3::protocol::update_security_credentials_command&)’ at /home/az01958/src/vsomeip/implementation/routing/src/routing_manager_client.cpp:2839:27:
/usr/include/boost/icl/detail/interval_set_algo.hpp:358:5: error: writing 9 bytes into a region of size 8 [-Werror=stringop-overflow=]
[... cut ...]
/home/az01958/src/vsomeip/implementation/routing/src/routing_manager_client.cpp:2835:44: note: at offset 40 into destination object ‘its_gid_set’ of size 48
 2835 |         boost::icl::interval_set<uint32_t> its_gid_set;
      |                                            ^~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/vsomeip3.dir/build.make:1126: CMakeFiles/vsomeip3.dir/implementation/routing/src/routing_manager_client.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:716: CMakeFiles/vsomeip3.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
`

fcmonteiro, 2024-05-03:
Probably the same as #527

xbroquer, 2024-05-03:
calling clear() fixes the compilation be I don't understand why

        boost::icl::interval_set<uint32_t> its_gid_set;
        its_gid_set.clear();

phongt, 2024-05-04:
Building with `-DCMAKE_BUILD_TYPE=Debug` can help.
[image]

duartenfonseca, 2024-11-19:
@xbroquer can you try with 3.5.1?