COVESA/vsomeip #690: [BUG]: vsomeip slow to restart with lots of EventGroup

joeyoravec, 2024-05-04:
### vSomeip Version

v3.4.10

### Boost Version

1.82

### Environment

Android and QNX

### Describe the bug

My automotive system has `*.fidl` with ~3500 attributes, one per CAN signal. My `*.fdepl` maps each attribute into a unique EventGroup.

Especially when resuming from suspend-to-ram it's possible that UDP SOMEIP-SD will be operational but TCP socket will be broken. This leads to tce `restart()` but during this time any Subscribe will receive SubscribeNack in response:
```
4191	105.781314	10.6.0.10	10.6.0.3	SOME/IP-SD	1408	SOME/IP Service Discovery Protocol [Subscribe]
4192	105.790868	10.6.0.3	10.6.0.10	SOME/IP-SD	1396	SOME/IP Service Discovery Protocol [SubscribeNack]
4193	105.792094	10.6.0.10	10.6.0.3	SOME/IP-SD	1410	SOME/IP Service Discovery Protocol [Subscribe]
4194	105.801525	10.6.0.10	10.6.0.3	SOME/IP-SD	1410	SOME/IP Service Discovery Protocol [Subscribe]
4195	105.802118	10.6.0.3	10.6.0.10	SOME/IP-SD	1398	SOME/IP Service Discovery Protocol [SubscribeNack]
4196	105.819610	10.6.0.3	10.6.0.10	SOME/IP-SD	1398	SOME/IP Service Discovery Protocol [SubscribeNack]
```
as the number of EventGroup scales to a large number, this become catastrophic to performance.

In `service_discovery_impl::handle_eventgroup_subscription_nack()` each EventGroup calls `restart()`:
https://github.com/COVESA/vsomeip/blob/cf497232adf84f55947f7a24e1b64e04b49f1f38/implementation/service_discovery/src/service_discovery_impl.cpp#L2517-L2521

and in `tcp_client_endpoint_impl::restart()` while `::CONNECTING` the code will "early terminate" from maximum 5 restarts:
https://github.com/COVESA/vsomeip/blob/cf497232adf84f55947f7a24e1b64e04b49f1f38/implementation/endpoints/src/tcp_client_endpoint_impl.cpp#L77-L85

thereafter the code will fall through, calling `shutdown_and_close_socket_unlocked()` and perform the full restart even while a connection is in progress.

As the system continues processing 1000s of SubscribeNack this will be a tight loop of 100% cpu load and multiple seconds to plow-through the workload. This can easily exceed a 2s ServiceDiscovery interval and cascade to further problems.

### Reproduction Steps

My reproduction was:
- start with fully-established communication between tse and tce
- tce enters suspend-to-ram with TCP socket established
- allow tse to continue running, exceed TCP keepalive timeout, and close the TCP socket
- tce resumes from suspend-to-ram thinking TCP socket is still established, then discovers it to be closed

but any use-case where tse closes the TCP socket but UDP is functional should be sufficient.

### Expected behaviour

Performance should be better.

joeyoravec, 2024-05-04:
We came up with 3 possible solutions;
1. eliminate the tce `restart()` call from `service_discovery_impl::handle_eventgroup_subscription_nack()`. It's not clear why this is required or how it would help
2. modify tce `restart()` to "early terminate" better, perhaps an unlimited number of times within the 5 second timeout
3. ensure that SOMEIP-SD gets inhibited around any event like suspend-to-ram where network communication will be lost. Try to prevent Subscribe until the TCP socket gets re-established

Interested in feedback on what would be most effective

FishPickBeans, 2025-05-27:
I encountered the same problem where the client detected a server restart and TCP connected first, but received a NACK and immediately called Restart(). TCP disconnected first and then connected again. Is there any way to keep it going?

1. tcpdump log(see attachment)

2. ----my log----------（“VSIP” is vsomeip3.4.10 log）:
66343 2025/05/26 17:23:28.337754 4602.6149 235 ECU1 SPND CLIE 2366 log info verbose 2 {321} SomeipProxyCreate enter
66344 2025/05/26 17:23:28.337798 4602.6155 65 ECU1 SPND VSIP 2366 log info verbose 1 REGISTER EVENT(1f02): [1005.0001.8001:is_provider=false]
66345 2025/05/26 17:23:28.337841 4602.6157 66 ECU1 SPND VSIP 2366 log info verbose 1 REQUEST(1f02): [1005.0001:1.4294967295]
66346 2025/05/26 17:23:28.338009 4602.6249 250 ECU1 SPND CAPI 2366 log info verbose 1 waitUntilConnected ...
66347 2025/05/26 17:23:28.338052 4602.6251 251 ECU1 SPND CAPI 2366 log info verbose 1 waitUntilConnected() no need wait stat
66348 2025/05/26 17:23:28.338221 4602.6316 252 ECU1 SPND CAPI 2366 log info verbose 1 availabilityCondition_ wait ...
66359 2025/05/26 17:23:29.068976 4603.4045 77 ECU1 SPND VSIP 2366 log warn verbose 1 Didn't receive a multicast SD message for 1100ms.
66373 2025/05/26 17:23:30.169648 4604.5051 91 ECU1 SPND VSIP 2366 log warn verbose 1 Didn't receive a multicast SD message for 1100ms.
66397 2025/05/26 17:23:31.272181 4605.6059 115 ECU1 SPND VSIP 2366 log warn verbose 1 Didn't receive a multicast SD message for 1100ms.
66404 2025/05/26 17:23:31.428065 4605.7635 122 ECU1 SPND VSIP 2366 log info verbose 1 Reboot detected: IP=172.16.59.3
66405 2025/05/26 17:23:31.428147 4605.7638 123 ECU1 SPND VSIP 2366 log info verbose 1 configuration_impl:find_specific_port #1: service: 1005 instance: 1 reliable: 1 return specific port: 30501
66406 2025/05/26 17:23:31.428192 4605.7640 124 ECU1 SPND VSIP 2366 log info verbose 1 endpoint_manager_impl::create_remote_client: 172.16.59.3:30500 reliable: 1 using local port: 30501
66407 2025/05/26 17:23:31.428235 4605.7641 125 ECU1 SPND VSIP 2366 log warn verbose 1 start_connecting_timerclient_endpoint_impl<Protocol>
66408 2025/05/26 17:23:31.428277 4605.7643 126 ECU1 SPND VSIP 2366 log warn verbose 1 tcp_client_endpoint::connect socket_->open err=Success
66409 2025/05/26 17:23:31.428319 4605.7644 127 ECU1 SPND VSIP 2366 log warn verbose 1 tcp_client_endpoint::connect bind()
66410 2025/05/26 17:23:31.435464 4605.7699 128 ECU1 SPND VSIP 2366 log warn verbose 1 cancel_and_connect_cbk connect_cbk
66411 2025/05/26 17:23:31.435543 4605.7725 129 ECU1 SPND VSIP 2366 log error verbose 1 connect_cbk: Success
66412 2025/05/26 17:23:31.449819 4605.7778 253 ECU1 SPND CAPI 2366 log info verbose 1 isAvailableBlocking() end
66413 2025/05/26 17:23:31.449877 4605.7779 236 ECU1 SPND CLIE 2366 log info verbose 2 {270} SomeipSubscribeEvent sub enter
66414 2025/05/26 17:23:31.449908 4605.7781 130 ECU1 SPND VSIP 2366 log info verbose 1 SUBSCRIBE(1f02): [1005.0001.0003:8001:1]
66415 2025/05/26 17:23:31.449938 4605.7783 237 ECU1 SPND CLIE 2366 log info verbose 2 {329} SomeipProxyCreate create success
66416 2025/05/26 17:23:31.449968 4605.7783 238 ECU1 SPND CLIE 2366 log info verbose 2 {358} SomeipLinkStatusNotify old state=0, new state=1
66417 2025/05/26 17:23:31.449998 4605.7783 239 ECU1 SPND CLIE 2366 log info verbose 2 {364} SomeipLinkStatusNotify notify state=1
66418 2025/05/26 17:23:31.452509 4605.7882 254 ECU1 SPND CAPI 2366 log info verbose 1 waitUntilConnected ...
66419 2025/05/26 17:23:31.452543 4605.7883 255 ECU1 SPND CAPI 2366 log info verbose 1 waitUntilConnected() no need wait stat
66420 2025/05/26 17:23:31.452573 4605.7884 0 ECU1 SPND CAPI 2366 log info verbose 1 availabilityCondition_ wait ...
66421 2025/05/26 17:23:31.464839 4605.8004 131 ECU1 SPND VSIP 2366 log warn verbose 1 handle_eventgroup_subscription_nack restart
66422 2025/05/26 17:23:31.473589 4605.8063 132 ECU1 SPND VSIP 2366 log warn verbose 1 shutdown_and_close_socket_unlocked shutdown and close
66423 2025/05/26 17:23:31.473625 4605.8063 133 ECU1 SPND VSIP 2366 log warn verbose 1 shutdown_and_close_socket_unlocked reset()
66424 2025/05/26 17:23:31.473655 4605.8064 134 ECU1 SPND VSIP 2366 log warn verbose 1 tce::restart: local: 172.16.59.10:30501 remote: 172.16.59.3:30500
66425 2025/05/26 17:23:31.473685 4605.8064 135 ECU1 SPND VSIP 2366 log warn verbose 1 start_connect_timer client_endpoint_impl
66426 2025/05/26 17:23:31.473716 4605.8066 136 ECU1 SPND VSIP 2366 log warn verbose 1 tcp_client_endpoint receive_cbk: End of file(2) local: remote: 172.16.59.3:30500
66427 2025/05/26 17:23:31.473746 4605.8066 137 ECU1 SPND VSIP 2366 log warn verbose 1 tcp_client_endpoint receive_cbk already restarting172.16.59.3:30500
66428 2025/05/26 17:23:31.498722 4605.8330 240 ECU1 SPND CLIE 2366 log info verbose 2 {69} push, rcv src:34, mid:2601,len=49
66429 2025/05/26 17:23:31.498754 4605.8332 241 ECU1 SPND CLIE 2366 log info verbose 2 {92} s data=0x00,0x00,0x00,0x00,0x05,0x10,0x00,0x00,0x01,0x00,
66430 2025/05/26 17:23:31.573702 4605.9095 138 ECU1 SPND VSIP 2366 log warn verbose 1 start_connecting_timerclient_endpoint_impl<Protocol>
66431 2025/05/26 17:23:31.573880 4605.9099 139 ECU1 SPND VSIP 2366 log warn verbose 1 tcp_client_endpoint::connect socket_->open err=Success
66432 2025/05/26 17:23:31.574748 4605.9116 140 ECU1 SPND VSIP 2366 log warn verbose 1 tcp_client_endpoint::connect bind()
66433 2025/05/26 17:23:31.576315 4605.9131 141 ECU1 SPND VSIP 2366 log warn verbose 1 cancel_and_connect_cbk connect_cbk
[... cut ...]
66460 2025/05/26 17:23:31.898180 4606.2342 162 ECU1 SPND VSIP 2366 log error verbose 1 connect_cbk: Cannot assign requested address
66461 2025/05/26 17:23:31.898352 4606.2344 163 ECU1 SPND VSIP 2366 log warn verbose 1 shutdown_and_close_socket_unlocked shutdown and close
66462 2025/05/26 17:23:31.899066 4606.2360 164 ECU1 SPND VSIP 2366 log warn verbose 1 shutdown_and_close_socket_unlocked reset()
66463 2025/05/26 17:23:31.899870 4606.2370 165 ECU1 SPND VSIP 2366 log warn verbose 1 service_endpoint_disconnected: lost connection to remote service: [1005.0001]

[image]