COVESA/vsomeip #697: [BUG]: SomeIpEventGroups reported as deprecated, but replacement with SomeIpNotifierEventGroups does not work

srl100, 2024-05-08:
### vSomeip Version

v3.4.10

### Boost Version

1.81

### Environment

Embedded Linux

### Describe the bug

I have a .fidl and .fdepl file (attached) which define attributes that I wish to support notifications on.

TestAttribute1 uses SomeIpEventGroups to declare the group it is in and appears to work OK, but the deployment checker complains that this is deprecated.
TestAttribute2 uses SomeIpNotifierEventGroups instead, but the deployment checker complains that it has no valid SomeIpNotifierEventGroups specified:

```
<path>/commonapi_core_generator/commonapi-core-generator-linux-x86_64 -sk ../fidl/Test.fidl
Executing CommonAPI Core Code Generation...

Skeleton code will be created
Using Franca Version 0.13.1.201807231814
and CommonAPI Version 3.2.14.v202310241605
log4j:WARN No appenders could be found for logger (org.eclipse.xtext.parser.antlr.AbstractInternalAntlrParser).
log4j:WARN Please initialize the log4j system properly.
log4j:WARN See http://logging.apache.org/log4j/1.2/faq.html#noconfig for more info.
Generating code for <path>/build/../fidl/Test.fidl

<path>/commonapi_someip_generator/commonapi-someip-generator-linux-x86_64 ../fidl/Test.fdepl 
Executing CommonAPI SomeIP Code Generation...

Using Franca Version 0.13.1.201807231814
log4j:WARN No appenders could be found for logger (org.eclipse.xtext.parser.antlr.AbstractInternalAntlrParser).
log4j:WARN Please initialize the log4j system properly.
log4j:WARN See http://logging.apache.org/log4j/1.2/faq.html#noconfig for more info.
Warning: SomeIP validation: Deprecated: Attribute "TestAttribute1" is using 'SomeIpEventGroups', use 'SomeIpNotifierEventGroups'. (file:<path>/build/../fidl/Test.fdepl line : 8 column : 15)
Warning: SomeIP validation: Attribute "TestAttribute2" has no valid 'SomeIpNotifierEventGroups' specified. (file:<path>/build/../fidl/Test.fdepl line : 15 column : 15)
SomeIP validation: Validation of deployment finished with: 0 errors, 2 warnings.
Generating code for <path>/build/../fidl/Test.fdepl
```

Use of the deprecated 'SomeIpEventGroups' version at least allows my code to build.

If I use the new, improved (!), 'SomeIpNotifierEventGroups' I get subsequent failures in compilation, with errors mentioning ‘UNDEFINED_EVENTGROUP_ID’.

### Logs and Screenshots

[Test_fdepl.txt](https://github.com/COVESA/vsomeip/files/15245641/Test_fdepl.txt)
[Test_fidl.txt](https://github.com/COVESA/vsomeip/files/15245643/Test_fidl.txt)

RicardoCarreira, 2024-05-13:
Hello @srl100,

Regarding the Test_fdepl.fdepl file, in the first line you define the deployment specs 
> "CommonAPI-SOMEIP_deployment_spec.fdepl" 

and it should be defined the deployment specs 

> "CommonAPI-**4**-SOMEIP_deployment_spec.fdepl"

you can find both deployment specifications [here](https://cc-github.bmwgroup.net/networking/commonapi-someip-generator/tree/maintain/3.2/org.genivi.commonapi.someip/deployment) and should always use the latest version (CommonAPI-4).

The corrected version of the file should look like this:
```
import "platform:/plugin/org.genivi.commonapi.someip/deployment/CommonAPI-4-SOMEIP_deployment_spec.fdepl"
import "Test.fidl"

define org.genivi.commonapi.someip.deployment for interface commonapi.examples.Test {
    SomeIpServiceID = 0x1234

    attribute TestAttribute1 {
	SomeIpGetterID = 1001
	SomeIpNotifierID = 32769

	SomeIpNotifierEventGroups = { 2001 }
    }

    attribute TestAttribute2 {
	SomeIpGetterID = 1002
	SomeIpNotifierID = 32770

	SomeIpNotifierEventGroups = { 2002 }
    }
}

define org.genivi.commonapi.someip.deployment for provider as Service {
    instance commonapi.examples.Test {
	InstanceId = "commonapi.examples.Test"

	SomeIpInstanceID = 254

	SomeIpUnicastAddress = "127.0.0.1"
	SomeIpUnreliableUnicastPort = 5000
    }
}
```

Please note that the keyword _'SomeIpEventGroups'_ was replaced with _'SomeIpNotifierEventGroups'_ as it is deprecated and should no longer be used, it is also not recommended to use deprecated keywords in conjunction with it's replacement.

> If I use the new, improved (!), 'SomeIpNotifierEventGroups' I get subsequent failures in compilation, with errors mentioning ‘UNDEFINED_EVENTGROUP_ID’.

Since the deployment specification in use on your files doesn't support the _'SomeIpNotifierEventGroups'_ keyword, the generated code show up with the _‘UNDEFINED_EVENTGROUP_ID’_ as it was unable to determine the group id defined.

srl100, 2024-05-13:
Confimed as not a problem if CommonAPI-4-SOMEIP_deployment_spec.fdepl is used.