COVESA/vsomeip #709: how to get vsomeip header when I do e2e include vsomeip header?

fengmao31, 2024-05-24:
### vSomeip Version

V3.4.10

### Boost Version

1.71

### Environment

Ubuntu20.04

### Describe the bug

When I do e2e include vsomeip header, our company request the crc32 profile04 need include someip header. but in my past experice to use vsomeip, I donnot sure it has the interface to get the information and  I find the session id is managed by vsomeip library. If I cannot get the header, I set the session id by myself. Can I impleiment the e2e function with someip header?

fengmao31, 2024-09-12:
The vsomeip e2e functionality works, and I made a mistake in understanding that part of the code. My apologies.

GHARBIAchref, 2024-09-21:
Hello @fengmao31  @duartenfonseca  
 I want to use end-to-end protection in my application. Is using end-to-end protection done solely through configuration in .json configuration files on both the client and server sides of the services, or does it also require code implementation within my algorithm? And how can I verify if the E2E protection profile 4 is working correctly in Wireshark? 
 thanks