COVESA/vsomeip #711: During compilation, it is prompted that the size of the ucred data structure cannot be calculated

yfan0000004, 2024-05-28:
### vSomeip Version

v3.4.10

### Boost Version

1.78

### Environment

QNX7.1

### Describe the bug

During compilation, it is prompted that the size of the ucred data structure cannot be calculated.How to solve this problem?

### Logs and Screenshots

COVESA/vsomeip/implementation/endpoints/src/../include/local_server_endpoint_impl_receive_op.hpp:75:71: error: invalid application of 'sizeof' to incomplete type 'vsomeip_v3::local_endpoint_receive_op::receive_cb(std::__1::shared_ptr<vsomeip_v3::local_endpoint_receive_op::storage>)::<lambda(boost::system::error_code)>::ucred'                  control_un.cmh.cmsg_len = CMSG_LEN(sizeof(struct ucred));

yfan0000004, 2024-05-28:
There are also the following issues that exist simultaneously：

COVESA/vsomeip/implementation/endpoints/src/../include/local_server_endpoint_impl_receive_op.hpp:77:44: error: 'SCM_CREDENTIALS' was not declared in this scope
                 control_un.cmh.cmsg_type = SCM_CREDENTIALS;

Lynn-1997, 2024-08-30:
The QNX environment is modified as follows:
control_un.cmh.cmsg_len = CMSG_LEN(sizeof(struct sockcred));
control_un.cmh.cmsg_level = SOL_SOCKET;
control_un.cmh.cmsg_type = SCM_CREDS;

duartenfonseca, 2024-10-24:
@Lynn-1997 is that the fix for the bug that was described by @yfan0000004 ?

Lynn-1997, 2024-11-14:
Yes, I have the same problem on QNX when I compile with the same environment. Use sockcred structure on QNX, use ucred structure on other systems. Also, the problem has been solved in version vsomeip3.5.1. @duartenfonseca

closed 2024-11-14 by duartenfonseca