COVESA/vsomeip #715: [BUG]: Facing Issue while compiling vsomeip in gcc version 12.2.0 (Debian 12.2.0-14)

itsavi7879, 2024-05-31:
### vSomeip Version

#https://github.com/COVESA/vsomeip/commit/6c0e9db200fbcfd37879c4b2ff0c8523a29d8eb5

### Boost Version

1.74

### Environment

GCC version 12.2.0 (debian 12.2.0-14)

### Describe the bug

When i am trying to compile the vsomeip library in gcc 9.4.0 (ubuntu 9.4.0-1ubuntu1~20.04.2) i am able to compile vsomeip with below mentioned commands:-
**cd vsomeip
rm -rf build
mkdir build
cd build
cmake ..
make -j $(nproc)
make install**

it is compiling without throwing any error.
But when i try to run the same set of command in gcc version 12.2.0(Debian 12.2.0-14) , i am getting the error for which i have attached the screenshots.
ERROR: 
**inlined from 'bool vsomeip_v3::policy::deserialize(const vsomeip_v3::byte_t*&, uint32_t&)' at /tmp/vsomeip/implementation/security/src/policy.cpp:71:23:
/usr/include/boost/icl/detail/interval_set_algo.hpp:358:5: error: writing 9 bytes into a region of size 8 [-Werror=stringop-overflow=]
  358 |     const_cast<value_type&>(key_value<Type>(first_))**

### Reproduction Steps

**cd vsomeip
rm -rf build
mkdir build
cd build
cmake ..
make -j $(nproc)
make install**

### Expected behaviour

It should build successfully.
[image]

### Logs and Screenshots

[image]
[image]

phongt, 2024-06-14:
seem like same problem https://github.com/COVESA/vsomeip/issues/688

duartenfonseca, 2024-10-28:
@itsavi7879 can you retry with a more recent version? thanks

duartenfonseca, 2024-12-17:
no response in a month

closed 2024-12-17 by duartenfonseca