COVESA/vsomeip #736: [BUG]: possible double free on app_->stop()

mogar, 2024-07-09:
### vSomeip Version

v3.4.10

### Boost Version

1.68

### Environment

Yocto 3.1 on ARM Cortex M78AE

### Describe the bug

I have a double free in my someip application. It occurs on shutdown, and seems to be from within vsomeip itself.

This may be related to the fact that I'm connecting to multiple (4) identical someip devices. Each device has a distinct instance ID, but offers the same services/events. My vsomeip host subscribes to multiple events from each device. Each device subscribes to one event offered by my service.

### Reproduction Steps

I haven't managed to create minimal reproduction steps yet.

In my application, shutting down the app by calling `app_->stop()` will usually (but not always) lead to a double free. 

### Expected behaviour

No double frees on shutdown.

### Logs and Screenshots

coredump:

```
#0  0x0000ffffb41cb81c in raise () from /home/mredfield/.cache/bazel/_bazel_mredfield/9b432d07059908728084e68c8bfb39a3/external/yocto_aarch64_gcc_sysroot_x86_64/lib/libc.so.6
#1  0x0000ffffb41b8dd4 in abort () from /home/mredfield/.cache/bazel/_bazel_mredfield/9b432d07059908728084e68c8bfb39a3/external/yocto_aarch64_gcc_sysroot_x86_64/lib/libc.so.6
#2  0x0000ffffb4203d34 in ?? () from /home/mredfield/.cache/bazel/_bazel_mredfield/9b432d07059908728084e68c8bfb39a3/external/yocto_aarch64_gcc_sysroot_x86_64/lib/libc.so.6
#3  0x0000ffffb420b2bc in ?? () from /home/mredfield/.cache/bazel/_bazel_mredfield/9b432d07059908728084e68c8bfb39a3/external/yocto_aarch64_gcc_sysroot_x86_64/lib/libc.so.6
#4  0x0000ffffb420cbdc in ?? () from /home/mredfield/.cache/bazel/_bazel_mredfield/9b432d07059908728084e68c8bfb39a3/external/yocto_aarch64_gcc_sysroot_x86_64/lib/libc.so.6
#5  0x0000aaaab5737fac in std::_Rb_tree<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool>, std::pair<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool> const, std::shared_ptr<vsomeip_v3::endpoint_definition> >, std::_Select1st<std::pair<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool> const, std::shared_ptr<vsomeip_v3::endpoint_definition> > >, std::less<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool> >, std::allocator<std::pair<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool> const, std::shared_ptr<vsomeip_v3::endpoint_definition> > > >::_M_erase(std::_Rb_tree_node<std::pair<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool> const, std::shared_ptr<vsomeip_v3::endpoint_definition> > >*) ()
#6  0x0000aaaab5737f88 in std::_Rb_tree<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool>, std::pair<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool> const, std::shared_ptr<vsomeip_v3::endpoint_definition> >, std::_Select1st<std::pair<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool> const, std::shared_ptr<vsomeip_v3::endpoint_definition> > >, std::less<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool> >, std::allocator<std::pair<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool> const, std::shared_ptr<vsomeip_v3::endpoint_definition> > > >::_M_erase(std::_Rb_tree_node<std::pair<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool> const, std::shared_ptr<vsomeip_v3::endpoint_definition> > >*) ()
#7  0x0000aaaab5737f88 in std::_Rb_tree<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool>, std::pair<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool> const, std::shared_ptr<vsomeip_v3::endpoint_definition> >, std::_Select1st<std::pair<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool> const, std::shared_ptr<vsomeip_v3::endpoint_definition> > >, std::less<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool> >, std::allocator<std::pair<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool> const, std::shared_ptr<vsomeip_v3::endpoint_definition> > > >::_M_erase(std::_Rb_tree_node<std::pair<std::tuple<unsigned short, unsigned short, boost::asio::ip::address, unsigned short, bool> const, std::shared_ptr<vsomeip_v3::endpoint_definition> > >*) ()
#8  0x0000ffffb41cdc48 in ?? () from /home/mredfield/.cache/bazel/_bazel_mredfield/9b432d07059908728084e68c8bfb39a3/external/yocto_aarch64_gcc_sysroot_x86_64/lib/libc.so.6
#9  0x0000ffffb41cdddc in exit () from /home/mredfield/.cache/bazel/_bazel_mredfield/9b432d07059908728084e68c8bfb39a3/external/yocto_aarch64_gcc_sysroot_x86_64/lib/libc.so.6
#10 0x0000ffffb41b9128 in __libc_start_main ()
   from /home/mredfield/.cache/bazel/_bazel_mredfield/9b432d07059908728084e68c8bfb39a3/external/yocto_aarch64_gcc_sysroot_x86_64/lib/libc.so.6
#11 0x0000aaaab565e518 in _start () at ../sysdeps/aarch64/start.S:94
```

duartenfonseca, 2024-09-24:
@mogar were you able to reproduce it, or give some indications on how we could reproduce it on our side?

duartenfonseca, 2024-10-07:
we have no reproduction steps, and no way of testing on this environment

closed 2024-10-07 by duartenfonseca