COVESA/vsomeip #741: [BUG]: Facing an issue in while communicating between host and virtual machine

DharaniAllada, 2024-07-13:
### vSomeip Version

v3.4.10

### Boost Version

1.71

### Environment

Ubuntu 20.04.6

### Describe the bug

The Problem statement i have to do is " Automate client server communication between the host PC and QEMU using SOMEIP. Validate the communication using wireshark."

So from the starting the steps i followed are:

sudo apt install cmake git gcc g++ libboost-system-dev libboost-thread-dev libboost-log-dev libboost-chrono-dev libboost-date-time-dev libboost-atomic-dev libboost-program-options-dev
git clone https://github.com/GENIVI/vsomeip.git
cd vsomeip
mkdir build
cd build
cmake ..
make
sudo make install

In the host machine i wrote the client code and vsomeip.json under the vsomeip/examples

**client-sample.cpp:** : 

#include <vsomeip/vsomeip.hpp>
#include <iostream>
#include <vector>

#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678
#define SAMPLE_METHOD_ID 0x0421

class Client {
	public:
	    	Client() : app_(vsomeip::runtime::get()->create_application("Client")) {}

    		bool init()
		{
        		if(!app_->init()) {
				std::cerr << "Failed to initialize application" << std::endl;
        	    		return false;
	        	}

	        	app_->register_message_handler(
        	    		SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_METHOD_ID,
            	  	  	std::bind(&Client::on_response, this, std::placeholders::_1));

	        	return true;
    		}

    		void start()
		{
			std::cout << "Client application started and initialized." << std::endl;
        		app_->request_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID);
        		std::shared_ptr<vsomeip::message> request = vsomeip::runtime::get()->create_request();
			std::cout << "Sending request to service" << std::endl;
        		app_->send(request);
        		app_->start();
    		}

    		void stop()
		{
        		app_->stop();
[... 210 lines cut ...]
2024-07-13 12:34:59.112443 [info] create_local_server: Listening @ /tmp/vsomeip-5678

Sending request to service
2024-07-13 12:34:59.112758 [error] Routing info for remote service could not be found! (5678): [0000.0000.0000] 0001

2024-07-13 12:34:59.112855 [info] Starting vsomeip application "Client" (5678) using 2 threads I/O nice 255
2024-07-13 12:34:59.113666 [info] Client [5678] routes unicast:192.168.122.1, netmask:255.255.255.0
2024-07-13 12:34:59.113787 [info] main dispatch thread id from application: 5678 (Client) is: 7fdc550a4700 TID: 5498
2024-07-13 12:34:59.114274 [info] shutdown thread id from application: 5678 (Client) is: 7fdc548a3700 TID: 5499
2024-07-13 12:34:59.114675 [info] Watchdog is disabled!
2024-07-13 12:34:59.114946 [info] io thread id from application: 5678 (Client) is: 7fdc559ad000 TID: 5496
2024-07-13 12:34:59.114969 [info] io thread id from application: 5678 (Client) is: 7fdc4f7fe700 TID: 5501
2024-07-13 12:34:59.119681 [info] vSomeIP 3.4.10 | (default)
2024-07-13 12:34:59.120100 [info] Network interface "virbr0" state changed: up
2024-07-13 12:35:09.122365 [info] vSomeIP 3.4.10 | (default)
2024-07-13 12:35:19.124010 [info] vSomeIP 3.4.10 | (default)
2024-07-13 12:35:29.126866 [info] vSomeIP 3.4.10 | (default)
2024-07-13 12:35:39.127914 [info] vSomeIP 3.4.10 | (default)

Here in the log file of host it is showing like:

Sending request to service
2024-07-13 12:34:59.112758 [error] Routing info for remote service could not be found! (5678): [0000.0000.0000] 0001

i think there might some error there in finding the service can any one please help me with this

lutzbichler, 2024-08-09:
Do your ever call to offer_service to actually offer your service to the client? From the log it seems you do not.

duartenfonseca, 2024-09-11:
@DharaniAllada is this still an issue from your side?

closed 2024-10-22 by duartenfonseca