COVESA/vsomeip #745: Client not receiving the Multiple response from the server for sing request.

ajiththangavel, 2024-07-29:
### vSomeip Version

v3.4.0

### Boost Version

1.85

### Environment

ubuntu

### Describe the bug

i am running the Server and client from the two different ubuntu machines. when i run it in the single same machine the client recieves all the response, but when it is in the different devices the client receives the first response only.
can somebody help me on this.
[client.json](https://github.com/user-attachments/files/16409741/client.json)

client.cpp:
#include <iomanip>
#include <iostream>
#include <sstream>
#include <condition_variable>
#include <thread>
#include <vsomeip/vsomeip.hpp>

#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678
#define SAMPLE_METHOD_ID 0x0421

std::shared_ptr< vsomeip::application > app;
std::mutex mutex;
std::condition_variable condition;

void run() {
    std::unique_lock<std::mutex> its_lock(mutex);
    condition.wait(its_lock);

    std::shared_ptr<vsomeip::message> request;
    request = vsomeip::runtime::get()->create_request();
    request->set_service(SAMPLE_SERVICE_ID);
    request->set_instance(SAMPLE_INSTANCE_ID);
    request->set_method(SAMPLE_METHOD_ID);

    std::shared_ptr<vsomeip::payload> its_payload = vsomeip::runtime::get()->create_payload();
    std::string message = "Hello";
    std::vector<vsomeip::byte_t> its_payload_data(message.begin(), message.end());
    its_payload->set_data(its_payload_data);
    request->set_payload(its_payload);

    std::cout << "CLIENT: Sending request with payload: " << message << std::endl;

    app->send(request);
}

void on_message(const std::shared_ptr<vsomeip::message> &_response) {
    std::shared_ptr<vsomeip::payload> its_payload = _response->get_payload();
    vsomeip::length_t l = its_payload->get_length();

    // Get payload as string
    std::string received_message(reinterpret_cast<const char*>(its_payload->get_data()), l);

    std::cout << "CLIENT: Received message with Client/Session ["
        << std::setw(4) << std::setfill('0') << std::hex << _response->get_client() << "/"
        << std::setw(4) << std::setfill('0') << std::hex << _response->get_session() << "] "
        << received_message << std::endl;
}

void on_availability(vsomeip::service_t _service, vsomeip::instance_t _instance, bool _is_available) {
    std::cout << "CLIENT: Service ["
              << std::setw(4) << std::setfill('0') << std::hex << _service << "." << _instance
              << "] is " << (_is_available ? "available." : "NOT available.") << std::endl;
    if (_is_available) {
        condition.notify_one();
    }
}

int main() {
    app = vsomeip::runtime::get()->create_application("Hello");
    app->init();
    app->register_availability_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, on_availability);
    app->request_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID);
    app->register_message_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_METHOD_ID, on_message);

    std::thread sender(run);
    app->start();
     sender.join();
}

server.cpp:

#include <iomanip>
#include <iostream>
#include <sstream>
#include <thread>
#include <chrono>

#include <vsomeip/vsomeip.hpp>

#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678
#define SAMPLE_METHOD_ID 0x0421

std::shared_ptr<vsomeip::application> app;

void on_message(const std::shared_ptr<vsomeip::message> &_request) {

    std::shared_ptr<vsomeip::payload> its_payload = _request->get_payload();
    vsomeip::length_t l = its_payload->get_length();

    // Get payload as string
    std::string received_message(reinterpret_cast<const char*>(its_payload->get_data()), l);

    std::cout << "SERVICE: Received message with Client/Session ["
        << std::setw(4) << std::setfill('0') << std::hex << _request->get_client() << "/"
        << std::setw(4) << std::setfill('0') << std::hex << _request->get_session() << "] "
        << received_message << std::endl;

    // Create multiple responses
    for (int i = 0; i < 5; ++i) { // Example: Send 5 responses
        std::shared_ptr<vsomeip::message> its_response = vsomeip::runtime::get()->create_response(_request);
        its_payload = vsomeip::runtime::get()->create_payload();
        
        std::string response_message = "Response " + std::to_string(i);
        std::vector<vsomeip::byte_t> its_payload_data(response_message.begin(), response_message.end());
        its_payload->set_data(its_payload_data);
        
        its_response->set_payload(its_payload);
        app->send(its_response);

        std::this_thread::sleep_for(std::chrono::milliseconds(100)); // Optional: Add delay between responses
    }
}

int main() {

   app = vsomeip::runtime::get()->create_application("World");
   app->init();
   app->register_message_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_METHOD_ID, on_message);
   app->offer_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID);
   app->start();
}

k08eYLx, 2024-08-01:
Basically, this "one request with multiple responses" usecase is just not supported. It works "accidently"(not designed to) while client and server are in the single same machine. Because, when inter-host communication, the server needs to record the request client/session ids for later response, while the server application response, the server has to find the target by client/session then erase them. As intra-host communication, the server and the client request/response directly through unix domain socket ipc, the server proxy already knows where the client is, it doesn't need to record/erase the client/session ids, so it works.
reference to implementation/endpoints/src/udp_server_endpoint_impl.cpp and server_endpoint_impl.cpp
[image]
[image]

ajiththangavel, 2024-08-12:
> Basically, I think, request/response are one on one, this "one request with multiple responses" usecase is just not supported. It works "accidently"(not designed to) while client and server are in the single same machine. Because, when inter-host communication, the server needs to record the request client/session ids for later response, while the server application response, the server has to find the target by client/session then erase them. As intra-host communication, the server and the client request/response directly through unix domain socket ipc, the server proxy already knows where the client is, it doesn't need to record/erase the client/session ids, so it works. reference to implementation/endpoints/src/udp_server_endpoint_impl.cpp and server_endpoint_impl.cpp <img alt="image" width="539" src="https://private-user-images.githubusercontent.com/3823635/354300542-7a9f88dc-4653-4cf4-a91b-8e3eb3673ce1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM0NDgyNjksIm5iZiI6MTcyMzQ0Nzk2OSwicGF0aCI6Ii8zODIzNjM1LzM1NDMwMDU0Mi03YTlmODhkYy00NjUzLTRjZjQtYTkxYi04ZTNlYjM2NzNjZTEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MTJUMDczMjQ5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDM1N2U5NmQwZTg3MDcwZDBhMGQzMTJmZWE4ODJkOGY1NjY5ZmI3MTJjMWM4MDAxODE1ZGViNTNkYzZmOGJjZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.K9SF-5MdkSNrpb_9pvxIi8E1iaBIfHqSWIBbd8tTSRU"> <img alt="image" width="526" src="https://private-user-images.githubusercontent.com/3823635/354301357-be3b37d6-a120-4b71-a832-a7abe72920b9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM0NDgyNjksIm5iZiI6MTcyMzQ0Nzk2OSwicGF0aCI6Ii8zODIzNjM1LzM1NDMwMTM1Ny1iZTNiMzdkNi1hMTIwLTRiNzEtYTgzMi1hN2FiZTcyOTIwYjkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgxMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MTJUMDczMjQ5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Y2I5NTRlZjg2YmYxNzIwMjVmYmQ1MjFiNjQ0NzIyODE4MTMyMDQ2NzJkZTI4MTRjNzhjNTBhYjcyNTQyN2Y0ZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Z05LZj1OSBxBEdtzL5aPMNbCZjQ7r0uaNUEScXwYlHg">

closed 2024-08-12 by ajiththangavel

ajiththangavel, 2024-08-12:
Then what is the solution i want acheive this multiple response for a single request. is there any kind of mechanism where i can apply the client details(session) in the program and send the multiple responses.