COVESA/vsomeip #746: HELP: Config files likely not being read properly

lukas162, 2024-07-30:
### vSomeip Version

v3.4.10

### Boost Version

1.74.0.3

### Environment

Debian 12 (Raspberry Pi OS)

### Describe the bug

Simple request/response method between two Raspberry Pi 4.

Config files are adjusted, but Client routes unicast: 127.0.0.1.
This likely is the reason for "[error] Routing info for remote service could not be found", but I am not sure.
It also seems like as if the application ids from the config files are not used either and instead given the standard value "0100".
I think that my config files are not being read properly, but again, I am uncertain if that is the case.

These error do not occur when I run the server and client on my pc througt localhost, only when I try to communicate between my Raspberry Pis.

**I have already tried many solutions which were suggested in prior issues, but feel free to link me a certain issue if you think it might help. Otherwise feel free to comment, every message is highly appreciated.**

Here is the code of all relevant files, alternatively you can download my code here
[server_client.zip](https://github.com/user-attachments/files/16420418/server_client.zip):
**vsomeip_client.json:**
{
    "unicast": "192.168.178.141",
    "logging": {
        "level": "debug"
    },
    "applications": [
        {
            "name": "HelloWorldClient",
            "id": "0x5678"
        }
    ],
    "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "unicast" : "192.168.178.104",
            "unreliable" : "30509"
        }
    ],
    "routing" : "HelloWorldClient",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}

**vsomeip_server.json:**
{
    "unicast": "192.168.178.104",
    "logging": {
        "level": "debug"
[... 239 lines cut ...]
2024-07-29 23:29:35.496418 [info] create_local_server: Listening @ /tmp/vsomeip-100
2024-07-29 23:29:35.497883 [info] OFFER(0100): [1234.5678:0.0] (true)
2024-07-29 23:29:45.455556 [info] vSomeIP 3.4.10 | (default)

_Client logs:_
2024-07-29 23:30:04.890580 [info] Parsed vsomeip configuration in 0ms
2024-07-29 23:30:04.892200 [info] Configuration module loaded.
2024-07-29 23:30:04.892353 [info] Security disabled!
2024-07-29 23:30:04.892466 [info] Initializing vsomeip (3.4.10) application "HelloWorldClient".
2024-07-29 23:30:04.892907 [info] Instantiating routing manager [Host].
2024-07-29 23:30:04.894928 [info] create_routing_root: Routing root @ /tmp/vsomeip-0
2024-07-29 23:30:04.896944 [info] Service Discovery enabled. Trying to load module.
2024-07-29 23:30:04.902296 [info] Service Discovery module loaded.
2024-07-29 23:30:04.903081 [info] Application(HelloWorldClient, 0100) is initialized (11, 100).
2024-07-29 23:30:04.903497 [info] Starting vsomeip application "HelloWorldClient" (0100) using 2 threads I/O nice 255
2024-07-29 23:30:04.905468 [info] Client [0100] routes unicast:127.0.0.1, netmask:255.255.255.0
2024-07-29 23:30:04.905237 [info] main dispatch thread id from application: 0100 (HelloWorldClient) is: 7fb658f180 TID: 5536
2024-07-29 23:30:04.909092 [info] Watchdog is disabled!
Client registered successfully.
**2024-07-29 23:30:04.910652 [error] Routing info for remote service could not be found! (0100): [1234.5678.0421] 0001**
2024-07-29 23:30:04.905494 [info] shutdown thread id from application: 0100 (HelloWorldClient) is: 7fb5d7f180 TID: 5537
2024-07-29 23:30:04.911294 [info] io thread id from application: 0100 (HelloWorldClient) is: 7fb76d6040 TID: 5534
2024-07-29 23:30:04.911396 [info] io thread id from application: 0100 (HelloWorldClient) is: 7fb4d5f180 TID: 5539
2024-07-29 23:30:04.913263 [info] vSomeIP 3.4.10 | (default)
2024-07-29 23:30:04.914018 [info] Network interface "lo" state changed: up

lutzbichler, 2024-08-06:
Please also provide the network settings (ip a, ip r) on both nodes. It seems they do not match the configured values and therefore vsomeip uses the local network interface (lo).

lukas162, 2024-08-06:
Thank you for your reply!

Node 192.169.178.104 (server):
ge95cos@raspberrypi:~/ba_maslic $ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether dc:a6:32:32:5a:e9 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether dc:a6:32:32:5a:eb brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.104/24 brd 192.168.178.255 scope global dynamic noprefixroute wlan0
       valid_lft 863742sec preferred_lft 863742sec
    inet6 2a01:c23:c47f:2c00:5300:2c7e:da8b:a25d/64 scope global dynamic noprefixroute 
       valid_lft 6963sec preferred_lft 3363sec
    inet6 fe80::12c4:a51f:5c31:56ad/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
4: eth0.10@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state LOWERLAYERDOWN group default qlen 1000
    link/ether dc:a6:32:32:5a:e9 brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.1/24 brd 192.168.10.255 scope global eth0.10
       valid_lft forever preferred_lft forever
ge95cos@raspberrypi:~/ba_maslic $ ip r
default via 192.168.178.1 dev wlan0 proto dhcp src 192.168.178.104 metric 600 
192.168.10.0/24 dev eth0.10 proto kernel scope link src 192.168.10.1 linkdown 
192.168.178.0/24 dev wlan0 proto kernel scope link src 192.168.178.104 metric 600

Node 192.168.178.141 (client):
ge95cos@raspberrypi2:~/ba_maslic $ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether d8:3a:dd:b0:37:5d brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether d8:3a:dd:b0:37:5f brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.141/24 brd 192.168.178.255 scope global dynamic noprefixroute wlan0
       valid_lft 863716sec preferred_lft 863716sec
    inet6 2a01:c23:c47f:2c00:50f4:9ef1:55cf:ef7c/64 scope global dynamic noprefixroute 
       valid_lft 6919sec preferred_lft 3319sec
    inet6 fe80::939a:e2e0:32da:730d/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
4: eth0.10@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state LOWERLAYERDOWN group default qlen 1000
    link/ether d8:3a:dd:b0:37:5d brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.2/24 brd 192.168.10.255 scope global eth0.10
       valid_lft forever preferred_lft forever
ge95cos@raspberrypi2:~/ba_maslic $ ip r
default via 192.168.178.1 dev wlan0 proto dhcp src 192.168.178.141 metric 600 
192.168.10.0/24 dev eth0.10 proto kernel scope link src 192.168.10.2 linkdown 
192.168.178.0/24 dev wlan0 proto kernel scope link src 192.168.178.141 metric 600

lutzbichler, 2024-08-09:
It seems it does not read the configuration files (permissions / path). Do you seem something like:
2024-08-09 08:22:40.663418 [info] Using configuration file: "config/vsomeip-udp-client.json".
in your log?

lukas162, 2024-08-14:
Please excuse my late response.

I have provided all the logs in the issue.
As you can see, I do not get a line with "Using configuration file". I only have a line which says "Configuration module loaded".