COVESA/vsomeip #750: [BUG]: <Multiple apps simultaneously send TP messages, resulting in TP message disorder and packet loss>

wulijiao, 2024-08-09:
### vSomeip Version

3.3.8

### Boost Version

1.71.0

### Environment

Ubuntu 20.04

### Describe the bug

Multiple application nodes and services simultaneously send messages that require TP processing (these messages are events, and the services under the application are all offer attributes), with a frequency of 20Hz/50ms.
From the captured messages, it appears that there is confusion and loss of TP messages.
The basic phenomenon is like this, thanks!

### Reproduction Steps

[SOMEIPInit_1.json](https://github.com/user-attachments/files/16558323/SOMEIPInit_1.json)
[SOMEIPInit_2.json](https://github.com/user-attachments/files/16558324/SOMEIPInit_2.json)
[SOMEIPInit_3.json](https://github.com/user-attachments/files/16558327/SOMEIPInit_3.json)
Run application nodes with similar attachment configurations

### Expected behaviour

Send TP data in an orderly and correct manner.

### Logs and Screenshots

[test_00007_20240808175718.zip](https://github.com/user-attachments/files/16558398/test_00007_20240808175718.zip)

wulijiao, 2024-08-14:
I'm very sorry, I have updated the correct PCAP file for 50ms. The PCAP message log cycle I previously provided was not accurate
[mrr_4cr_0814_someiptimer.zip](https://github.com/user-attachments/files/16610213/mrr_4cr_0814_someiptimer.zip)

vidz12, 2024-08-14:
Is it happening always?

wulijiao, 2024-08-15:
> Is it happening always?

Yes.

closed 2024-11-14 by wulijiao