COVESA/vsomeip #757: [BUG]: <title>Compilation issue while building the code on Android 14

shivamkumar123, 2024-08-17:
### vSomeip Version

5b4d2a64f11e23240728a7722dc6ac37a07b0752

### Boost Version

1.71

### Environment

Android 14

### Describe the bug

Compilation issue while building the code on Android 14 with Flag -DUSE_DLT Enabled:

external/fev/vsomeip/implementation/tracing/src/connector_impl.cpp:279:35: error: cannot assign to variable 'its_data_size' with const-qualified type 'const uint16_t' (aka 'const unsigned short')
                    its_data_size = VSOMEIP_FULL_HEADER_SIZE;
                    ~~~~~~~~~~~~~ ^
external/fev/vsomeip/implementation/tracing/src/connector_impl.cpp:217:20: note: variable 'its_data_size' declared const here
    const uint16_t its_data_size

### Logs and Screenshots

$ make
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=14
PRODUCT_INCLUDE_TAGS=com.android.mainline
TARGET_PRODUCT=mek_8q_car2
TARGET_BUILD_VARIANT=userdebug
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=cortex-a53
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=cortex-a9
HOST_OS=linux
HOST_OS_EXTRA=Linux-5.15.0-117-generic-x86_64-Ubuntu-20.04.6-LTS
HOST_CROSS_OS=windows
BUILD_ID=UQ1A.231205.015
OUT_DIR=out
============================================
[ 10% 7/67] //external/fev/vsomeip:libvsomeip3 clang++ implementation/tracing/src/connector_impl.cpp [arm]
FAILED: out/soong/.intermediates/external/fev/vsomeip/libvsomeip3/android_vendor.34_arm_armv7-a-neon_cortex-a9_shared/obj/external/fev/vsomeip/implementation/tracing/src/connector_impl.o
PWD=/proc/self/cwd prebuilts/clang/host/linux-x86/clang-r487747c/bin/clang++ -c -D__ANDROID_VNDK__ -D__ANDROID_VENDOR__ -mthumb -Os -fomit-frame-pointer -DANDROID -fmessage-length=0 -W -Wall -Wno-unused 
-Winit-self -Wpointer-arith -Wunreachable-code-loop-increment -no-canonical-prefixes -DNDEBUG -UDEBUG -fno-exceptions -Wno-multichar -O2 -g -fdebug-default-version=5 -fno-strict-aliasing -Werror=date-tim
e -Werror=pragma-pack -Werror=pragma-pack-suspicious-include -Werror=string-plus-int -Werror=unreachable-code-loop-increment -Wno-error=deprecated-declarations -D__compiler_offsetof=__builtin_offsetof -f
addrsig -fcommon -Werror=int-conversion -Wno-reserved-id-macro -fcolor-diagnostics -Wno-sign-compare -Wno-inconsistent-missing-override -Wno-c99-designator -Wno-gnu-designator -Wno-gnu-folding-constant -
Wunguarded-availability -D__ANDROID_UNAVAILABLE_SYMBOLS_ARE_WEAK__ -ffp-contract=off -fdebug-prefix-map=/proc/self/cwd= -ftrivial-auto-var-init=zero -enable-trivial-auto-var-init-zero-knowing-it-will-be-
removed-from-clang -Wno-unused-command-line-argument -ffunction-sections -fdata-sections -fno-short-enums -funwind-tables -fstack-protector-strong -Wa,--noexecstack -D_FORTIFY_SOURCE=2 -Wstrict-aliasing=
2 -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Werror=format-security -nostdlibinc -fdebug-info-for-profiling -Wno-enum-compare -Wno-enum-compare-switch -Wno-null-
pointer-arithmetic -Wno-null-dereference -Wno-pointer-compare -Wno-final-dtor-non-final-class -Wno-psabi -Wno-null-pointer-subtraction -Wno-string-concatenation -Wno-deprecated-non-prototype -msoft-float
 -march=armv7-a -mfloat-abi=softfp -mfpu=neon  -target armv7a-linux-androideabi34 -fPIC -Wsign-promo -Wimplicit-fallthrough -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS -Wno-gnu-include-next -fvisibility-i
nlines-hidden  -Iexternal/fev/vsomeip/interface -Iexternal/fev/vsomeip/interface -Iexternal/fev/vsomeip/implementation/helper -Iexternal/fev/vsomeip -Iexternal/libcxxabi/include -Iexternal/fev/boost -Isy
stem/logging/liblog/include_vndk -Isystem/core/libutils/include -Isystem/libbase/include -Iexternal/fmtlib/include -Isystem/core/libcutils/include_outside_system -Isystem/core/libprocessgroup/include -Is
ystem/core/libsystem/include -Iexternal/libcxx/include -isystem out/soong/.intermediates/bionic/libc/libc/android_vendor.34_arm_armv7-a-neon_cortex-a9_shared/gen/include -isystem bionic/libc/kernel/uapi/
asm-arm -isystem bionic/libc/kernel/uapi -isystem bionic/libc/kernel/android/scsi -isystem bionic/libc/kernel/android/uapi -DVSOMEIP_BOOST_VERSION=107100 -DVSOMEIP_INTERNAL_SUPPRESS_DEPRECATED -DWITHOUT_
SYSTEMD '-DVSOMEIP_VERSION="3.4.10"' '-DVSOMEIP_COMPAT_VERSION="3.4.10"' '-DVSOMEIP_BASE_PATH="/vendor/run/someip/"' -DUSE_DLT -std=c++17 -fexceptions -Wno-non-virtual-dtor -Wno-unused-const-variable -Wn
o-unused-parameter -Wno-unused-private-field -Wno-unused-lambda-capture -Wno-unused-variable -Wno-unused-local-typedef -Wno-sign-compare -Wno-format -Wno-header-guard -Wno-overloaded-virtual -Wno-implici
t-fallthrough -Wno-error -Wno-shorten-64-to-32 -D_GTHREAD_USE_MUTEX_INIT_FUNC -D_GTHREAD_USE_RECURSIVE_MUTEX_INIT_FUNC -frtti  -Werror=bool-operation -Werror=format-insufficient-args -Werror=implicit-int
-float-conversion -Werror=int-in-bool-context -Werror=int-to-pointer-cast -Werror=pointer-to-int-cast -Werror=xor-used-as-pow -Wno-void-pointer-to-enum-cast -Wno-void-pointer-to-int-cast -Wno-pointer-to-
int-cast -Werror=fortify-source -Werror=address-of-temporary -Werror=null-dereference -Werror=return-type -Wno-tautological-constant-compare -Wno-tautological-type-limit-compare -Wno-reorder-init-list -W
no-implicit-int-float-conversion -Wno-tautological-overlap-compare -Wno-deprecated-copy -Wno-range-loop-construct -Wno-zero-as-null-pointer-constant -Wno-deprecated-anon-enum-enum-conversion -Wno-pessimi
zing-move -Wno-non-c-typedef-for-linkage -Wno-align-mismatch -Wno-error=unused-but-set-variable -Wno-error=unused-but-set-parameter -Wno-error=deprecated-builtins -Wno-error=deprecated -Wno-error=single-
bit-bitfield-constant-conversion -Wno-error=enum-constexpr-conversion -Wno-format-insufficient-args -Wno-sizeof-array-div -Wno-incompatible-function-pointer-types -Wno-unused-but-set-variable -Wno-unused
-but-set-parameter -Wno-unqualified-std-cast-call -Wno-bitwise-instead-of-logical -Wno-misleading-indentation -Wno-array-parameter -Wno-gnu-offsetof-extensions -MD -MF out/soong/.intermediates/external/f
ev/vsomeip/libvsomeip3/android_vendor.34_arm_armv7-a-neon_cortex-a9_shared/obj/external/fev/vsomeip/implementation/tracing/src/connector_impl.o.d -o out/soong/.intermediates/external/fev/vsomeip/libvsome
ip3/android_vendor.34_arm_armv7-a-neon_cortex-a9_shared/obj/external/fev/vsomeip/implementation/tracing/src/connector_impl.o external/fev/vsomeip/implementation/tracing/src/connector_impl.cpp
external/fev/vsomeip/implementation/tracing/src/connector_impl.cpp:279:35: error: cannot assign to variable 'its_data_size' with const-qualified type 'const uint16_t' (aka 'const unsigned short')
[... 15 lines cut ...]
-null-pointer-arithmetic -Wno-null-dereference -Wno-pointer-compare -Wno-final-dtor-non-final-class -Wno-psabi -Wno-null-pointer-subtraction -Wno-string-concatenation -Wno-deprecated-non-prototype -march
=armv8-a -mcpu=cortex-a53 -target aarch64-linux-android34 -fPIC -Wsign-promo -Wimplicit-fallthrough -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS -Wno-gnu-include-next -fvisibility-inlines-hidden  -Iexterna
l/fev/vsomeip/interface -Iexternal/fev/vsomeip/interface -Iexternal/fev/vsomeip/implementation/helper -Iexternal/fev/vsomeip -Iexternal/libcxxabi/include -Iexternal/fev/boost -Isystem/logging/liblog/incl
ude_vndk -Isystem/core/libutils/include -Isystem/libbase/include -Iexternal/fmtlib/include -Isystem/core/libcutils/include_outside_system -Isystem/core/libprocessgroup/include -Isystem/core/libsystem/inc
lude -Iexternal/libcxx/include -isystem out/soong/.intermediates/bionic/libc/libc/android_vendor.34_arm64_armv8-a_cortex-a53_shared/gen/include -isystem bionic/libc/kernel/uapi/asm-arm64 -isystem bionic/
libc/kernel/uapi -isystem bionic/libc/kernel/android/scsi -isystem bionic/libc/kernel/android/uapi -DVSOMEIP_BOOST_VERSION=107100 -DVSOMEIP_INTERNAL_SUPPRESS_DEPRECATED -DWITHOUT_SYSTEMD '-DVSOMEIP_VERSI
ON="3.4.10"' '-DVSOMEIP_COMPAT_VERSION="3.4.10"' '-DVSOMEIP_BASE_PATH="/vendor/run/someip/"' -DUSE_DLT -std=c++17 -fexceptions -Wno-non-virtual-dtor -Wno-unused-const-variable -Wno-unused-parameter -Wno-
unused-private-field -Wno-unused-lambda-capture -Wno-unused-variable -Wno-unused-local-typedef -Wno-sign-compare -Wno-format -Wno-header-guard -Wno-overloaded-virtual -Wno-implicit-fallthrough -Wno-error
 -Wno-shorten-64-to-32 -D_GTHREAD_USE_MUTEX_INIT_FUNC -D_GTHREAD_USE_RECURSIVE_MUTEX_INIT_FUNC -frtti  -Werror=bool-operation -Werror=format-insufficient-args -Werror=implicit-int-float-conversion -Werro
[... cut ...]
external/fev/vsomeip/implementation/tracing/src/connector_impl.cpp:279:35: error: cannot assign to variable 'its_data_size' with const-qualified type 'const uint16_t' (aka 'const unsigned short')
                    its_data_size = VSOMEIP_FULL_HEADER_SIZE;
                    ~~~~~~~~~~~~~ ^
external/fev/vsomeip/implementation/tracing/src/connector_impl.cpp:217:20: note: variable 'its_data_size' declared const here
    const uint16_t its_data_size
    ~~~~~~~~~~~~~~~^~~~~~~~~~~~~
1 error generated.
01:55:55 ninja failed with: exit status 1

duartenfonseca, 2024-08-28:
hi there, the error was introduced on a recent PR (https://github.com/COVESA/vsomeip/pull/727) and was identified internally. The correction will be introduced. The fix was to change `const uint16_t its_data_size` to ` uint16_t its_data_size`

duartenfonseca, 2024-08-28:
@shivamkumar123 could you try this? thanks

closed 2024-09-10 by duartenfonseca