COVESA/vsomeip #758: [BUG]: Compiler error on Ubuntu 20.04 due the missing vector header

nate-li-parasoft, 2024-08-29:
### vSomeip Version

v3.4.10

### Boost Version

1.7*

### Environment

Ubuntu 20.04

### Describe the bug

The header file "interface/vsomeip/handler.hpp" uses the "std::vector" type without including the relative header file "<vector>". It compiles OK on higher versions of GCC on Ubuntu 22.04, but it is just pure luck.

The "handler.h" header should explicitly include <vector> to avoid relying on the implementational details of other headers.

duartenfonseca, 2024-09-06:
hi @nate-li-parasoft thanks for the input. 
 what are the reproduction steps for this bug? building vsomeip on Ubuntu 20.04?

nate-li-parasoft, 2024-09-09:
I think the compiling error can be reproducted by compiling the project using GCC-9.

duartenfonseca, 2024-09-10:
here is a log of me building vsomeip on gcc9:
[logs.txt](https://github.com/user-attachments/files/16942767/logs.txt)

is there any other specific characteristic of the environment that could be creating this fail? could you send logs?

nate-li-parasoft, 2024-09-20:
The compiler error appears when I tried to compile the library using javacpp(a JNI helper library).

I managed to extract the step generating the error as follows.
Put the following code inside a "test.cpp" at the root of the repository, and run command "g++ -I./interface test.cpp"
```cpp
#include <vsomeip/enumeration_types.hpp>
#include <vsomeip/handler.hpp>

int main()
{
  return 1;
}