COVESA/vsomeip #759: How to use tcp transport large data?

KudoWu, 2024-08-30:
### vSomeip Version

v3.4.10

### Boost Version

1.71

### Environment

Ubuntu 20.04

### Describe the bug

Hello, COVESA!

I have read the guide and reviewed other similar issues. I understand that the 1416-byte limit is specific to using UDP as the transport protocol.

According to the guide, the payload size settings are described as follows:

> **payload-sizes (array)**  
> Array to limit the maximum allowed payload sizes per IP and port. If not specified otherwise, the allowed payload sizes are unlimited. The settings in this array only affect communication over TCP. To limit the local payload size, `max-payload-size-local` can be used.

Based on this, I believe I am correctly using TCP to transport large data. However, I am still encountering the same issue:

```shell
[error] sei::send_intern: Dropping too big message (733962 Bytes). Maximum allowed message size is: 1416 Bytes.
```

Could you please provide an example or guidance on how to resolve this issue? Thank you very much!

### Reproduction Steps

This is my services config:

```shell
  "services": [                                                                
      {                                                                        
          "service": "0x80e9",                                                 
          "instance": "0x814e",                                                
          "reliable" : { "port" : "30501", "enable-magic-cookies" : "true" },  
          "unreliable": "30510",                                               
          "events" : {                                                         
              "event": "0x80fd",                                               
              "is_reliable": "true"                                            
          },                                                                   
          "someip-tp" : {                                                      
              "service-to-client": [                                           
                   { "service": "0x80e9", "instance": "0x814e", "eventgroup": "0x80fc" }                                    
              ]                                                                
          }                                                                    
      }                                                                        
  ],  
```

### Expected behaviour

I can use TCP or another ways to transport large data .

KudoWu, 2024-08-30:
I also try to use someip-tp, but it has the smae problem.

```
    "services": [
        {
            "service": "0x80e9",
            "instance": "0x814e",
            "unreliable": "30510",
            "events": [{
                "event": "0x80fe",
                "is_field": "false",
                "is_reliable": true
            }],
            "someip-tp": {
                "service-to-client": [ "0x80e9", "0x80fe"]
            }
        }
    ],
```

Billkouts96, 2024-08-30:
Hello KudoWu,
As I can see from your configuration, it seems that you define both unreliable(UDP) and reliable ports (TCP).
You can try, remove 'unreliable' and hopefully will work.

lutzbichler, 2024-08-30:
Unfortunately, you provide zero information on what you actually do. Did you set the reliable flag on the message you are trying to send (message_base::set_reliable or at creation time)?

KudoWu, 2024-08-30:
> Hello KudoWu, As I can see from your configuration, it seems that you define both unreliable(UDP) and reliable ports (TCP). You can try, remove 'unreliable' and hopefully will work.

Hello, [Billkouts96](https://github.com/Billkouts96)
Thank you for your help! Actually, I have tried to remove 'unreliable' , but I had the other problem:
```shell
2024-08-30 19:50:00.692114 [info] endpoint_manager_impl::create_remote_client: 192.168.3.208:30501 reliable: 1 using local port: 0
2024-08-30 19:50:00.739588 [warning] create_eventgroup_entry: Cannot create subscription as unreliable endpoint is zero: [80e9.814e.80fc]
2024-08-30 19:50:00.741354 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [80e9.814e.80fc] 2
2024-08-30 19:50:00.689206 [warning] create_eventgroup_entry: Cannot create subscription as unreliable endpoint is zero: [80e9.814e.80fc]
2024-08-30 19:50:00.691456 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [80e9.814e.80fc] 2
2024-08-30 19:50:00.686764 [warning] create_eventgroup_entry: Cannot create subscription as unreliable endpoint is zero: [80e9.814e.80fc]
2024-08-30 19:50:00.689158 [warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [80e9.814e.80fc] 2
2024-08-30 19:50:01.433518 [info] REMOTE SUBSCRIBE(0000): [80e8.814d.80fc] from 192.168.3.208:35097 reliable was accepted
```

KudoWu, 2024-08-30:
> Unfortunately, you provide zero information on what you actually do. Did you set the reliable flag on the message you are trying to send (message_base::set_reliable or at creation time)?
I'm sorry. Actually, I don't find any interface that I can set reliable.
Here's your response refined in English:

Hello, [lutzbichler](https://github.com/lutzbichler), thank you for being willing to help me!

I'm very sorry for not providing enough information earlier. First, I would like to clarify that I am using CommonAPI C with SomeIP+ vsomeip, and I am trying to transmit large amounts of data within the `PublishEvent`. I haven't found any interface to set the reliable flag. I'm wondering if there's a significant difference in how the reliable flag is handled between CommonAPI C SomeIP and the standard vsomeip interfaces.

Additionally, I noticed in the "CommonAPI-C---SomeIP-in-10-minutes" example that the `method` can have `SomeIpReliable = true` set in the fdepl file. So, I added a similar configuration in my own fdepl file:

```plaintext
define org.genivi.commonapi.someip.deployment for interface xxxxx {
    SomeIpServiceID = 33001

    broadcast FirmwarePublish {
        SomeIpEventID = 33021
        SomeIpEventGroups = { 33020 } 
        SomeIpReliable = true
    }
}

```
That's great! I can use tcp transport large data. So, if there are other methods to set_reliable in CommonAPI C SomeIP? Thank you!

lutzbichler, 2024-08-30:
If you deploy your broadcast to be reliable ("SomeIpReliable = true") CommonAPI will do the setting of the reliable flag in vsomeip for you. Setting the flag manually only is needed if you use the vsomeip API.

As I think the problem is solved, I will close the issue.

closed 2024-08-30 by lutzbichler