COVESA/vsomeip #768: how configure e2e protection profile 4

GHARBIAchref, 2024-09-21:
### vSomeip Version

v3.4.10

### Boost Version

1.83

### Environment

Windows and linux MSVC

### Describe the bug

 Hello  
 I want to use end-to-end protection in my application. Is using end-to-end protection done solely through configuration in .json configuration files on both the client and server sides of the services, or does it also require code implementation within my algorithm? And how can I verify if the E2E protection profile 4 is working correctly in Wireshark? 
 thanks

duartenfonseca, 2024-09-24:
hi @GHARBIAchref ,
you have one example of a E2E application: [client](https://github.com/COVESA/vsomeip/blob/master/test/network_tests/e2e_tests/e2e_profile_04_test_client.cpp) and [service](https://github.com/COVESA/vsomeip/blob/master/test/network_tests/e2e_tests/e2e_profile_04_test_service.cpp). they use this configs: [client](https://github.com/COVESA/vsomeip/blob/master/test/network_tests/e2e_tests/conf/e2e_profile_04_test_client_external.json.in) and [service](https://github.com/COVESA/vsomeip/blob/master/test/network_tests/e2e_tests/conf/e2e_profile_07_test_service_external.json.in). 
i attach also the pcap of the test, so that you have an example of something you should see on your side
[e2e_profile_04_test_external.zip](https://github.com/user-attachments/files/17113186/e2e_profile_04_test_external.zip)

closed 2024-09-24 by duartenfonseca

fcmonteiro, 2025-02-05:
we have updated with our documentation with this information and examples please check,
https://github.com/COVESA/vsomeip/blob/master/documentation/vsomeipConfiguration.md#20-e2e

khaled-tantawi, 2025-02-05:
@fcmonteiro Actually, I used profile 04 for a method that sends data. I configured it as indicated in the vsomeip configuration. The method sends:
00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D

But I have a problem: the received message is:
00 26 00 00 99 00 00 FF 0E 39 24 4C 0C 0D 0E 0F 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D

So, the profile 04 header, which is 12 bytes in size (without data), has replaced the first 12 bytes of the sent data. However, I want it to be added at the beginning of the payload, not to replace the sent data.