COVESA/vsomeip #781: Vsomeip-3.5.1 Client UDP Port Closed, Server Still Sending Events

FishPickBeans, 2024-10-15:
### vSomeip Version

v3.5.1

### Boost Version

1.84

### Environment

Linux

### Describe the bug

Linux use Vsomeip-3.5.1 code, Steps:
Firstly, client subscribe server's event-group successed.（vsomip_server.json noly support UDP） 
Secondly, client reboot, But server still sending client's sub event.

### Reproduction Steps

Linux use Vsomeip-3.5.1 code, Steps:
Firstly, client subscribe server's event-group successed.（vsomip_server.json noly support UDP） 
Secondly, client reboot, But server still sending client's sub event.
Thirdly, tcpdump parse by wireshark.

### Expected behaviour

after client reboot, server should stop sending before client's sub event.

### Logs and Screenshots

[vsomeip-3.5.1 log.txt](https://github.com/user-attachments/files/17371111/vsomeip-3.5.1.log.txt)

FishPickBeans, 2024-10-15:
[Uploading tcpdump_20241014200558.zip…]()

duartenfonseca, 2024-12-12:
> Uploading tcpdump_20241014200558.zip…

@FishPickBeans cannot open the tcpdump

FishPickBeans, 2024-12-17:
sorry, we had select vsomeip3.4.10, this issue no appare! 
Tks!

anthonypayne-GM, 2025-02-18:
Hello, 
I have seen this issue as well. It seems that if the client exits and does not send a StopSubscribe the server keeps publishing the Event even after the UDP socket has been closed.

If the client sends the StopSubscribe before exiting, the service stops publishing the event as expected. 

The TTL setting in the service discovery should kick in when the client exits and it stops sending the Subscribe. 

I can provide logs/packet capture if necessary. It is pretty easy to replicate.

duartenfonseca, 2025-02-19:
hi @anthonypayne-GM, which version do you use?  please provide your pcap's, and how you got that reproduction.

anthonypayne-GM, 2025-02-19:
Hello @duartenfonseca ,

See attachment. It has packet capture, json configuration, and test code that replicates the issue. 

[vsomeip.issue781.zip](https://github.com/user-attachments/files/18873201/vsomeip.issue781.zip)

anthonypayne-GM, 2025-02-19:
> which version do you use?

Sorry forgot to answer this, but this was using vsomeip version 3.5.1.