COVESA/vsomeip #782: Can not call a method with TCP, bud UDP works, SD seems to be working fine

callanchen, 2024-10-15:
### vSomeip Version

3.1.14.1

### Boost Version

1.71

### Environment

Ubuntu 22.04

### Describe the bug

When I configure vsomeip, everything works fine with UDP, but when I switch to TCP, it seems that I can't call the method normally, but I can see that SD is working normally in pcap.
Client vsomeip.json
```json
{
    "unicast": "192.168.7.4",
    "netmask": "255.255.255.0",
    "logging":
    {
        "level": "debug",
        "console": "true",
        "file":
        {
            "enable": "true",
            "path": "/var/vsomeip.log"
        },
        "dlt": "true"
    },
    "applications":
    [
        {
            "name": "ACCApplication_instance",
            "id": "0x3cba"
        },
        {
            "name": "statemanager_acc_instance",
            "id": "0x518b"
        }
    ],
    "services":
    [
    ],
    "routing": "routingmanagerd",
    "service-discovery":
    {
        "enable": "true",
        "multicast": "224.224.224.245",
        "port": "30491",
        "initial_delay_min": "10",
        "initial_delay_max": "100",
        "repetitions_base_delay": "200",
        "repetitions_max": "3",
        "ttl": "10",
        "protocol": "udp"
    }
}
```

Service vsomeip.json

```json
{
    "unicast": "192.168.7.2",
    "netmask": "255.255.255.0",
    "logging":
    {
        "level": "debug",
        "console": "false",
        "file":
        {
            "enable": "false",
            "path": "/var/log/vsomeip.log"
        },
        "dlt": "false"
    },
    "applications":
    [
        {
            "name": "PPDMApplication_instance",
            "id": "0xb16b"
        },
        {
            "name": "PPDMApplication_instance",
            "id": "0xb16b"
        },
        {
            "name": "statemanager_instance",
            "id": "0xa70a"
        }
    ],
    "services":
    [
        {
            "service": "0x2",
            "instance": "0x1f",
            "reliable":
            {
                "port": "8000",
                "enable-magic-cookies": "true"
            },
            "events":
            [
            ],
            "eventgroups":
            [
            ]
        }
    ],
    "routing": "routingmanagerd",
    "service-discovery":
    {
        "enable": "true",
        "multicast": "224.224.224.245",
        "port": "30491",
        "initial_delay_min": "10",
        "initial_delay_max": "100",
        "repetitions_base_delay": "200",
        "repetitions_max": "3",
        "ttl": "10",
        "cyclic_offer_delay": "2000",
        "request_response_delay": "1500",
        "protocol": "udp"
    }
}
```
I should have configured the routing configuration under Ubuntu, but it still doesn't work.
Service:
```
/opt/PPDMApplication_instance # ip route
default via 192.168.7.1 dev eth0 
192.168.7.0/24 dev eth0 scope link  src 192.168.7.2 
224.224.224.245 via 192.168.7.1 dev eth0
```

Client:
```
/opt/ACCApplication_instance # ip route
default via 192.168.7.1 dev eth0 
192.168.7.0/24 dev eth0 scope link  src 192.168.7.4 
224.224.224.245 via 192.168.7.1 dev eth0
```

pcap file using UDP: [success_udp.zip](https://github.com/user-attachments/files/17375741/success_udp.zip)
pcap file using TCP: [tcp.zip](https://github.com/user-attachments/files/17375751/tcp.zip)

Can any kind soul help me? Thanks a lot.

fcmonteiro, 2024-10-15:
Hi @callanchen, can you retest with vSomeIP 3.4.x or 3.5.x
Currently we only support those two versions.
Thanks

lutzbichler, 2024-10-15:
The transport protocol must be set  on the message object (message::set_reliable).

callanchen, 2024-10-18:
Ok, I think I have solved the problem.
The problem is that I make the client to find "ANY" instances of a service.
I am not sure this is the root reason, but when I use instance id instead of "ANY", the problem solved.
I aming using Autosar Ara Sdk, maybe there are some bugs in it.

callanchen, 2024-10-18:
Anyway, I will look into it when I am in free time.
I close the issue.

closed 2024-10-18 by callanchen