COVESA/vsomeip #795: [BUG]: Ignored SD message from unknown port

manojkumarms1999, 2024-10-25:
### vSomeip Version

v3.1.20

### Boost Version

1.73

### Environment

QNX & Android

### Describe the bug

The isssue is observed on while QNX is sending a OFFER with a port of 30490
[image]
While the QNX is sending an OFFER message, there are erros printed on binding to the IP address.

> [error] udp_server_endpoint_impl: bind failed (Can't assign requested address)

The reason beyond the error is yet to be idenfied

**Verisoning 👍**  :
vSomeIP 3.4.10 on QNX
vSomeIP 3.1.20 on Android

### Reproduction Steps

SOMEIP should send offer via listed port number

### Expected behaviour

SOMEIP is not send offer via listed port number and client says OFFEER from unknown port is not accepted.

KareemLMR, 2024-10-27:
[error] udp_server_endpoint_impl: bind failed (Can't assign requested address) seems to be a network configuration issue. Are you sure the multicast and unicast VLANs are configured and up and the port 30490 is not busy used by another service?
You can check netstat and wireshark tcpdump logs for your vsomeip network.

manojkumarms1999, 2024-10-28:
Hi @KareemLMR This issue is fixed now

After altering logic the network configuration, this issue is resolved
- we had made a 2 condition on the QNZ startup script to check whether the VLAN status is ACTIVE and MAC address is set
Thanks

closed 2024-10-28 by manojkumarms1999