COVESA/vsomeip #797: [BUG]: <title>Supporting multiple versions of the same service.

zhaoxin39913, 2024-10-26:
### vSomeip Version

v3.4.10

### Boost Version

1.7.6

### Environment

Ubuntu 20.04

### Describe the bug

Hi vsomeip experts,

One issue I would like to check with you guys.
 
In order to support multiple versions of the same service (defined in **SOME/IP SD spec PRS_SOMEIPSD_00512**), there could be multiple services offered with the same serviceID and instanceID but different major versions within a network.  Using API request_service(serviceId, instanceId, ANY_MAJOR) should be able to find all services with different major versions for a given serviceID/instanceID pair. Next are_available() API should be able to retrieve all those major versions.

The problem is how does **availability_handler()** get triggered when those services with different major versions are detected, since only the serviceID and instanceID is passing to the handler without major version.

This is one of our customer user case, what's the solution do you suggest to handle it in our application?

### Reproduction Steps

N/A

### Expected behaviour

N/A

### Logs and Screenshots

N/A

KareemLMR, 2024-11-16:
Hello,
Did you tried calling request_service API with setting major version?
[image]

zhaoxin39913, 2024-11-18:
Hi @KareemLMR,

Yes, I have tested request_service() API for the same service-instances with different major versions, it works without any errors from the log. But I am not able to verify it using vsomeip server API since the offer_service() API doesn't support offering the same service-instances with different major versions within the same VLAN network which I mentioned in https://github.com/COVESA/vsomeip/issues/809.

zhaoxin39913, 2024-11-21:
Hi @KareemLMR ,

Even we can specify the versions in some client side APIs like **request_service**(), **subscribe**() and **register_availability_handler**()/**unregister_availability_handler**(), there are still some APIs don't support passing the versions, which means the serviceID-instanceID-majVersion is not fully manageable independently. Thus I think we need to have below changes to fully support PRS_SOMEIPSD_00512 on client side to make sure the version is present in pair.

`virtual void release_service(service_t _service, instance_t _instance) = 0;`  
changed to
`virtual void release_service(service_t _service, instance_t _instance, major_version_t _major = ANY_MAJOR, minor_version_t _minor = ANY_MINOR) = 0;`

 `virtual void unsubscribe(service_t _service, instance_t _instance, eventgroup_t _eventgroup) = 0;`
change to
` virtual void unsubscribe(service_t _service, instance_t _instance, eventgroup_t _eventgroup, major_version_t _major = DEFAULT_MAJOR) = 0;`

And request_event()/release_event() shall all add versions as the parameter.

zhaoxin39913, 2024-11-22:
I tested the  request_service() for the same instance but with different major versions, and it return error message as below:
`[error] rm_base::request_service service property mismatch (0102): [1234.5678:18:878082192] passed: 34:4294967295`

In  my test I have two devices (A and B) in the network. First device B offers service 0x1234/5678 with majorVersion 0x12. Next device B request service 0x1234/5678 with majorVersion 0x12 and it works. Then device B request service 0x1234/5678 with majorVersion 0x22, it reported such error.

From the source code, it seems like the majorVersion is not taken into consideration when maintaining the service instances available in the network.

_```
void routing_manager_base::request_service(client_t _client,
        service_t _service, instance_t _instance,
        major_version_t _major, minor_version_t _minor) {
    auto its_info = find_service(_service, _instance);
    if (its_info) {
        if ((_major == its_info->get_major()
                || DEFAULT_MAJOR == its_info->get_major()
                || ANY_MAJOR == _major)
                && (_minor <= its_info->get_minor()
                    || DEFAULT_MINOR == its_info->get_minor()
                    || _minor == ANY_MINOR)) {
            its_info->add_client(_client);
        } else {
            VSOMEIP_ERROR << "rm_base::request_service service property mismatch ("
                    << std::hex << std::setfill('0')
                    << std::setw(4) << _client << "): ["
                    << std::setw(4) << _service << "."
                    << std::setw(4) << _instance << ":"
                    << std::dec
                    << static_cast<std::uint32_t>(its_info->get_major()) << ":"
                    << its_info->get_minor() << "] passed: "
                    << static_cast<std::uint32_t>(_major) << ":"
                    << _minor;
        }
    }
}
```_

zhaoxin39913, 2024-11-22:
Hi @KareemLMR,

From my investigation, It seems like the **PRS_SOMEIPSD_00512** in SOME/IP SD Spec is totally not supported in vsomeip library, can you confirm ?

KareemLMR, 2024-11-23:
Hello @zhaoxin39913
I need to check something. Open the vsomeip source code and navigate to the file implementation/routing/src/routing_manager_base.cpp at this function routing_manager_base::offer_service to be specific
[image]
And add a log to this if condition marked by the red arrow:
if (!its_info->is_local()) {
return false;
}
and if you can add more logs throughout this function then recompile and reinstall the binaries and let me see the logs of your test. If the log you added in that if condition appeared, then there is an issue as you said. If not then it is something else we will see.

KareemLMR, 2024-11-28:
Ok, I see the log you added appeared. That's good. I saw what I wanted to see. Let me try a change at my side and send it to you to try when it is successful.

zhaoxin39913, 2024-12-12:
Hi @KareemLMR ,

Any updates for your patch ?

KareemLMR, 2024-12-12:
Hi @zhaoxin39913 
Unfortunately, the change was much larger than what I expected at the beginning as all data structures and searching functions in the code are set to deal with only service and instance and I had to change the strategy many times after starting to decrease the impact of the change but the good news I have decided the most suitable one and I am about to finish the implementation. I will send it once done.

zhaoxin39913, 2024-12-13:
Thanks @KareemLMR  for sharing the update.

zhaoxin39913, 2024-12-31:
Hi @KareemLMR , 

Thanks for sharing the sample. One question here, for client API  request_service(), Do we still support passing ANY_MAJOR when build the **unique_version_t** parater ? Or we must pass a non-0xFF version ? 

```
virtual void request_service(service_t _service, unique_version_t _unique,
            major_version_t _major = ANY_MAJOR,
            minor_version_t _minor = ANY_MINOR,
            bool _use_exclusive_proxy = false) = 0;
```

KareemLMR, 2024-12-31:
Hello @zhaoxin39913
Unfortunately the current patch set in case of using unique version to use the new feature of offering multiple instances with different major versions you have to request the service specifically by its major version
Using 0xFF as major won't work as expected in the current version but I was planning to fix that in another patch set just after checking your test results to know if it is the only thing missing before committing another change.
Have you tested it with requesting the specific major version and the rest of APIs?

zhaoxin39913, 2025-01-08:
Hi @KareemLMR , 

Currently I am only testing the client APIs which request different major versions for the same instance ID. It's working fine so far. I will continue to test those server APIs and keep you updated once my test is done.

zhaoxin39913, 2025-01-14:
Hi @KareemLMR,

Update some test results here for the patch(let's call it 3.5.3p) here. The server app can offer two major version instances for the same service/instance ID successfully. And the client app can also detect different version instances when registering different major version availability handlers.

However the basic request-response and event notification don't work in the patch. Here are my test setup:

The server app running on device B offers service 0x238/0x1 and always sends response with "Version4.1: Reply Cnt=X" for any methods. In the meantime it periodically sends event 0x4465/0x8778 with payload "Version4.1: Event Cnt=X“ every 5s.
The client app running on device A request service 0x238/0x1 and always send request every 1s. And register message handler to receive both response and event from device B.

**Test Case1:** Both client app and server app are running with vsomeip 3.5.3p. The client app log is as below:
```
2025-01-14 16:19:08.551257  [info] Using configuration file: "/home/zhax/work/COVESA/multiple_majVersion_patch_src/config/vsomeip.json".
2025-01-14 16:19:08.553718  [info] Parsed vsomeip configuration in 1ms
2025-01-14 16:19:08.554187  [info] Configuration module loaded.
2025-01-14 16:19:08.554483  [info] Security disabled!
2025-01-14 16:19:08.554704  [info] Initializing vsomeip (3.5.3) application "request-sample".
2025-01-14 16:19:08.555695  [info] Instantiating routing manager [Host].
2025-01-14 16:19:08.560971  [info] create_routing_root: Routing root @ /tmp/vsomeip-0
2025-01-14 16:19:08.567016  [info] Service Discovery enabled. Trying to load module.
2025-01-14 16:19:08.587358  [info] Service Discovery module loaded.
2025-01-14 16:19:08.591667  [info] Application(request-sample, 0103) is initialized (11, 100).
2025-01-14 16:19:08.592841  [info] REQUEST(0103): [0238.0001:4.4294967295]
2025-01-14 16:19:08.602663  [info] create_local_server: Listening @ /tmp/vsomeip-103
2025-01-14 16:19:08.607122  [info] REGISTER EVENT(0103): [0238.0001.8778:is_provider=false]
2025-01-14 16:19:08.608108  [info] SUBSCRIBE(0103): [0238.0001.4465:ffff:4]
2025-01-14 16:19:08.608728  [info] Starting vsomeip application "request-sample" (0103) using 2 threads I/O nice 0
2025-01-14 16:19:08.611914  [debug] Thread created. Number of active threads for request-sample : 1
2025-01-14 16:19:08.613286  [info] main dispatch thread id from application: 0103 (request-sample) is: ffff90f5f060 TID: 6430
2025-01-14 16:19:08.614579  [info] Client [0103] routes unicast:192.168.225.3, netmask:255.255.255.0
Service [238.1:4] is NOT available.
2025-01-14 16:19:08.614740  [info] shutdown thread id from application: 0103 (request-sample) is: ffff9074f060 TID: 6431
2025-01-14 16:19:08.621289  [info] Watchdog is disabled!
2025-01-14 16:19:08.623689  [info] io thread id from application: 0103 (request-sample) is: ffff92c05020 TID: 6427
2025-01-14 16:19:08.623910  [info] io thread id from application: 0103 (request-sample) is: ffff8f72f060 TID: 6433
2025-01-14 16:19:08.625012  [info] vSomeIP 3.5.3 | (default)
2025-01-14 16:19:08.630391  [info] Network interface "eth0" state changed: up
2025-01-14 16:19:08.635098  [info] Route "default route (0.0.0.0/0) if: eth0 gw: 192.168.225.1" state changed: up
2025-01-14 16:19:08.639554  [debug] Joining to multicast group 224.0.0.1 from 192.168.225.3
2025-01-14 16:19:08.641245  [info] SOME/IP routing ready.
2025-01-14 16:19:08.644673  [info] udp_server_endpoint_impl<multicast>: SO_RCVBUF is: 212992 (1703936) local port:30490
2025-01-14 16:19:08.647025  [warning] Route "224.0.0.1/32 if: eth0 gw: n/a" state changed: up
2025-01-14 16:19:10.713993  [info] endpoint_manager_impl::create_remote_client: 192.168.225.1:13501 reliable: 0 using local port: 0
2025-01-14 16:19:10.716441  [info] udp_client_endpoint_impl::connect: SO_RCVBUF is: 212992 (1703936) local port:0 remote:192.168.225.1:13501
2025-01-14 16:19:10.718382  [debug] routing_manager_stub::on_offer_service: ON_OFFER_SERVICE(0000): [0238.0001:4.1]
Service [238.1:4] is available.
2025-01-14 16:19:10.522534  [error] Routing info for remote service could not be found! (0103): [0238.0001.0421] 0001
2025-01-14 16:19:11.527970  [error] Routing info for remote service could not be found! (0103): [0238.0001.0421] 0002
2025-01-14 16:19:12.533606  [error] Routing info for remote service could not be found! (0103): [0238.0001.0421] 0003
2025-01-14 16:19:13.539502  [error] Routing info for remote service could not be found! (0103): [0238.0001.0421] 0004
2025-01-14 16:19:14.545933  [error] Routing info for remote service could not be found! (0103): [0238.0001.0421] 0005
```

**Test Case 2:** Client app is running with vsomeip 3.4.10, server app is running with vsomeip 3.5.3p. The client app log is as below:
```
zhax@zhax-desktop:~/work/COVESA/src/vsomeip-3.4.10/build/examples$ ./request-sample 
2025-01-14 16:24:54.412258 [info] Using configuration file: "/home/zhax/work/COVESA/src/config/vsomeip.json".
2025-01-14 16:24:54.415145 [info] Parsed vsomeip configuration in 1ms
2025-01-14 16:24:54.415730 [info] Configuration module loaded.
2025-01-14 16:24:54.416294 [info] Security disabled!
2025-01-14 16:24:54.416635 [info] Initializing vsomeip (3.4.10) application "request-sample".
2025-01-14 16:24:54.419252 [info] Instantiating routing manager [Host].
2025-01-14 16:24:54.427170 [info] create_routing_root: Routing root @ /tmp/vsomeip-0
2025-01-14 16:24:54.431621 [info] Service Discovery enabled. Trying to load module.
[... cut ...]
2025-01-14 16:30:45.244899 [warning] sd::get_eventgroup_reliability: couldn't determine eventgroup reliability type for [0238.0001.4465] using reliability type:  0002
Service [238.1:4] is available.
[Version4.1: Reply Cnt=0.]
[Version4.1: Reply Cnt=1.]
[Version4.1: Reply Cnt=2.]
[Version4.1: Event Cnt=1.]
[Version4.1: Reply Cnt=3.]
[Version4.1: Reply Cnt=4.]
[Version4.1: Reply Cnt=5.]
[Version4.1: Reply Cnt=6.]
[Version4.1: Reply Cnt=7.]
[Version4.1: Event Cnt=2.]
[Version4.1: Reply Cnt=8.]
2025-01-14 16:30:55.215905 [info] vSomeIP 3.4.10 | (default)
[Version4.1: Reply Cnt=9.]
[Version4.1: Reply Cnt=10.]
```

zhaoxin39913, 2025-01-14:
Here is the client app source code:

```
#include <vsomeip/vsomeip.hpp>
#include <iomanip>
#include <iostream>
#include <vector>
#include <string>
#include <chrono>
#include <thread>
#include "sample-ids.hpp"

class requestSample
{
    public:
        requestSample() :
            app_(vsomeip::runtime::get()->create_application("request-sample")),
            service2Available_(false),
            service4Available_(false),
            sender_(std::bind(&requestSample::run, this))
        {
        }

	void init()
        {
            app_->init();
            app_->register_message_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID,vsomeip::ANY_METHOD,
               std::bind(&requestSample::on_message, this, std::placeholders::_1));

            request_ = vsomeip::runtime::get()->create_request();
            request_->set_service(SAMPLE_SERVICE_ID);
            request_->set_instance(SAMPLE_INSTANCE_ID);
            request_->set_method(SAMPLE_METHOD_ID);
            request_->set_reliable(false);

#if 0
            vsomeip_v3::unique_version_t unique2 = vsomeip_v3::get_unique_version(SAMPLE_INSTANCE_ID, 0x2);
	    vsomeip_v3::unique_version_t unique4 = vsomeip_v3::get_unique_version(SAMPLE_INSTANCE_ID, 0x4);
            app_->request_service(SAMPLE_SERVICE_ID, unique2, 0x2);
            app_->request_service(SAMPLE_SERVICE_ID, unique4, 0x4);
            app_->register_availability_handler(SAMPLE_SERVICE_ID, unique2,
                std::bind(&requestSample::on_availability, this, 0x2,
                          std::placeholders::_1, std::placeholders::_2, std::placeholders::_3), 0x2);
            app_->register_availability_handler(SAMPLE_SERVICE_ID, unique4,
                std::bind(&requestSample::on_availability, this, 0x4,
                          std::placeholders::_1, std::placeholders::_2, std::placeholders::_3), 0x4);

            app_->request_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, 0x2);
	    app_->register_availability_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID,
                std::bind(&requestSample::on_availability, this, 0x2,
                          std::placeholders::_1, std::placeholders::_2, std::placeholders::_3), 0x2);
#endif

            app_->request_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, 0x4);
            app_->register_availability_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID,
                std::bind(&requestSample::on_availability, this, 0x4,
                          std::placeholders::_1, std::placeholders::_2, std::placeholders::_3), 0x4);

            std::set<vsomeip::eventgroup_t> its_groups;
            its_groups.insert(SAMPLE_EVENTGROUP_ID);
            app_->request_event(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_EVENT_ID, its_groups);
            app_->subscribe(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_EVENTGROUP_ID, 0x4);
        }

	void start()
        {
            app_->start();
        }

        void run()
        {
            while(1)
            {
                if (service2Available_)
                {
                    request_->set_interface_version(0x2);
                    app_->send(request_);
                }

                if (service4Available_)
                {
                    request_->set_interface_version(0x4);
                    app_->send(request_);
                }

                std::this_thread::sleep_for(std::chrono::milliseconds(1000));
            }
	}

        void on_message(const std::shared_ptr<vsomeip::message> & _response)
        {
            std::shared_ptr<vsomeip::payload> payload = _response->get_payload();
            vsomeip::length_t length = payload->get_length();
            const vsomeip::byte_t* data = payload->get_data();
            std::string response_str(data, data + length);

            std::cout << response_str  << std::endl;     
	}

        void on_availability(uint8_t majver, vsomeip::service_t service, vsomeip::instance_t instance, bool is_available)
        {
            std::cout << "Service ["
            << std::hex << std::setfill('0') << std::setw(4) <<  std::setw(2) << service << "." << instance << ":" <<  +majver << "] is "
	    << (is_available ? "available." : "NOT available.") << std::endl;

            if (majver == 0x2)
	    {
               if (is_available)
               {
                   service2Available_ = true;
               }
               else
               {
                   service2Available_ = false;
	       }
               return;
            }

            if (majver == 0x4)
            {
               if (is_available)
               {
                   service4Available_ = true;
               }
               else
               {
                   service4Available_ = false;
               }
               return;
            }	    
        }

    private:
        std::shared_ptr<vsomeip::application> app_;
        std::shared_ptr<vsomeip::message> request_;
        bool service2Available_;
        bool service4Available_;
        std::thread sender_;
};
```

int main()
{
    requestSample client;
    client.init();
    client.start();
    return 0;
}

KareemLMR, 2025-01-14:
Hello @zhaoxin39913 
I think I found the issue. I will share a fix for that with you today.

KareemLMR, 2025-01-14:
Thank you for the feedback
I believe this should fix the issue: https://github.com/KareemLMR/vsomeip
Please try the test again and tell me the results but this time you will need to set the major version of the request in your code like this:
request_->set_instance(major)
Also the message handler availability_state_handler_t is now of type:
typedef std::function< void (service_t, unique_version_t, availability_state_e) > availability_state_handler_t;
instead of:
typedef std::function< void (service_t, instance_t, availability_state_e) > availability_state_handler_t;

And don't forget to use unique_version_t in APIs that need them in the latest patch I just shared with you.

zhaoxin39913, 2025-01-15:
> Thank you for the feedback I believe this should fix the issue: https://github.com/KareemLMR/vsomeip Please try the test again and tell me the results but this time you will need to set the major [...]

Hi @KareemLMR ,

We need to keep the vsomeip API backward compatible in the patch, otherwise we will encounter regression issue, eg. COMMON API proxy/stub code is still using the old vsomeip API.  If we introduce new API for this feature, all related things need to be considered as well, then it will be a very big impact.

KareemLMR, 2025-01-15:
I believe it is backward compatible as I didn't change any API in the last patch I shared. I said that last line just to confirm with you that you correctly used it from the older patch.

zhaoxin39913, 2025-01-15:
> I believe it is backward compatible as I didn't change any API in the last patch I shared. I said that last line just to confirm with you that you correctly used it from the older patch. You can even try it and tell me what you have found

Thanks @KareemLMR , that's great!  What I expect is the old APIs are still working in the network only has single major version instance. And we can use the new APIs to handle multiple major version instances.

zhaoxin39913, 2025-01-16:
Hi @KareemLMR,

Tested the second patch (3.5.3p2) you shared for only client app(server app using 3.4.10), still see the same error log on client side:

```
2025-01-14 16:19:10.716441  [info] udp_client_endpoint_impl::connect: SO_RCVBUF is: 212992 (1703936) local port:0 remote:192.168.225.1:13501
2025-01-14 16:19:10.718382  [debug] routing_manager_stub::on_offer_service: ON_OFFER_SERVICE(0000): [0238.0001:4.1]
Service [238.1:4] is available.
2025-01-14 16:19:10.522534  [error] Routing info for remote service could not be found! (0103): [0238.0001.0421] 0001
2025-01-14 16:19:11.527970  [error] Routing info for remote service could not be found! (0103): [0238.0001.0421] 0002
2025-01-14 16:19:12.533606  [error] Routing info for remote service could not be found! (0103): [0238.0001.0421] 0003
2025-01-14 16:19:13.539502  [error] Routing info for remote service could not be found! (0103): [0238.0001.0421] 0004
2025-01-14 16:19:14.545933  [error] Routing info for remote service could not be found! (0103): [0238.0001.0421] 0005 
```

git log :

```
commit 531ca1f7aadd89568e4be41f8b5484e58f0e8ec3 (HEAD -> master, origin/master, origin/HEAD)
Author: Kareem Ibrahim <karem.ibrahim46@gmail.com>
Date:   Tue Jan 14 21:41:49 2025 +0200

    change instance_t type in sync_handler to unique_version_t.

commit d7b01668cb415ea6de6b4f39393da978cf3243ab
Author: Kareem Ibrahim <karem.ibrahim46@gmail.com>
Date:   Tue Jan 14 19:45:47 2025 +0200

    Fix request/response issue.

commit 59c8bb5db2644bdf0125c24ef5f8e88cf3e16c10
Author: Kareem Ibrahim <karem.ibrahim46@gmail.com>
Date:   Fri Dec 20 02:11:23 2024 +0200

    Support handling multiple services with the same service and instance IDs but different major version.
```

KareemLMR, 2025-01-16:
Hello @zhaoxin39913 
Just a question, did you set the major version in the response as I said in my last comment request_->set_instance(major)?

KareemLMR, 2025-01-16:
I tried to reproduce that issue of "[error] Routing info for remote service could not be found!". It only happened if I didn't set the major version in request : request_->set_instance(major). When I set it, everything works fine. So I said you might forgot it as it was not in the old API but unfortunately this change in the API can't be avoided as the old API only takes service and instance but if we have multiple services with same instance but different major version, the send API won't work (how will it differentiate between them). So I believe it is impossible to implement this feature without this change, do you think of something different?

We can negotiate about that but can you try setting the major version as I told you just to check everything is fine?

zhaoxin39913, 2025-01-18:
> Hello [@zhaoxin39913](https://github.com/zhaoxin39913) Just a question, did you set the major version in the response as I said in my last comment request_->set_instance(major)?

Hi @KareemLMR ,

Are you talking about the server API or client API ?

closed 2025-01-18 by zhaoxin39913

zhaoxin39913, 2025-01-18:
Hi @KareemLMR,

vsomeip already has message API **set_interface_version**() to do the same thing, which is duplicate with set_major_version(). Can you confirm ?

zhaoxin39913, 2025-01-18:
> Hello [@zhaoxin39913](https://github.com/zhaoxin39913) Just a question, did you set the major version in the response as I said in my last comment request_->set_instance(major)?

Hi @KareemLMR ,

vsomeip already has message API set_interface_version() to do the same thing, which is duplicate with set_major_version(). Can you confirm ?

KareemLMR, 2025-01-18:
I was thinking of that too. However, in the vsomeip code intervace_version was never used as major_version. They even have different types although both are uint8_t (interface_version_t and major_version_t). I tried to check the SOME/IP Specification Document, it is defined as "major version of the service interface" but in some requirments like this one attached below seems diffrent from the fixed service major version. It seems more like session id or protocol version so I am not sure so I didn't use it fearing it might not be known globally by vsomeip users that they are the same. However, anyway this will not affect my change much if we made sure both interface version and major version are the same. I will just replace the call of my newely defined method with this one in the code and it will be fine, if you confirm it works fine with my method set_major_version()

[image]

zhaoxin39913, 2025-01-20:
Hi @KareemLMR,

It's very clearly defined in AUTOSAR SOME/IP Spec [**PRS_SOMEIP_00053**] that the interface version is major version:

```
4.1.2.7 Interface Version [8 Bit]
[PRS_SOMEIP_00053] Interface Version shall be an 8 Bit field that contains the Major
Version of the Service Interface.c(RS_SOMEIP_00003, RS_SOMEIP_00027)** 
```
And according to the Spec, the interfaceVer/MajVer is used to track the message format(request/response/event) change while the minor version is used to track the implementation change for the same interfaceVer/MajVer.

So could you please update the patch to use the set_interface_version() directly? Thanks to set_interface_version() , we can make our changes compatible with old APIs :).

zhaoxin39913, 2025-02-12:
Hi @KareemLMR ,

Happy new year!  I am just back from vocation.

I have tested your new patch 3.5.3p3. **However, it still does not work**.  

In my test I see the availabilityHandler is triggered but the msgHandler is always not triggered which means my client app is not able to receive any response/event messages from server. I also opened the wireshirk and found the request and response are present, but the event is not present on client system side.

Then I switch to use official 3.4.10 with the same client app, and it works as expected. all request/response/event communications are working fine.

So could you please help to test and confirm your patch and make sure no regression issues ?

KareemLMR, 2025-02-12:
Hi @zhaoxin39913 
I can share with you the example codes I tested the change with. So at first, can you please run this example just as it is on your environment without changing anything to check if it is working just as on mine. If it is working then I think we can compare the 2 codes and reuse part by part from it until we identify the issue. If it is not working then there is something different between our 2 environments we can investigate.

Service.cpp:
#include <iomanip>
#include <iostream>
#include <sstream>

#include <vsomeip/vsomeip.hpp>

#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678
#define SAMPLE_METHOD_ID 0x0421

std::shared_ptr<vsomeip::application> app;

void on_message(const std::shared_ptr<vsomeip::message> &_request) {

    std::shared_ptr<vsomeip::payload> its_payload = _request->get_payload();
    vsomeip::length_t l = its_payload->get_length();

    // Get payload
    std::stringstream ss;
    for (vsomeip::length_t i=0; i<l; i++) {
       ss << std::setw(2) << std::setfill('0') << std::hex
          << (int)*(its_payload->get_data()+i) << " ";
    }

    std::cout << "SERVICE: Received message with Client/Session ["
        << std::setw(4) << std::setfill('0') << std::hex << _request->get_client() << "/"
        << std::setw(4) << std::setfill('0') << std::hex << _request->get_session() << "] "
        << ss.str() << std::endl;

    // Create response
    std::shared_ptr<vsomeip::message> its_response = vsomeip::runtime::get()->create_response(_request);
    its_payload = vsomeip::runtime::get()->create_payload();
    std::vector<vsomeip::byte_t> its_payload_data;
    for (int i=9; i>=0; i--) {
        its_payload_data.push_back(i % 256);
    }
    its_payload->set_data(its_payload_data);
    its_response->set_payload(its_payload);
    app->send(its_response);
}

int main() {

   app = vsomeip::runtime::get()->create_application("World");
   app->init();
   vsomeip_v3::unique_version_t unique = vsomeip_v3::get_unique_version(SAMPLE_INSTANCE_ID, 10);
   app->register_message_handler(SAMPLE_SERVICE_ID, unique, SAMPLE_METHOD_ID, on_message);
   app->offer_service(SAMPLE_SERVICE_ID, unique, 10, 0);
   app->start();
}

Client.cpp:
#include <iomanip>
#include <iostream>
#include <sstream>

#include <condition_variable>
#include <thread>

#include <vsomeip/vsomeip.hpp>

#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678
#define SAMPLE_METHOD_ID 0x0421

std::shared_ptr< vsomeip::application > app;
std::mutex mutex;
std::condition_variable condition;

void run() {
  std::unique_lock<std::mutex> its_lock(mutex);
  condition.wait(its_lock);

  std::shared_ptr< vsomeip::message > request;
  request = vsomeip::runtime::get()->create_request();
  request->set_service(SAMPLE_SERVICE_ID);
  request->set_instance(SAMPLE_INSTANCE_ID);
  request->set_interface_version(10);
  request->set_method(SAMPLE_METHOD_ID);

  std::shared_ptr< vsomeip::payload > its_payload = vsomeip::runtime::get()->create_payload();
  std::vector< vsomeip::byte_t > its_payload_data;
  for (vsomeip::byte_t i=0; i<10; i++) {
      its_payload_data.push_back(i % 256);
  }
  its_payload->set_data(its_payload_data);
  request->set_payload(its_payload);
  while(1){
  	app->send(request);
        std::this_thread::sleep_for(std::chrono::milliseconds(1000));
  }
}

void on_message(const std::shared_ptr<vsomeip::message> &_response) {

  std::shared_ptr<vsomeip::payload> its_payload = _response->get_payload();
  vsomeip::length_t l = its_payload->get_length();

  // Get payload
  std::stringstream ss;
  for (vsomeip::length_t i=0; i<l; i++) {
     ss << std::setw(2) << std::setfill('0') << std::hex
        << (int)*(its_payload->get_data()+i) << " ";
  }

  std::cout << "CLIENT: Received message with Client/Session ["
      << std::setw(4) << std::setfill('0') << std::hex << _response->get_client() << "/"
      << std::setw(4) << std::setfill('0') << std::hex << _response->get_session() << "] "
      << ss.str() << std::endl;
}

void on_availability(vsomeip::service_t _service, vsomeip::instance_t _instance, bool _is_available) {
    std::cout << "CLIENT: Service ["
            << std::setw(4) << std::setfill('0') << std::hex << _service << "." << _instance
            << "] is "
            << (_is_available ? "available." : "NOT available.")
            << std::endl;
    condition.notify_one();
}

int main() {

    app = vsomeip::runtime::get()->create_application("Hello");
    app->init();
    vsomeip_v3::unique_version_t unique = vsomeip_v3::get_unique_version(SAMPLE_INSTANCE_ID, 10);
    app->register_availability_handler(SAMPLE_SERVICE_ID, unique, on_availability);
    app->request_service(SAMPLE_SERVICE_ID, unique, 10, 0);
    app->register_message_handler(SAMPLE_SERVICE_ID, unique, SAMPLE_METHOD_ID, on_message);
    std::thread sender(run);
    app->start();
}

KareemLMR, 2025-02-12:
On running these 2 examples here are the logs:
Server Side:

[image]

Client Side:

[image]

If I changed the service major version or the number in set_interface_version() of my client, I got the same error you got as expected because client can no longer find that service. So try that and tell me if you have the same test results on your environment.

zhaoxin39913, 2025-02-13:
Hi @KareemLMR ,

Thanks for sharing your code and test log.

I checked your code and found the only difference is both your client app and server app have called **vsomeip_v3::get_unique_version()** before offer service and request service.  However, as we discussed we need to keep backward compatible and no regressing issues for old applications. So please remove "vsomeip_v3::get_unique_version" in both your client app and server app and test again,  since in old application and commonapi stub/proxy code get_unique_version() is never used.

zhaoxin39913, 2025-02-24:
Hi @KareemLMR ,

Any updates for the new patch ?

KareemLMR, 2025-02-25:
Yes, I have made the change and tested it. You can try it now with the latest patch I just pushed.
Here is the updated version of the server client test apps:
`#include <iomanip>
#include <iostream>
#include <sstream>

#include <vsomeip/vsomeip.hpp>

#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678
#define SAMPLE_METHOD_ID 0x0421

std::shared_ptr<vsomeip::application> app;

void on_message(const std::shared_ptr<vsomeip::message> &_request) {

    std::shared_ptr<vsomeip::payload> its_payload = _request->get_payload();
    vsomeip::length_t l = its_payload->get_length();

    // Get payload
    std::stringstream ss;
    for (vsomeip::length_t i=0; i<l; i++) {
       ss << std::setw(2) << std::setfill('0') << std::hex
          << (int)*(its_payload->get_data()+i) << " ";
    }

    std::cout << "SERVICE: Received message with Client/Session ["
        << std::setw(4) << std::setfill('0') << std::hex << _request->get_client() << "/"
        << std::setw(4) << std::setfill('0') << std::hex << _request->get_session() << "] "
        << ss.str() << std::endl;

    // Create response
    std::shared_ptr<vsomeip::message> its_response = vsomeip::runtime::get()->create_response(_request);
    its_payload = vsomeip::runtime::get()->create_payload();
    std::vector<vsomeip::byte_t> its_payload_data;
    for (int i=9; i>=0; i--) {
        its_payload_data.push_back(i % 256);
    }
    its_payload->set_data(its_payload_data);
    its_response->set_payload(its_payload);
    app->send(its_response);
}

int main() {

   app = vsomeip::runtime::get()->create_application("World");
   app->init();
   app->register_message_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_METHOD_ID, on_message, 10);
   app->offer_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, 10, 0);
   app->start();
}
`
`#include <iomanip>
#include <iostream>
#include <sstream>

#include <condition_variable>
#include <thread>

#include <vsomeip/vsomeip.hpp>

#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678
#define SAMPLE_METHOD_ID 0x0421

std::shared_ptr< vsomeip::application > app;
std::mutex mutex;
std::condition_variable condition;

void run() {
  std::unique_lock<std::mutex> its_lock(mutex);
  condition.wait(its_lock);

  std::shared_ptr< vsomeip::message > request;
  request = vsomeip::runtime::get()->create_request();
  request->set_service(SAMPLE_SERVICE_ID);
  request->set_instance(SAMPLE_INSTANCE_ID);
  request->set_interface_version(10);
  request->set_method(SAMPLE_METHOD_ID);

  std::shared_ptr< vsomeip::payload > its_payload = vsomeip::runtime::get()->create_payload();
  std::vector< vsomeip::byte_t > its_payload_data;
  for (vsomeip::byte_t i=0; i<10; i++) {
      its_payload_data.push_back(i % 256);
  }
  its_payload->set_data(its_payload_data);
  request->set_payload(its_payload);
  while(1){
  	app->send(request);
        std::this_thread::sleep_for(std::chrono::milliseconds(1000));
  }
}

void on_message(const std::shared_ptr<vsomeip::message> &_response) {

  std::shared_ptr<vsomeip::payload> its_payload = _response->get_payload();
  vsomeip::length_t l = its_payload->get_length();

  // Get payload
  std::stringstream ss;
  for (vsomeip::length_t i=0; i<l; i++) {
     ss << std::setw(2) << std::setfill('0') << std::hex
        << (int)*(its_payload->get_data()+i) << " ";
  }

  std::cout << "CLIENT: Received message with Client/Session ["
      << std::setw(4) << std::setfill('0') << std::hex << _response->get_client() << "/"
      << std::setw(4) << std::setfill('0') << std::hex << _response->get_session() << "] "
      << ss.str() << std::endl;
}

void on_availability(vsomeip::service_t _service, vsomeip::instance_t _instance, bool _is_available) {
    std::cout << "CLIENT: Service ["
            << std::setw(4) << std::setfill('0') << std::hex << _service << "." << _instance
            << "] is "
            << (_is_available ? "available." : "NOT available.")
            << std::endl;
    condition.notify_one();
}

int main() {

    app = vsomeip::runtime::get()->create_application("Hello");
    app->init();
    app->register_availability_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, on_availability, 10, 0);
    app->request_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, 10, 0);
    app->register_message_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_METHOD_ID, on_message, 10);
    std::thread sender(run);
    app->start();
}
`

zhaoxin39913, 2025-02-25:
Hi @KareemLMR,

Thanks for the new patch. I will test it and share test result.

BTW, have you performed any regression tests for the event/field notification with the new patch as well?

KareemLMR, 2025-02-26:
And here are the logs:
Server Side:

[image]

Client Side:

[image]

You can try them and tell me the results too.

yunqiang512, 2025-03-18:
Hi @KareemLMR ，

After I updated the patch, I found that the service side can't send the offer service SD message.  So I tested the following cases: (patch commit 027dd3163222d73cf0ab48f72e7f038c26606503)

(1) Test the vsomeip lib reference case "vsomeip/examples/hello_world" for end-to-end communication：
With version 3.5.3, the offer service can be seen in the pcap trace. After updating the patch, there is no offer service.

(2) Test the use case you shared above for local loopback:
If run the client first and then the service, client can receive the event just as you shared.
But if run the service first and then the client, client can't receive the event.

I've tried to upload the log picture, but the upload keeps failing. Could you please help check the above scenarios?

zhaoxin39913, 2025-03-19:
Thanks @yunqiang512  for sharing the test report.

Hi @KareemLMR ,

According to our engineer test result,  we are still seeing regression issues in the new patch.  Could you please help to run all the vsomeip sample apps for RPC test and make sure there are no regressions? We can only use the patch until all the regression issues get fixed.

KareemLMR, 2025-03-19:
Thanks for sharing the test results. I will check these cases.

KareemLMR, 2025-03-21:
Hi @zhaoxin39913 @yunqiang512 
I found the root cause of the issue causing event not to be received correctly if server is run before client and fixed it. There was a function I forgot to replace instances in it with unique versions in the stub. It is now fixed in the latest f39b4b2deadb9d0c0014fd89acad89d13d50defe you can also try it. The event is now working whatever which app you start first.

KareemLMR, 2025-03-21:
But there shouldn't be issues in the offer as it is already working with my test apps. Can you share more info about the test steps you did in:
"(1) Test the vsomeip lib reference case "vsomeip/examples/hello_world" for end-to-end communication：
With version 3.5.3, the offer service can be seen in the pcap trace. After updating the patch, there is no offer service."
@yunqiang512

zhaoxin39913, 2025-03-21:
Hi @KareemLMR ,

The latest patch you shared 3.5.3p5 is still not working with my legacy client sample app. In my test I see the response is received on client device through wireshark, but the response message handler is NEVER triggered.  However the same app is working fine with vsomeip 3.4.10 libarary.

Have you tested vsomeip client/server sample apps between two devices using your patch?

KareemLMR, 2025-03-21:
Hi @zhaoxin39913 
That's strange as my test app is working, but let's investigate. Could you share your test apps with me to check?
And also have you tried my shared test apps?

zhaoxin39913, 2025-03-22:
Hi @KareemLMR ,

Actually I already shared my client app before in this ticket, but here I share it again for you to check.

Note the server app on service device is always running with vsomeip 3.4.10.

```
#include <vsomeip/vsomeip.hpp>
#include <iomanip>
#include <iostream>
#include <vector>
#include <string>
#include <chrono>
#include <thread>
#include "sample-ids.hpp"

class requestSample
{
    public:
        requestSample() :
            app_(vsomeip::runtime::get()->create_application("request-sample")),
            service2Available_(false),
            service4Available_(false),
            sender_(std::bind(&requestSample::run, this))
        {
        }

	void init()
        {
            app_->init();
            app_->register_message_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID,vsomeip::ANY_METHOD,
               std::bind(&requestSample::on_message, this, std::placeholders::_1));

            request_ = vsomeip::runtime::get()->create_request();
            request_->set_service(SAMPLE_SERVICE_ID);
            request_->set_instance(SAMPLE_INSTANCE_ID);
            request_->set_method(SAMPLE_METHOD_ID);
            request_->set_reliable(false);

            app_->request_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, 0x2);
            app_->register_availability_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID,
                std::bind(&requestSample::on_availability, this, 0x2,
                          std::placeholders::_1, std::placeholders::_2, std::placeholders::_3), 0x2);

            std::set<vsomeip::eventgroup_t> its_groups;
            its_groups.insert(SAMPLE_EVENTGROUP_ID);
            app_->request_event(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_EVENT_ID, its_groups);
            app_->subscribe(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_EVENTGROUP_ID, 0x2);
        }

	void start()
        {
            app_->start();
        }

        void run()
        {
            while(1)
            {
                if (service2Available_)
                {
                    request_->set_interface_version(0x2);
                    app_->send(request_);
                }

                if (service4Available_)
                {
                    request_->set_interface_version(0x4);
                    app_->send(request_);
                }

                std::this_thread::sleep_for(std::chrono::milliseconds(1000));
            }
	}

        void on_message(const std::shared_ptr<vsomeip::message> & _response)
        {
            std::shared_ptr<vsomeip::payload> payload = _response->get_payload();
            vsomeip::length_t length = payload->get_length();
            const vsomeip::byte_t* data = payload->get_data();
            std::string response_str(data, data + length);

            std::cout << response_str  << std::endl;     
	}

        void on_availability(uint8_t majver, vsomeip::service_t service, vsomeip::instance_t instance, bool is_available)
        {
            std::cout << "Service ["
            << std::hex << std::setfill('0') << std::setw(4) <<  std::setw(2) << service << "." << instance << ":" <<  +majver << "] is "
	    << (is_available ? "available." : "NOT available.") << std::endl;

            if (majver == 0x2)
	    {
               if (is_available)
               {
                   service2Available_ = true;
               }
               else
               {
                   service2Available_ = false;
	       }
               return;
            }

            if (majver == 0x4)
            {
               if (is_available)
               {
                   service4Available_ = true;
               }
               else
               {
                   service4Available_ = false;
               }
               return;
            }	    
        }

    private:
        std::shared_ptr<vsomeip::application> app_;
        std::shared_ptr<vsomeip::message> request_;
        bool service2Available_;
        bool service4Available_;
        std::thread sender_;
};

int main()
{
    requestSample client;
    client.init();
    client.start();
    return 0;
}
```

KareemLMR, 2025-03-22:
Hi @zhaoxin39913 
I think I found the issue in the test app that made it work with vsomeip 3.4.11 but not with my patch. I had to make a change in the APIs register_message_handler() and request_event() but this change can't be avoided as there is no way vsomeip can distinguish services with the same instance but different major version without it as you even referred in your comment: https://github.com/COVESA/vsomeip/issues/797#issuecomment-2490005216.
I took your comment into my account and isolated the interface from unique version and kept APIs with major version as it is for backward compatibility, but for those with no major_version parameter I had to add it like this:

virtual void register_message_handler(service_t _service,
            instance_t _instance, method_t _method,
            const message_handler_t &_handler,
            major_version_t _major = DEFAULT_MAJOR) = 0;

    virtual void request_event(service_t _service, instance_t _instance,
            event_t _event, const std::set<eventgroup_t> &_eventgroups,
            event_type_e _type = event_type_e::ET_EVENT,
            reliability_type_e _reliability = reliability_type_e::RT_UNKNOWN,
            major_version_t _major = DEFAULT_MAJOR) = 0;

For this in my test app you will find:  
app->register_message_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, vsomeip::ANY_METHOD, on_message, 10);

app->request_event(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_EVENT_ID, its_groups, vsomeip::event_type_e::ET_EVENT, vsomeip::reliability_type_e::RT_UNKNOWN, 10);

I was even able to reproduce your issue by removing the major version 10 from the register_message_handler. So you have to add major version to your register_message_handler call or make it for any service instance: app->register_message_handler(vsomeip::ANY_SERVICE, vsomeip::ANY_INSTANCE, vsomeip::ANY_METHOD, on_message);

and also add major version to request_event call. I believe this should work with you, but if it didn't work tell me and I will prepare another environment with server using vsomeip 3.4.11 as you did to check if there are any issues.

zhaoxin39913, 2025-03-24:
Hi @KareemLMR ,

Thanks for identifying the root cause. Some comments here:

1: For register_message_handler(), the default value of parameter "major_version_t _major = DEFAULT_MAJOR" shall be "major_version_t _major = ANY_MAJOR", otherwise it's regression.

2: For unregister_message_handler(), shall also support with parameter "major_version_t _major = ANY_MAJOR".  Simliar to what we already have today in register_availability_handler() and unregister_availability_handler().

3: What's the impact to CommonAPI ? I believe we need to also change the commonapi code generator to support generating proxy code with the new parameters, otherwise it's regression , right ?

zhaoxin39913, 2025-03-25:
Hi @KareemLMR,

Thanks very much for the clarifications. I am OK with your solution now. Yes, I have concern for the commonAPI impact since our applications are implemented in both commonAPI way and vsomeip API way.  And now I understand that we need to update both vsomeip library and commonapi someip-runtime library and change our vsomeip applications once your patch is available. There is no impact to commonAPI applications.

However, since the backward compatibility of vsomeip applications is broken by the version parameter introduced in request_event(), does community have any plan to upgrade the vsomeip version from 3.x to 4.x ? or still stay in 3.x ?

zhaoxin39913, 2025-04-01:
Hi @KareemLMR,

Any update for your new patch ?

KareemLMR, 2025-04-01:
Hi @zhaoxin39913 
To be honest, I am not sure whether that patch would be merged officially. I didn't receive any reviews yet. So I don't know whether the version will be upgraded or not or if it is planned to fix that bug. If you want you can go on with the patch and I will keep supporting if any issues occurred during testing.

duartenfonseca, 2025-04-04:
@KareemLMR @zhaoxin39913 we have something similar to what you developed. it is in the works, but we don't do the interface changes. as soon as we add our part we can discuss all the changes

zhaoxin39913, 2025-04-08:
Hi @KareemLMR ,

OK, thanks for your great support. I will continue to test the patch and let you know our test result later.

yunqiang512, 2025-05-13:
Hi @KareemLMR, 
I tested with tag: f39b4b2, default hello_world case worked fine. But when I added major and minor parameters to offer_service and register_message_handler, PCAP trace did not observe the offer service being sent. 

My modifications only involve adding version parameters to **(un)register_message_handler** and **(stop_)offer_service**. Do I need to change anything else?

Below is my test case:

**hello_world_service.hpp:**
`
// Copyright (C) 2015-2023 Bayerische Motoren Werke Aktiengesellschaft (BMW AG)
// This Source Code Form is subject to the terms of the Mozilla Public
// License, v. 2.0. If a copy of the MPL was not distributed with this
// file, You can obtain one at http://mozilla.org/MPL/2.0/.
#include <vsomeip/vsomeip.hpp>
#include <chrono>
#include <thread>
#include <condition_variable>
#include <mutex>

#if defined ANDROID || defined __ANDROID__
#include "android/log.h"
#define LOG_TAG "hello_world_service"
#define LOG_INF(...) std::fprintf(stdout, __VA_ARGS__), std::fprintf(stdout, "\n"), (void)__android_log_print(ANDROID_LOG_INFO, LOG_TAG, ##__VA_ARGS__)
#define LOG_ERR(...) std::fprintf(stderr, __VA_ARGS__), std::fprintf(stderr, "\n"), (void)__android_log_print(ANDROID_LOG_ERROR, LOG_TAG, ##__VA_ARGS__)
#else
#include <cstdio>
#define LOG_INF(...) std::fprintf(stdout, __VA_ARGS__), std::fprintf(stdout, "\n")
#define LOG_ERR(...) std::fprintf(stderr, __VA_ARGS__), std::fprintf(stderr, "\n")
#endif

static vsomeip::service_t service_id = 0x1111;
static vsomeip::instance_t service_instance_id = 0x2222;
static vsomeip::method_t service_method_id = 0x3333;

class hello_world_service {
public:
    // Get the vSomeIP runtime and
    // create a application via the runtime, we could pass the application name
    // here otherwise the name supplied via the VSOMEIP_APPLICATION_NAME
    // environment variable is used
    hello_world_service() :
                    rtm_(vsomeip::runtime::get()),
                    app_(rtm_->create_application()),
                    stop_(false)
    {
        stop_thread_ = std::thread{&hello_world_service::stop, this};
    }

    ~hello_world_service()
    {
        if (std::this_thread::get_id() != stop_thread_.get_id()) {
            if (stop_thread_.joinable()) {
                stop_thread_.join();
            }
        } else {
            stop_thread_.detach();
        }
    }

    bool init()
    {
        // init the application
        if (!app_->init()) {
            LOG_ERR("Couldn't initialize application");
            return false;
        }

        // register a message handler callback for messages sent to our service
        app_->register_message_handler(service_id, service_instance_id,
                service_method_id,
[... 95 lines cut ...]
        if (hw_srv_ptr != nullptr &&
                (_signal == SIGINT || _signal == SIGTERM))
            hw_srv_ptr->terminate();
    }
#endif

int main(int argc, char **argv)
{
    (void)argc;
    (void)argv;

    hello_world_service hw_srv;
#ifndef VSOMEIP_ENABLE_SIGNAL_HANDLING
    hw_srv_ptr = &hw_srv;
    signal(SIGINT, handle_signal);
    signal(SIGTERM, handle_signal);
#endif
    if (hw_srv.init()) {
        hw_srv.start();
        return 0;
    } else {
        return 1;
    }
}
`

KareemLMR, 2025-05-13:
Hello @yunqiang512 
I didn't try to reproduce the issue the issue yet, but from your description of the issue I think I know the reason for that.
In your .json configuration files of your service you have to define major version too after service and instance IDs so as to differentiate services with the same instance and service IDs but different major version. I pointed to that here in this comment in its last few lines: https://github.com/COVESA/vsomeip/issues/797#issuecomment-2557451525 as it is one of my changes.
So you can try that and tell me. If that didn't work I will check.

To let you know what happened exactly is that when you called the APIs without passing major version, major version was defaulted to 0 and the same happens in json file so you offered the same configured service as external so the offer packets were sent normally. But when you passed the major version to the APIs without updating the json file, vsomeip now thought that the service you are trying to offer is not configured (as the default in json file is major version = 0) so it doesn't know whether it is external or not so the offer packets were not sent.

So to sum up, what you need to do is add major version in your json file service definition like that:
 {
            "service" : "0x1234",
            "instance" : "0x5678",
            "major" : "value",
            "unreliable" : "30509"
  }

zhaoxin39913, 2026-09-08:
Hi @KareemLMR,

I see this feature is supported in 3.7.5 with below description in RN:

**Enable multiple major version offerings**

So does it mean PRS_SOMEIPSD_00512 is fully supported in vsomeip stack for both client and server APIs ?

ricleite, 2026-09-08:
@zhaoxin39913 there's work underway for PRS_SOMEIPSD_00512 but it's a bit more subtle, that note in 3.7.5 only deals with the IPC side
in short: allows different applications under the same routing-manager to offer different major versions for the same service

As far as everything else goes (service-discovery, SOME/IP communication, vsomeip::application API..) it's as flawed as it was before. The plan at the moment is to go through the service-discovery + SOME/IP communication issues in the next releases. I don't want to change the libvsomeip API however, so if you have a case where you need PRS_SOMEIPSD_00512, you'll need to be using a different `vsomeip::application` per each offered/requested different major-version