COVESA/vsomeip #798: [BUG]: <title>QNX 7.1，RUN 3.4.10 notify-sample, can't sent multicast offer service message.

darduck18, 2024-10-28:
### vSomeip Version

v3.4.10

### Boost Version

1.63

### Environment

QNX 7.1

### Describe the bug

run 3.4.10 notify-sample,i can't catch offer service message use tcpdump.
i try iperf3 to send message to 224.224.224.245 , it work properly，and i can catch the message.

### Expected behaviour

offer service message in multicast 224.224.224.245

### Logs and Screenshots

```
/var/data/jing88# ./run_tsf.sh notify-sample --udp
[ 6581.319796]~DLT~2306148~INFO     ~Socket /tmp/dlt cannot be opened (errno=2). Retrying later...
1970-01-01 01:49:42.823238 [warning] Reliability type for event [1234.5678.8777] was not configured Using : RT_UNRELIABLE
1970-01-01 01:49:42.823238 [warning] Reliability type for event [1234.5678.8778] was not configured Using : RT_UNRELIABLE
1970-01-01 01:49:42.823248 [warning] Reliability type for event [1234.5678.8779] was not configured Using : RT_UNRELIABLE
1970-01-01 01:49:42.823248 [info] Using configuration file: "./etc/vsomeip/vsomeip-udp-service.json".
1970-01-01 01:49:42.823248 [info] Parsed vsomeip configuration in 6ms
1970-01-01 01:49:42.823248 [info] Configuration module loaded.
1970-01-01 01:49:42.823248 [info] Security disabled!
1970-01-01 01:49:42.823248 [info] Initializing vsomeip (3.4.10) application "service-sample".
1970-01-01 01:49:42.823258 [info] Instantiating routing manager [Host].
1970-01-01 01:49:42.823258 [info] create_routing_root: Routing root @ /var/vsomeip-0
1970-01-01 01:49:42.823268 [info] Service Discovery enabled. Trying to load module.
1970-01-01 01:49:42.823348 [info] Service Discovery module loaded.
1970-01-01 01:49:42.823358 [info] vsomeip tracing not enabled. . vsomeip service discovery tracing not enabled.
1970-01-01 01:49:42.823358 [info] Application(service-sample, 1277) is initialized (11, 100).
1970-01-01 01:49:42.823358 [info] offer_event: Event [1234.5678.8778] uses configured cycle time 0ms
1970-01-01 01:49:42.823358 [info] REGISTER EVENT(1277): [1234.5678.8778:is_provider=true]
1970-01-01 01:49:42.823358 [info] Starting vsomeip application "service-sample" (1277) using 2 threads I/O nice 255
1970-01-01 01:49:42.823358 [info] main dispatch thread id from application: 1277 (service-sample) is: 7
1970-01-01 01:49:42.823358 [info] shutdown thread id from application: 1277 (service-sample) is: 8
1970-01-01 01:49:42.823368 [info] udp_server_endpoint_impl: SO_RCVBUF is: 1703936 (1703936) local port:30490
1970-01-01 01:49:42.823368 [debug] Joining to multicast group 224.224.224.245 from 192.168.1.2
1970-01-01 01:49:42.823378 [info] create_local_server: Listening @ /var/vsomeip-1277
1970-01-01 01:49:42.823378 [info] OFFER(1277): [1234.5678:0.0] (true)
Setting event (Length=1).
1970-01-01 01:49:42.823378 [info] udp_server_endpoint_impl: SO_RCVBUF is: 1703936 (1703936) local port:30509
1970-01-01 01:49:42.823378 [info] udp_server_endpoint_impl<multicast>: SO_RCVBUF is: 1703936 (1703936) local port:30490
1970-01-01 01:49:42.823378 [info] SOME/IP routing ready.
1970-01-01 01:49:42.823388 [info] Watchdog is disabled!
Application service-sample is registered.
1970-01-01 01:49:42.823398 [info] io thread id from application: 1277 (service-sample) is: a
1970-01-01 01:49:42.823398 [info] io thread id from application: 1277 (service-sample) is: 1
1970-01-01 01:49:42.823398 [info] vSomeIP 3.4.10 | (default)
```

```
/mnt/root# tcpdump -i emac0 host 224.224.224.245
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on emac0, link-type EN10MB (Ethernet), capture size 262144 bytes
01:49:36.306791 IP 192.168.1.2 > 224.224.224.245: igmp leave 224.224.224.245
01:49:42.401797 IP 192.168.1.2 > 224.224.224.245: igmp v2 report 224.224.224.245
01:49:45.047131 IP 192.168.1.2 > 224.224.224.245: igmp v2 report 224.224.224.245
```

```
/mnt/root# netstat -r
Routing tables

Internet:
Destination        Gateway            Flags    Refs      Use    Mtu  Interface
127.0.0.1          127.0.0.1          UH          0      440  33136  lo0
172.16.118/24      link#37            UC          0        0      -  vp20
172.16.119/24      link#38            UC          0        0      -  vp30
192.168.0/24       link#35            UC          0        0      -  vp_la0
192.168.1/24       link#33            UC          3        0      -  emac0
192.168.1.1        link#33            UHLc        1        0      -  emac0
192.168.1.10       link#33            UHL1c       1      854      -  emac0
192.168.1.100      00:e0:4c:a0:19:fa  UHLc        4     1986      -  emac0
192.168.2/24       link#36            UC          0        0      -  vp10
224.224.224.245    192.168.1.1        UGHS        0        0      -  emac0

```

```
{
    "unicast" : "192.168.1.2",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/tmp/vsomeip.log" },
        "dlt" : "false"
    },
    "applications" :
    [
        {
            "name" : "service-sample",
            "id" : "0x1277"
        }
    ],
    "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "unreliable" : "30509",
            "events" :
            [
                {
                    "event" : "0x8777",
                    "is_field" : "true",
                    "update-cycle" : 2000
                },
                {
                    "event" : "0x8778",
                    "is_field" : "true",
                    "update-cycle" : 0
                },
                {
                    "event" : "0x8779",
                    "is_field" : "true"
                }
            ],
            "eventgroups" :
            [
                {
                    "eventgroup" : "0x4455",
                    "events" : [ "0x8777", "0x8778" ]

                },
                {
                    "eventgroup" : "0x4465",
                    "events" : [ "0x8778", "0x8779" ]
                },
                {
                    "eventgroup" : "0x4555",
                    "events" : [ "0x8777", "0x8779" ]
                }
            ]
        },
        {
            "service" : "0x1235",
            "instance" : "0x5678",
            "unreliable" : "30509"
        }
    ],
    "routing" : "service-sample",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}
```

darduck18, 2024-10-29:
<html><body>
<p id="">During my debugging, I found that after executing the following line:</p>
<pre><code class="language-cpp">
offer_debounce_timer_.async_wait(  
        std::bind(&service_discovery_impl::on_offer_debounce_timer_expired,  
                  this, std::placeholders::_1));
</div></code></pre>
<p id="">The <code class=" inline">on_offer_debounce_timer_expired</code> callback was never triggered. In <code class=" inline">application_impl.cpp</code>, <code class=" inline">io_.run()</code> was also executed normally. I tried using the Boost ASIO <code class=" inline">steady_timer</code> and it worked fine on QNX 7.1.
Can you give me some advice on how to check why the timer is not running properly?
</p>
</body>
</html>

closed 2025-05-14 by darduck18

rangasridhar07-droid, 2026-02-12:
i too face the same issue what should i do now?
I use boost 1.76