COVESA/vsomeip #799: [BUG]: Initial Event (FIELDS) Sent as Multicast Instead of Unicast for Clients Exceeding Threshold

Vignesh-Murugappan, 2024-10-29:
### vSomeip Version

v3.4.10

### Boost Version

1.76

### Environment

Ubuntu and Android

### Describe the bug

1. When a new client subscribes, after receiving a successful subscribe acknowledgment, the server sends an "initial event" (for FIELDS) as unicast.
2. If the same client subscribes again for the same event group, no initial event will be sent.

However:

1. When the threshold is set to 1, the expectation is that all events, except the initial event, should be sent over multicast to clients once at least one client is subscribed.
2. The initial event (FIELDS) should be sent as a unicast message to each newly subscribed client.
3. However, when the threshold is set to 1, the initial event is correctly sent over unicast for the first subscribing client. But, for any subsequent clients, the initial event is sent over multicast instead of unicast.

We have observed this issue for the (N+1)th subscribed client when the threshold is set to N.

In our code walkthrough, we observed that the multicast address is updated to an unreliable endpoint in the highlighted code section when the number of subscribed clients (clients_subscribed) is greater than or equal to the threshold.

This is not in-line with AutoSAR requirement **"[PRS_SOMEIPSD_00487] dThe initial events shall be transported using unicast from Server to Client."**

### Reproduction Steps

1. Set the threshold parameter to 1 in the server-side configuration.
2. Start two client applications on separate machines (in this case, two different Ubuntu machines).
3. Start the server application on a third machine.
4. Expected behavior: the initial event should be sent to both clients over unicast.
5. Observed behavior: the initial event is sent to only one client via unicast, while the other client receives the initial event over multicast. This causes the initial event to be broadcast to all subscribed clients, regardless of whether they have already received the initial event.

### Expected behaviour

**Expected behavior**: the initial event should be sent to both clients over unicast.

**Observed behavior**: the initial event is sent to only one client via unicast, while the other client receives the initial event over multicast. This causes the initial event to be broadcast to all subscribed clients, regardless of whether they have already received the initial event.

Vignesh-Murugappan, 2024-10-29:
[image]

zhaoxin39913, 2025-01-24:
It's not true. According to **PRS_SOMEIPSD_00851** in SOME/IP SD Spec:

```
if MULTICAST_THRESHOLD is configured to a value n with n=1, then automated
switching between client service unicast/multicast endpoints and the server service
multicast endpoint is disabled. Only multicast communication to the server
service multicast endpoint is supported.
```

It implies the initial events are completely disabled since the client service endpoints are not used for event communication when threshold=1. So the exepcted behavior is no initial events are sent on server side when client subscribes the event group and only regular events are sent on server side via the server service multicast endpoint later.